openthread/openthread #1612: Test case 9.2.2 problem

hubertmis, 2017-04-18:
I have a problem with test case 9.2.2 for a Leader role. It looks that the harness sends following commands to a cli device:
```
[1] commissioner mgmtset steeringdata ff
[2] commissioner start
[3] commissioner mgmtset sessionid 0 steeringdata ff
[4] commissioner mgmtset sessionid 0 locator 0x400
[5] commissioner mgmtset sessionid 0 steeringdata ff locator 0x400
```

and the device sends following CoAP `/c/cs` messages:
[1] (missing Commissioner Session ID TLV); Steering Data TLV (0xFF)
[2] Commissioner Session ID TLV; Steering Data TLV (0xFF)
[3] Commissioner Session ID TLV; Steering Data TLV (0xFF)
[4] Commissioner Session ID TLV; Border Agent Locator TLV (0x0400) (not allowed TLV)
[5] Commissioner Session ID TLV; Steering Data TLV (0xFF); Border Agent Locator TLV (0x0400) (not allowed TLV)

It seems that the harness does not expect duplicated [2] and [3] messages and it causes test failure. Is it possible to fix this problem in THCI?

librasungirl, 2017-04-21:
The additional /c/cs is introduced since #1486.  I am afraid we could not do anything in THCI for this issue.  
Personally I think Harness should correctly match the /c/cs response and /c/cs request it want to verify, not on some assumption on the message order. [DEV-1545](https://threadgroup.atlassian.net/browse/DEV-1545) is filed to track this.

librasungirl, 2017-04-30:
Finally agreed on updating OT to not introducing the additional /c/cs as the fix #1670.

closed 2017-04-30 by librasungirl