openthread/openthread #1619: MTD vrs FTD builds, and Makefile.am If/Else/Endif issues

DuaneEllis-TI, 2017-04-19:
As discussed in #1467 (Changes to Support IDEs)

These problems are intertwined and share a number of common traits and are being addressed together in a single commit, sadly I don't see a way to break this up.

**PROBLEM 1**   
In src/core/Makefile.am - lists numerous IF/ELSE/ENDIF which choose which files are added or not added to a specific library. While IDEs can provide some level of exclusion, they are not that flexible.

**PROBLEM 2**  
Need both --enable-cli-ftd --enable-cli-mtd, and likewise --enable-mtd

Need granularity such that CLI can be built with MTD only, And NCP can be built using MTD or FTD
Current ./configure solution is: FTD for both, or FTD for Neither

(there are no details for this item below)

**PROBLEM 3**
Linking the CLI application with the CLI library, and NCP with the NCP library

Currently when the CLI (or NCP) library is built - it is built using only *one* of the two internal header configurations, ie: FTD - with all of the structure/class elements, or  MTD with various structure/class elements removed.

This causes problems with link time optimization and static code analysis.

**PROBLEM 4**
Appvayor.

I do not have access to the version of Visual Studio used here, I have Visual Studio 2015, but the project appears to use an older version of Visual Studio, my experience has been:  VS2015 will upgrade the projects... this is a one-way process  

if someone can advise, it would be helpful.

DETAILS FOLLOW

================================================

**PROBLEM 1 - details**

a)  Makefile.am files should not longer use If/Else/Endif type constructs to add/remove files from a build.
Instead, the makefile should generally compile everything.

For example some IDEs - such as Eclipse CDT based IDEs - compile all files "*.c" and "*.cpp" and are of the view - if the file does not belong, or should not be compiled then it should not be present in the directory.  While other IDEs have other schemes, in IDE land - there does not exist a single common way to exclude/include files based upon the build configuration.

**Solution:** Where required, appropriate #if/#else/#endif type statements need to be added to various source files so that the entire quote "translation unit" - effectively becomes a pseudo-empty translation unit.

A number of files did not have, or do not #include "openthread-config.h" - which is required to support the #if/#else/#endif described above, this was added to a number of files.

Result: A number of places additional tests must be applied to the conditional.

For example in several places, the #if SOMETHING becomes: #if SOMETHING && OPENTHREAD_FTD

**PROBLEM 2** Details

This is described above, not expanded here.

**PROBLEM 3** Details

When the NCP or CLI library is built, it uses the internal thread headers, and thus structures and class members.

When the NCP or CLI application is linked, the openthread library effectively uses a different set of header files (either MTD or FTD configuration) - the net result is these internal structures are different.

What is needed is the CLI library must be built in FTD mode, and MTD mode so that it matches the openthread library, these differences are flagged as errors by compilers and static code analysis tools that do "link-time-optimization"

Thus, in all cases we have added two command line defines:

 When building anything-FTD,  "-DOPENTHREAD_FTD=1"  must be defined
When building anything-MTD, "-DOPENTHREAD_MTD=1" must be defined

**PROBLEM 4** - No access to Visual Studio OLD?

I have Visual Studio 2015 - I do not have older versions, I cannot make any of the required changes to support the above for appvayor.

I believe the change is simple, because I believe the Windows solution only builds the FTD configuration.

In the library build projects add:  -DOPENTHREAD_FTD=1

jwhui, 2017-04-21:
Resolved by #1620.

closed 2017-04-21 by jwhui

mstamer, 2017-04-26:
@jwhui @DuaneEllis-TI 
Just curious, does the file "dhcp6_client.cpp" need the "#if OPENTHREAD_ENABLE_DHCP6_CLIENT" macros?  Or has it been decided that the dhcp client should always be present?  

Thanks.

jwhui, 2017-04-26:
@mstamer, good catch!  DHCPv6 Client should be optional.  Would you like to submit a PR?

mstamer, 2017-04-26:
Will do.

DuaneEllis-TI, 2017-04-26:
@jwhui  - Agree, it probably should.

This file also will need to include 'openthread-config.h'
it is an example of several that do not - perhaps it does via other headers

mstamer, 2017-04-26:
missed that info in the ticket (#1663)...  Do you prefer I add that info to this ticket as well?

jwhui, 2017-04-26:
@mstamer, I think it makes sense to make the `#include` change in the same PR.