openthread/openthread #1625: Missing Commissioning Data TLV in full Thread Network Data

hubertmis, 2017-04-19:
According to Thread 1.1.0 8.4.3.2 the Thread Network Data must include an MLE Commissioning Data TLV. But there is no Commissioning Data TLV in the Network Data TLV after initialization. It is added when a commissioner registers or resigns.
I think that Commissioning Data TLV should be added to the Network Data TLV during network data initialization.

jwhui, 2017-04-19:
We noticed the same issue as well when testing against the latest harness.  You are right that Section 8.4.3.2 states that the Commissioning Data TLV must be included in full network data.  At the same time, when no Commissioner is active on the network, it seems completely unnecessary to include the Commissioning Data TLV.

We have filed [SPEC-710](https://threadgroup.atlassian.net/browse/SPEC-710) with the Thread Group for further discussion.

hubertmis, 2017-04-20:
If Commissioner is removed from a network Network Data should include Commissioning Data with incremented Session Id. After any Commissioner is active for any time in the network, Commissioning Data must be present in the Network Data. It seems to be small overhead to include Commissioner Data before first Commissioner connects to the network.

I think we should make OT compliant with Thread specification as it is and in case the specification changes we could change the implementation.

jwhui, 2017-04-20:
I agree.  We'll work towards always including the Commissioning Data TLV.

hubertmis, 2017-04-20:
I already have initial implementation on the branch https://github.com/NordicSemiconductor/openthread/tree/pr/set-commissioner-data-when-becoming-leader
But still working on python tests to pass.

jwhui, 2017-04-20:
Great to hear you're already working on it!  I'll reassign this issue to you :)

PR #1636 (2017-04-21): Set Commissioner Data when becoming Leader
This PR fixes #1625
closed 2017-04-21 by jwhui