openthread/openthread #1627: MT-Safe platform abstraction

srickardti, 2017-04-19:
OpenThread takes advantage of a single threaded view of the processor. Either the User application is running or the stack is running. neither can preempt the other. On a multi-threaded architecture such as an RTOS, the preemption can happen at any time. This becomes an issue if a stack tasklet in running in the stack processing RTOS Task, and a context switch occurs to another RTOS task that tries to call one of the OpenThread APIs.

The pending Riot port of OpenThread (https://github.com/RIOT-OS/RIOT/pull/5552) uses a mutex to lock and unlock the stack functionality. The onerous of locking and unlocking the mutex falls on the caller, and it is assumed that all calls to the OpenThread APIs are protected by this mutex.

It would be good to have a platform abstraction interface for MT environments to make the `C` APIs MT-Safe.

I think these two functions would be enough:
```c
/**
 * Enters the mutex gate for execution of OpenThread APIs.
 *
 * @return A key to be passed to the matching leave function.
 */
uintptr_t otApiEnter(void);

/**
 * Leaves the mutex gate for execution of OpenThread APIs.
 *
 * @param [in] key The value returned by the corresponding lock call.
 */
void otApiLeave(uintptr_t key);
```
And adding them to the beginning and end of the `C` API functions in `src/core/api` should be sufficient to make the `C` APIs MT-Safe.

The mutex object that backs these APIs would have to be a recursive mutex. Meaning that the same task can lock the mutex multiple times, provided they are unlocked in the same order. This would allow callback functions to use the `C` APIs, even though they are being called from a Task that has already locked the mutex.

jwhui, 2017-04-25:
Early on, we had fairly lengthy discussions around what concurrency model the OpenThread API would support.  For simplicity and testability, we landed on a design where the OpenThread API is assumed to be called from a single execution context.  Any mechanisms required to handle concurrency are out-of-scope and left to platform-specific code around the core OpenThread library.

Regarding this specific proposal, we can certainly see that this will work.  However, this places a burden on OpenThread API developers to ensure that the proper calls are in place for any new API's.  This can be easily overlooked as there is no test to ensure that the platform calls are made as part of the API implementation.  At the same time, there are other ways to handle concurrency (e.g. through a request queue that is served by the OpenThread task) which may be favorable depending on the platform software design.  This proposal does not align with that design choice.

srickardti, 2017-05-04:
We've altered our plans to accommodate this. Thanks for considering it though.

closed 2017-05-04 by srickardti