openthread/openthread #1638: Stack reverts to non-volatile settings when changed by application

LuDuda, 2017-04-21:
Recently #1586 has been applied to invalidate Operational Dataset in case one of change of some parameters like channel or PAN ID.

Unfortunatelly i can't see this behaviour using CLI example, at least in a way i would expect this to work:

```
> panid 0x1234
Done
> ifconfig up
Done
> thread start
Done
> state
leader
Done
> thread stop  
Done
> ifconfig down
Done
> channel
11
Done
> channel 14
Done
> ifconfig up
Done
> thread start
Done
> channel
11  <-- Channel has been reverted
Done
```

I found that the problem is here:
https://github.com/openthread/openthread/blob/master/src/core/meshcop/dataset_manager.cpp#L191

`mLocal `is actually never invalidated due to result of `Compare `method. If this result is omitted, then channel does not revert to one set in non-volatile memory.

@jwhui Should we just add a new parameter or method that will not perform this comparison or do you see more sophisticated solution?

jwhui, 2017-04-21:
Yes, good catch.  I actually think we can remove the `Compare` method from the `if-statement`.  Thoughts?

/cc @chshu @librasungirl

LuDuda, 2017-04-21:
Great! If there will be no objections i am willing to send this simple fix.

jwhui, 2017-04-21:
Go ahead and submit a PR, others can still object in the mean time :)

jwhui, 2017-04-22:
Resolved by #1639.

closed 2017-04-22 by jwhui