openthread/openthread #1662: UDP port re-use for CoAP application client and server

GiedriusM, 2017-04-26:
Real-life issue: application, which hosts CoAP application server, sends a registration (or update) to CoAP resource directory (https://tools.ietf.org/html/draft-ietf-core-resource-directory-10) enabled service with success, however once the service tries to contact the application server, it uses the information from registration (update) including the client UDP port, which in this case is ephemeral and not the same as the application servers port.

Solution required: ability to send requests from CoAP server or at least the same (re-used) UDP source port.

Currently for testing purposes I've forced client port to be the same as application server's and in client's ProcessReceivedMessage I've moved the destructive `aMessage.MoveOffset(responseHeader.GetLength());` to be executed only when the message is really processed so that the server's ProcessReceivedMessage could handle the remaining messages. This does work, but is more like a workaround.
Another solution would be to merge client and server, but this requires lots of changes and as such would most likely introduce a bunch of other issues.

What do you think would be the best way in this situation?

jwhui, 2017-04-26:
Can you point to the text in [draft-ietf-core-resource-directory](https://tools.ietf.org/html/draft-ietf-core-resource-directory-10) that indicates the using the UDP Header's Source Port of the CoAP Request?

GiedriusM, 2017-04-26:
Section 5.3, page 14:
```In the absence of this parameter the scheme of the protocol, source address and source port of the register request are assumed.```.

Another example could be [LwM2M 1.0 specification](http://www.openmobilealliance.org/release/LightweightM2M/V1_0-20170208-A/OMA-TS-LightweightM2M-V1_0-20170208-A.pdf), section 5.3.1, page 27: 
```Upon receiving a “Register” operation from the LwM2M Client, the LwM2M Server records the connection information of the registration message (e.g., source IP address and port or MSISDN) and uses this information for all future interactions with that LwM2M Client.```

jwhui, 2017-04-26:
> In the absence of this parameter the scheme of the protocol, source address and source port of the register request are assumed.

I noted this text as well.  However, [draft-ietf-core-resource-directory](https://tools.ietf.org/html/draft-ietf-core-resource-directory-10) does not specify how the source port is "assumed".

> Upon receiving a “Register” operation from the LwM2M Client, the LwM2M Server records the connection information of the registration message (e.g., source IP address and port or MSISDN) and uses this information for all future interactions with that LwM2M Client.

As noted at the end, this use of source address and port relates to future interactions with the LwM2M Client.  The text you copied here does not state any relationship between the Client and the service that the Client is registering/updating.

GiedriusM, 2017-04-26:
I am confused as to why this does not answer your question.

> I noted this text as well. However, draft-ietf-core-resource-directory does not specify how the source port is "assumed".

"How" is implementation specific and not related to the issue. Related part is that it is assumed (in case "con" is not present).

> As noted at the end, this use of source address and port relates to future interactions with the LwM2M Client. The text you copied here does not state any relationship between the Client and the service that the Client is registering/updating.

That is the problem. The registration itself is successful, but later on, when LwM2M server wants to contact LwM2M client (e.g. read resource value), it sends a request to LwM2M Client, using the connection information that was stored from registration request. In other words LwM2M server expects that the client is reachable back by the same address/port the registration came in. The transport layer is not split into two distinct parts (client and server), but is seen as a single endpoint having an address and port, which is used for bidirectional CoAP request communication. LwM2M Client/Server describe service functionality and should not be confused with CoAP client/server.

jwhui, 2017-04-26:
After rereading the text, I agree.  I guess I'm used to more traditional protocol interactions that do not have the client and server services bind to the same port.

What about simply putting a guard at the top of the respective `ProcessReceivedMessage()` methods such that:
1) `Client::ProcessReceivedMessage()` ignores all received CoAP Request messages and,
2) `Server::ProcessReceivedMessage()` ignores all received CoAP Response messages.

Thoughts?

GiedriusM, 2017-04-26:
That would probably be easiest way to keep client/server separation if there aren't any weird special case messages, that could be processed by both parties. I'll have to give it more though and will try to test it.

With this solution there should also be otCoapClientSetPort, which would set the port. However there is only one instance of CoAP client in OpenThread and both application and MLE uses it. Could there be any issues with MLE or other parts of OpenThread when CoAP requests are sent using some other source port instead of ephemeral one?  So far it seems to work on a simple setup, but I'm not very familiar with OpenThread core specifics.

jwhui, 2017-04-27:
Disregard my previous suggestion to simply ignore messages, it breaks CoAP Reset message generation in the general case.  I have some additional thoughts and will post them here in a bit.

jwhui, 2017-04-27:
One possible option: introduce a `CoapSocket` object, that serves as the intermediary between a UDP socket (or DTLS session) and one or both of `CoapClient` and `CoapServer`.  `CoapClient` and `CoapServer` would be modified to bind to `CoapSocket` rather than directly to a UDP socket.  The `CoapSocket` would be responsible for sending a CoAP Reset message if neither the Client nor Server instances could handle the CoAP message.

FWIW, it seems that others CoAP implementers have [suffered](https://github.com/mcollina/node-coap/issues/39) from needing to support the CoAP client and server binding to the same port.

@GiedriusM @rlubos - thoughts?

GiedriusM, 2017-04-27:
> Disregard my previous suggestion to simply ignore messages, it breaks CoAP Reset message generation in the general case. I have some additional thoughts and will post them here in a bit.

Is it really an issue? Like you've said, lets assume:
1. `Client::ProcessReceivedMessage()` - ignores CON/NON request messages, i.e. processes CON/NON with response code, ACK and RST messages.
2. `Server::ProcessReceivedMessage()` - ignores CON/NON response messages, i.e. processes CON/NON with request code, ACK and RST messages.

If either server or client receives unknown ACK/RST message, they simply ignore it.
If CON/NON could not be handled then at most only one RST would be sent, because either a client would generate a RST if `msg.code >= kCoapResponseCodeMin` OR a server if `msg.code < kCoapResponseCodeMin`.
Or am I missing something?

P.S. If I'm correct current Coap::Server does not generate a RST message in any case.

P.P.S. Is there really a case where a server MUST generate a RST? `A Reset message indicates that a specific message (Confirmable or Non-confirmable) was received, but some context is missing to properly process it.` - I understand that client could have no "context" when it receives an unknown response, but what could be the "context" for the server? All requests come "out of the blue" and could be responded with 404 or a generic 500/501 response.

> FWIW, it seems that others CoAP implementers have suffered from needing to support the CoAP client and server binding to the same port.

Unrelated to current issue, but [this pull](https://github.com/mcollina/node-coap/pull/131) resolved the issue.

rlubos, 2017-04-27:
Well, probably the architecure of our CoAP could be revised a little bit. Maybe the CoapBase (or it could become simpy CoAP) class, which currently is a common codebase for both client and server, could receive more responsiblity. It could receive all CoAP traffic, handle retransmission and deduplication, and dispatch requests/responses between client/server which in this case would be members of the CoapBase. In this case both entities would operate on a single socket or DTLS session.

bukepo, 2017-04-27:
I agree that we could revise the architecture of CoAP. According to the RFC, both the client and the server can send confirmable messages(separate response), as a result, both side should have the retransmission mechanism.

jwhui, 2017-04-27:
> Is it really an issue? Like you've said, lets assume:
> 1. `Client::ProcessReceivedMessage()` - ignores CON/NON request messages, i.e. processes CON/NON with response code, ACK and RST messages.
> 2. [...]

This logic assumes that a given UDP port implements both CoAP Client and Server functionality.  I think it is important to support the following scenarios:
1. Client only
2. Server only
3. Client and Server

It has become clear to me that there needs to be some shared message processing between the Client and Server components to support scenario 3 above.

> Unrelated to current issue, but [this pull](https://github.com/mcollina/node-coap/pull/131) resolved the issue.

My point here was that CoAP surfaces a scenario (Client and Server bound to same port) that is not common practice in more traditional web technologies.

GiedriusM, 2017-04-27:
> This logic assumes that a given UDP port implements both CoAP Client and Server functionality

And it doesn't? As far as I understand both CoAP client and server could bind their UDP sockets to the same port. It would be two separate sockets, but an incoming message would reach both of them and therefore the server and the client, where a mentioned filtering would be done.

jwhui, 2017-04-27:
> And it doesn't? As far as I understand both CoAP client and server could bind their UDP sockets to the same port. It would be two separate sockets, but an incoming message would reach both of them and therefore the server and the client, where a mentioned filtering would be done.

In the case that a CoAP Client is bound to a given UDP port, but *not* a CoAP Server, how do CoAP Reset messages get generated in response to receiving CoAP Response messages?  Having the CoAP Client simply ignore CoAP Response messages does not address this scenario.

GiedriusM, 2017-04-27:
> In the case that a CoAP Client is bound to a given UDP port, but not a CoAP Server, how do CoAP Reset messages get generated in response to receiving CoAP Request messages? Having the CoAP Client simply ignore CoAP Request messages does not address this scenario.

Aha, I was missing the `not a CoAP Server` parth. How about adding a single boolean flag mStandalone/mIgnoreRequests to the client and setting it when appropriate? Together with a filtering it could be simple solution. Be it a temporary one, until client and server cores are combined?

jwhui, 2017-04-28:
As a temporary solution, I am fine with having the Client and Server components simply ignoring Request and Response messages, respectively - even if that means not sending Reset messages in cases when we should be.  I can see how adding an `mIgnoreRequests` flag would provide a workaround, but I would like to avoid projecting temporary workarounds through the OpenThread API.

Thoughts?

GiedriusM, 2017-04-28:
> I would like to avoid projecting temporary workarounds through the OpenThread API.

I agree. However at the moment I can't think of any good long-term solutions. Maybe it would be helpful to state the reasons (functionality) why there MUST be a separate CoAP client/server instance. From that it could be worked out what needs to be implemented in CoAP Base/Endpoint and what in Client/Server.

GiedriusM, 2017-05-02:
Few additional current CoAP engine shortcomings that I came across:
1) No default request handler - it allows more efficient handling (and smaller code) when CoAP resources are in tree-like structure with many nodes, which require common processing.
2) No response "response" handler - this is needed in CoAP [observe](https://tools.ietf.org/html/draft-ietf-core-observe-14#section-3.6) to detect cancelled observation.

jwhui, 2017-05-03:
The idea behind separating the Client and Server functionality is to only use/allocate resources when they are actually used.  For example, if a device only needs to implement Client functionality, it does not need to allocate resources needed to support Server functionality.  The current split also makes it easy to have the Client and Server bind to different ports.  However, as you noted, it does not support binding both a Client and Server to the same port.

jwhui, 2017-05-03:
Thanks for pointing out additional feature requests.  Can you submit separate issues for these so that we can track them separate?  Would you also be willing to contribute some of your thoughts around the design of those additional features (e.g. APIs, where code should be modified, etc.)?  Even better, would you be willing to submit a PR with these additional features?

GiedriusM, 2017-05-03:
I already have these and a couple of other changes in my development repo and will probably PR them during a coffee break tomorrow.

As for the main issue I probably won't be able to PR any reasonable code anytime soon. I've had an idea about a single CoAP Endpoint to which sockets could be registered and then coap messages would be sent and received based on messageInfo, but then I remembered DTLS client/server. Not sure how to handle that nicely and bug-prone.

jwhui, 2017-05-04:
@GiedriusM, thanks and look forward to your PRs.

I'm willing to take a look into expanding the responsibility of `CoapBase` as @rlubos suggested above.

jwhui, 2017-05-09:
@GiedriusM, I just submitted #1745, which I think you might appreciate :) . Please review.

GiedriusM, 2017-05-09:
@jwhui, great job, I'm looking forward to merging this into my repo! I've skimmed through the changes and it seems good. My plan is to test the changes once it gets merged into the master, but if you want a pre-test, with some delay I could do that too.

P. S. What about #1717, #1718 and #1719? Are you going to append them or want me to resolve the conflicts once #1745 is merged?

jwhui, 2017-05-09:
I don't have a strong preference either way on #1718 and #1719.  I have some reservation with #1717 since that makes changes to the API that will be removed by #1745.  Do you have a strong preference?

GiedriusM, 2017-05-09:
Well, #1717 was supposed to be a temporary workaround for this issue, but #1745 should provide the proper solution, so I guess it's redundant and can be closed.
I will definitely need the other two for my application, so I might as well submit them to OpenThread.

PR #1745 (2017-05-12): Combine CoAP Client and Server into a single CoAP agent.
Resolves #1662.

This also has the nice side effect of reducing both code and RAM requirements.

Using arm-none-eabi-gcc 5.4.1 20160919.

Before:
```
$ arm-none-eabi-size output/cc2538/bin/ot-cli-ftd
   text	   data	    bss	    dec	    hex	filename
 214056	   4588	  27880	 246524	  3c2fc	output/cc2538/bin/ot-cli-ftd
```

After:
```
$ arm-none-eabi-size output/cc2538/bin/ot-cli-ftd
   text	   data	    bss	    dec	    hex	filename
 213504	   4588	  27680	 245772	  3c00c	output/cc2538/bin/ot-cli-ftd
```

text: -1002 bytes
bss: -200 bytes
closed 2017-05-12 by jwhui