openthread/openthread #1663: DHCP6 Client should be optional

mstamer, 2017-04-26:
With the recent update to the build environment, the dhcp6_client.cpp file missed its enable macro check.
The macro check and matching "endif" should be added to the file.

```
#if OPENTHREAD_ENABLE_DHCP6_CLIENT
...
#endif
```

DuaneEllis-TI, 2017-04-26:
I think you will need to include openthread-config.h  (via the if/else/endif scheme used in other files)
otherwise this pre-processor define will default to ZERO and this will be disabled.

mstamer, 2017-04-27:
@DuaneEllis-TI @jwhui 
Hello, for this fix, do we just want to blankly add the if/else/endif scheme to all files that use some OPENTHRED_ENABLE macro?  I did notice that some headers also include this scheme, such as thread_netif.hpp, which is why the dhcp6_client.cpp will work since it includes this header...  And if we include this scheme in all files that use OPENTHREAD_ENABLE macros, I suspect we will want some macros to prevent multiple inclusions of this openthred-config.h header...

DuaneEllis-TI, 2017-04-27:
1) Multiple inclusion of 'openthread-config.h - technically not a problem, and does not have anything other then pound-defines. if it was defining TYPES - then yea, you would get errors.

2) All internal thread source files need to include 'openthread-config.h' - PERIOD, and it needs to be the first file that is included.

3) This is a NIT - I don't like the if/else/endif - that occurs in **EVERY** source file, i would like to see this moved to some header file instead, so that there is a *SINGLE* place the if/else/endif occurs rather then every source file.

-Duane.

jwhui, 2017-04-28:
I agree with @DuaneEllis-TI's points above.

mstamer, 2017-05-08:
General question about the build environment...  I have the code done, but it fails when building/testing step:
1.9  Compiler: clang BUILD_TARGET="posix-distcheck" VERBOSE=1

Failure is due to "cannot find file".  
Specific errors are:

../../../src/core/api/instance_api.cpp:36:39: fatal error: openthread-enable-defines.h: No such file or directory
 #include <openthread-enable-defines.h>

My change is introducing a new file to include instead of the if/else/endif in every file.  New file will contain the if/else/endif

All other tests and builds can find the file without issue...  It is include as:

`#include <openthread-enable-defines.h>`

Any idea what make file I might need to update to get the correct the "-I" included in the build?  (or other way to fix?)

Still digging, but any pointer will be appreciated.

Thanks.

mstamer, 2017-05-08:
Forgot to mention, I placed the new file in:
openthread/include/

DuaneEllis-TI, 2017-05-09:
With automake - the compile commands are "quite" and not noisy.

If you type:  "V=1 make", or   "make V=1" - instead of saying:  "CC  somefile.c"
the makefile outputs the complete command line.

You could - for test purposes, tweak the travis scripts or the makefile that it invokes to add "V=1" and get a more detail from the compile.

Have you tried:  ```#include "filename"``` instead?  I wonder if it is the <> vrs "" rules in clang vrs gcc?

jwhui, 2017-05-09:
Should the new include file be placed in `src/core`?  Does it need to be a public include file?

jwhui, 2017-06-01:
@mstamer, can you provide any updates on this issue?

mstamer, 2017-06-01:
Sorry for the delay.  I have the code done and figured out the issues.  Issue was in the distcheck, it actually copies and zips all files,  then unzips in different location to build that.  So, main issue was that I needed to include this new header file in one of the make files...  That solved the problem...  Since it has be quite a while, I plan to do a re-pull of the repo, and then re-apply my changes to that...  I'll try to get a pull request in by early next week.

jwhui, 2017-06-12:
Resolved by #1877.

closed 2017-06-12 by jwhui