openthread/openthread #1681: Configure error `link tests are not allowed after AC_NO_EXECUTABLES`

abtink, 2017-04-28:
Moving to OpenThread master, we are seeing build failures triggered during `configure`:
```
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking for strlcpy... (cached) no
checking for strlcat... (cached) no
checking for strnlen... configure: error: link tests are not allowed after AC_NO_EXECUTABLES
make[4]: *** [config.status] Error 1
make[3]: *** [build] Error 2
make[2]: *** [tps/openthread/Makefile] Error 2
```

This seems to happen after the changes in this commit https://github.com/openthread/openthread/commit/d65418c12fbf820dd1ade22087541bff691a8f96.

@darconeous, @DuaneEllis-TI , any ideas?

DuaneEllis-TI, 2017-04-28:
Why is this not being caught by travis checks?

What target CPU are you using here?  What compiler version?
can you provide a more complete config.log to examine?

DuaneEllis-TI, 2017-04-28:
Also - does this help resolve the issue: https://github.com/openthread/openthread/pull/1677

abtink, 2017-04-28:
Compiler is : `arm-none-eabi-gcc (v.4,9).`
```
gcc version 4.9.3 20141119 (release) [ARM/embedded-4_9-branch revision 218278] (GNU Tools for ARM Embedded Process
```
Let me check with https://github.com/openthread/openthread/pull/1677....

abtink, 2017-04-28:
I still see this with the https://github.com/openthread/openthread/pull/1677 changes included.

DuaneEllis-TI, 2017-04-28:
so - i just pulled head, and using (the same compiler as you)
```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20141119 (release) [ARM/embedded-4_9-branch revision 218278]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

./bootstrap 
make -f examples/Makefile-cc2538
```
Works just fine.
What else can you tell me?

darconeous, 2017-04-28:
This isn't an OpenThread bug. We can close this issue and discuss internally.

abtink, 2017-04-28:
Thanks RQ . Also thanks @DuaneEllis-TI for your quick help with this.
Seems like it's related to an internal makefile  issue that we need to figure out.
Closing this issue.

closed 2017-04-28 by abtink

darconeous, 2017-04-28:
See https://github.com/openthread/openthread/commit/812cc838ebe8bec59831273d5e296f32ade0ffa4#commitcomment-21953634