openthread/openthread #1700: otThreadSetMasterkey

thcu-gp, 2017-05-02:
The api call is given by:
```
ThreadError otThreadSetMasterKey(otInstance *aInstance, const uint8_t *aKey, uint8_t aKeyLength)
```
If this is called with a masterkey that is not 16 bytes in length, e.g.
```
uint8_t key = {0x5E, 0xA1, 0xED};
otThreadSetMasterKey(aInstance, const &key, 3);
```
The 16 byte masterkey buffer will look like
```
0x5E, 0xA1, 0xED, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 0x99, 0xAA, 0xBB, 0xCC, 0xDD, 0xEE, 0xFF
```
because of the default masterkey that was used before. (One could consider clearing the buffer first)

The above results in the following
1. The actual masterkey used is: `0x5E, 0xA1, 0xED, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00`
2. Getting the masterkey results in `0x5E, 0xA1, 0xED, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 0x99, 0xAA, 0xBB, 0xCC, 0xDD, 0xEE, 0xFF` with length 16

If another thread devices joins, it will get the entire 16 byte buffer as masterkey. Since Thread does not provide a mechanic to communicate the masterkey's length to the new device, the joiner router and the joiner will have a different master key.

There are two issues:
1. The length field does not seem to be taken into account consistently (code problem)
2. The length field cannot be communicated to a new joiner device (api flaw)

jwhui, 2017-05-02:
@thcu-gp, thanks for raising this issue.

The Thread Specification does not make a clear statement on possible lengths for the Network Master Key (or _thrMasterKey_).  That said, there are a number of areas which indicate that the length must be 16 bytes:

* Section 7.1.4 Test Vectors all have a 16-byte _thrMasterKey_.
* Section 8.10.1.6 indicates that the Network Master Key TLV Length is always set to 16.

What if we changed the API to always require passing a 16-byte value?

thcu-gp, 2017-05-03:
I would suggest removing the length field entirely and always use the 16 byte masterkey.
One could keep the length field in the api, but then it should be made clear that `16 - length` `0x00` bytes are added to the masterkey (and still use the full 16 bytes internally)

GiedriusM, 2017-05-04:
How about reusing the otMasterKey type and, as such, forcing 16 byte length? From my experience it's often better to restrict the user to a certain requirement than to perform "under the hood" operation if the input is non conforming.

jwhui, 2017-05-04:
@GiedriusM, exactly what I was thinking as well, and also satisfies @thcu-gp's suggestion.

jwhui, 2017-05-04:
Submitted #1724.  Please review.

PR #1724 (2017-05-05): Require Thread Master Key to be 16 bytes.
- Update code to use `otMasterKey` type.

Resolves #1700.
closed 2017-05-05 by jwhui