openthread/openthread #1701: use of the channelmask is unintuitive and misses a check

thcu-gp, 2017-05-02:
1. 'all channels' in OpenThread is specified by `0xffffffff`, which well exceeds the allowed channels in MAC 802154. 
2. the 'all channels' mask from the MAC 802154 spec uses `0x7fff800` to specify channels 11 to 26. To only uses the channels 11 to 26 in OpenThread, one has to specify `0x0000ffff` as a mask (unintuitive)
3. If one specifies e.g. the channelmask `0x000f0000` (channels 27 to 31). The check in [mesh_forwarder.cpp](https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder.cpp) on line 1588, is only triggered after the radio tried to send packets on channel 27 (a check is missing)

jwhui, 2017-05-02:
@thcu-gp, thanks for raising these issues.  Responses below:

> 1. 'all channels' in OpenThread is specified by `0xffffffff`, which well exceeds the allowed channels in MAC 802154. 

Strictly speaking, `0xffffffff` well exceeds the channels for the **2.4 GHz band** in IEEE 802.15.4-2006.  Thread has provisions for supporting other IEEE 802.15.4 PHYs (i.e. Channel Pages) which may have different channel ranges.  Rather than having separate "all channels" define for each Channel Page, we simplify on one.

> 2. the 'all channels' mask from the MAC 802154 spec uses `0x7fff800` to specify channels 11 to 26. To only uses the channels 11 to 26 in OpenThread, one has to specify `0x0000ffff` as a mask (unintuitive)

The `OT_CHANNEL_*_MASK` [definitions](https://github.com/openthread/openthread/blob/master/include/openthread/types.h#L383) are consistent with `0x7fff800` to specify channels 11 through 26.

That said, there is a bug in `MeshForwarder` that interprets the channel mask incorrectly.  Will submit a PR soon to fix this.

> 3. If one specifies e.g. the channelmask `0x000f0000` (channels 27 to 31). The check in [mesh_forwarder.cpp](https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder.cpp) [...]

Agree that this is an issue.  Will submit a PR to address this issue.

Thanks!

jwhui, 2017-05-02:
Submitted #1702.  Please review.

PR #1702 (2017-05-03): Fix channel mask in MLE Discover process.
- Fix channel mask interpretation (bit 0 is channel 0).
- Do not send Discover Request on invalid channels.

Fixes #1701.
closed 2017-05-03 by jwhui