openthread/openthread #1709: Noise floor detection

chshu, 2017-05-03:
As defined in spec 5.9.4, some Link Metics like Link Margin/Quality/Cost are all calculated based on the received signal strength (RSS) and local noise floor.

But in current OT, the noise floor detection is not implemented yet, and the kDefaultNoiseFloor (-100 dbm) is used by default for the Link Metrics calculation.

I have some questions:
- Do you think it is necessary to implement noise floor detection feature?
- And if so, what's the suggest way to implement Noise floor detection feature?

jwhui, 2017-05-03:
We thought about this in the past and decided not to compute the noise floor based on sampling the RSSI.  Some of the challenges that came out:

1. There is little guidance on how to compute the noise floor based on samples.  For example, how often should samples be performed?  What is the duration of the averaging window?  If something like exponentially-weighted moving average is used, what are the coefficients?  The specific implementation necessarily needs to find a happy medium on being responsive, but not overly reactive.
2. Sampling the noise floor on a sleepy device is energy intensive and may not provide significant added value.

For this reason, we decided to take the initial path where the Link Margin is computed against the device's receive sensitivity, which is hardware-specific and provides a lower bound on the noise floor.  It is also the same reason why the Thread Specification chose the term "Link Margin" rather than "Signal-to-Noise Ratio".  Having said that, I agree we need to enhance the platform code to provide a receive sensitivity value.

Happy to entertain other proposals to enhance the link metric computation.

chshu, 2017-05-04:
@jwhui, Thanks a lot for the detailed clarification.

Given that there are many challenges to implement noise floor detection, I also prefer to use hardware-specific sensitivity value as the basic noise floor for link metric computation.

So I propose the following changes:
- Add `#define OPENTHREAD_CONFIG_RADIO_RECEIVE_SENSITIVITY -100` configuration
(platforms could define their own values separately, here I assume the sensitivity should be same among hardwares under the same platform, and does not change during runtime);
- Change `kDefaultNoiseFloor` to `kNoiseFloor` and set it to the receive sensitivity value;
- Simplify noise floor related implementation since we don't need it anymore.

@jwhui, @abtink, Please let me know if you have any suggestions, thanks.

abtink, 2017-05-04:
@chshu simplifying and removing the noise floor sounds great. Thanks for initiating this.

I think the radio sensitivity may not be well-suited as an OpenThread configuration.  OT config option settings are mainly used to change some behavior/parameter within OT itself. Radio sensitivity should really be provided by the radio driver/platform which is closer to hardware.

My suggestion is that we instead add a new API like `otPlatRadioGetReceiveSensetivity()` in `radio.h` which the platform radio code defines.  This way it is tied to all other radio functions and will be added  by the platform developer enabling OT on a new radio.

Thoughts?

chshu, 2017-05-04:
@abtink, It does make sense to add a new radio API to get sensitivity value, which makes it more flexible for platforms to determine their own sensitivity value.

I will push a PR to address it, thanks for your suggestion.

jwhui, 2017-05-05:
Resolved by #1715.

closed 2017-05-05 by jwhui