openthread/openthread #1710: joiner does not reset Crc16 calculations after joining attempt

thcu-gp, 2017-05-03:
When the commissioner calculates the steeringdata, `Crc16::mCrc` is always `0` at the start of each calculation.
When the joiner calculates the steeringdata, `Crc16::mCrc` is only `0` at boot or after a reset, but not at each join attempt. Therefore if a joiner tries to join a second time (the first attempt might have failed), the steeringdata are calculated incorrectly and the joining fails.

`src/core/meshcop/joiner.cpp` on the lines `107-111` should start from `Crc16::mCrc=0` instead of from a previously calculated version

chshu, 2017-05-03:
I think the issue will be fixed by #1706.

jwhui, 2017-05-03:
@thcu-gp, does #1706 address your issue?

thcu-gp, 2017-05-03:
yes, the fix in #1702 should do the trick

jwhui, 2017-05-04:
Resolved by #1706.

closed 2017-05-04 by jwhui