openthread/openthread #1713: Patching mbedtls - vrs IDEs

DuaneEllis-TI, 2017-05-03:
There's an IDE based issue with patching mbedtls that IDEs cannot easily handle.

In the ./configure world - (typically linux, but also cygwin, or msys) one often installs the package called: "build-essentials" and/or other sw-development packages or bundles. In the end, the "patch" command is present. Life is good, and simple.... and joyful.

IDEs do allow for a a pre/post build command. But the trouble is the way the are implemented. What they  typically have is one single dialog box you an enter your commands, typically 1 command. Some but not all support 1 command per line.  This feature is often implemented via the standard linux C library function  system()  [see: man 3 system]

If the command is a script... and often it is a script. Creating a platform independent script is not possible, IDEs do not have a script language, nor do they have a "dos-command" vrs "linux" vrs "cygwin" type dialog box.  Macro names in these commands vary greatly, ie: IAR uses $VAR$ with two dollar signs, Others use the posix shell variables, and windows batch files use %NAME%

We cannot demand/require that a WINDOWS user has Python installed and in their $PATH$ - as often developers have multiple versions of python,  (I have 5 versions, it is painfully complex and too far off topic to describe the reasons here)

Putting tools like CYGWIN or MSYS in their default path will often cause countless other problems, as those tools are very pseudo-environment specific.

While we can launch and IDE from a batch file - IDEs are normally not launched from a batch file (that script would for example modify the $PATH variable, or set other variables). The customer (or in this case: our victim) normally double-clicks an ICON and expects it to work.  

What about "git apply patch"?  While some IDEs include builtin support for GIT ... in some cases it a exotic version of git... (eclipse: jgit) which is not available via the command line.

It gets wierder, and complicated supporting windows users:  They have  Eclipse GIT (with its own settings), there is "git-bash" - with its msys based settings,  and Tortoise GIT, downright painful.

**I see two options here:**

**Option 1:**
The best option for an IDE is to have the mbedtls package "prepatched" and ready to go.

i know this is counter to the idea of a "virgin and untouched" third-party sub directory, my counter would be:  Rename the directory:

for example:  ${openthread}/thirdparty/mbedtls 
Becomes: ${openthread}/thirdparty/mbedtls.patched

And provide a top level script in thirdparty that can re-apply the patch to a new version of the 3rd party package.

**Option 2:**
Is Do this some other way,  and I'm stuck here. 
I'm open to ideas and need some suggestions.

Gr3yh0und, 2017-05-04:
Another option would be to use TinyDTLS? I'm currently using it additionally on top of OpenThread. Maybe create a sub-layer/API between OT and SSL and then a configure switch to choose between both?

jwhui, 2017-05-04:
@DuaneEllis-TI, another option may be to include the patched file in `third_party/mbedtls/patched` while keeping `third_party/mbedtls/repo` pristine.

jwhui, 2017-05-04:
@Gr3yh0und, the [TinyDTLS site](https://projects.eclipse.org/projects/iot.tinydtls/governance) indicates:

> The cipher suites supported by tinydtls are limited to TLS_PSK_WITH_AES_128_CCM_8 and TLS_ECDHE_ECDSA_WITH_AES_128_CCM_8 that are mandatory-to-implement for CoAP.

Unfortunately, Thread 1.1 requires an EC-JPAKE cipher suite.

Gr3yh0und, 2017-05-04:
@jwhui , Ah damn. Thought that there would be a cave eat somewhere... Thanks for pointing that out.

DuaneEllis-TI, 2017-05-04:
>> @jwhui  another option may be to include the patched file in third_party/mbedtls/patched while keeping third_party/mbedtls/repo pristine.

That does not work very well, IDEs often want to compile *ALL* source code in a directory.

It sort of needs to be the *entire* directory - all or none,

If I create two libraries  MBED-not-patched (containing all source files) and  MBED-PATCHED (containing the single patched file) -  I have no means to force the single patched file  to be picked out of the library instead of the unpatched file.  

Well.. we could - force a dummy reference to the patched function in main.c, then force the patched library to be first... but that solution is very fragile.

What if we had two "library" directories.

Directory: repo/library
And: repo/library.patched

This would require additionally patch the upper level (configure, and various Makefile.am) that recurses into library... changing it to use library.patched instead.

nibanks, 2017-05-04:
VS manages just fine with its [mbedtls_k](https://github.com/openthread/openthread/blob/master/etc/visual-studio/mbedtls_k.vcxproj#L60) project file. It just explicitly declares which files to compile.

DuaneEllis-TI, 2017-05-04:
@nibanks >> VS manages just fine with its mbedtls_k project file. It just explicitly declares which files to compile.

yea, with the caviate that if you modify the source file list (add, delete, rename a file) - you must manually modify the project file.

Other IDEs just compile the directory full of files.  Eclipse is an example of this.  Two ends of the spectrum.

nibanks, 2017-05-04:
@DuaneEllis-TI,  mbedtls is fairly static as far as changes go. I don't think it is unreasonable to just rely on modifying a source list if it every actually changes.

DuaneEllis-TI, 2017-05-04:
Agreed, Mbed is exactly 99.999% static.

However - Some IDEs have lists of files, some do not and will compile all files that match "*.c" and/or "*.cpp" - there is no list.

PR #2074 (2017-08-09): [mbedtls] apply patch to checked-in source
This commit applies the patch directly to checked-in source files rather
than at build time.  This commit helps to simplify the build process.

This commit also introduces a shell script that allows anyone to reproduce
the patched mbedtls source.

Resolves #1713.
closed 2017-08-09 by jwhui