openthread/openthread #1741: travis test: posix-ncp, and 'spinel.py'

DuaneEllis-TI, 2017-05-09:
Currently, the .travis/setup.sh script

At the very top - installs (via pip) the package spinel:
For example the log reads:
```
+pip install --user pyserial
Requirement already satisfied: pyserial in /usr/lib/python2.7/dist-packages
+pip install --user git+https://github.com/openthread/pyspinel
Collecting git+https://github.com/openthread/pyspinel
  Cloning https://github.com/openthread/pyspinel to ./pip-1KZKeL-build
```

This seems a bit strange, if there is a change to spinel - you would want to test the SPINEL that is *PRESENT* in the distribution under test, not the spinel that is in the master branch.

that said, it would be a nice thing - if rather then installing spinel, and thus spinel-cli being in your "python path" - it probably should be used in a quasi 'installed location'.

To describe the problem another way - without spinel installed - the "make posix-ncp" test fails with error messages like this:
```
ERROR: test (__main__.Cert_5_1_01_RouterAttach)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/a0226692local/ti-openthread-vagrant/openthread/examples/../tests/scripts/thread-cert/Cert_5_1_01_RouterAttach.py", line 48, in setUp
    self.nodes[i] = node.Node(i)
  File "/home/a0226692local/ti-openthread-vagrant/openthread/tests/scripts/thread-cert/node.py", line 42, in __init__
    self.interface = node_cli.otCli(nodeid)
  File "/home/a0226692local/ti-openthread-vagrant/openthread/tests/scripts/thread-cert/node_cli.py", line 44, in __init__
    self.__init_ncp_sim(nodeid)
  File "/home/a0226692local/ti-openthread-vagrant/openthread/tests/scripts/thread-cert/node_cli.py", line 79, in __init_ncp_sim
    self.pexpect = pexpect.spawn(cmd, timeout=4)
  File "/usr/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 194, in __init__
    self._spawn(command, args, preexec_fn, dimensions)
  File "/usr/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 267, in _spawn
    'executable: %s.' % self.command)
ExceptionPexpect: The command was not found or was not executable: spinel-cli.py.
```

jwhui, 2017-05-09:
As a result of #961, spinel-cli is now maintained in a separate repo [openthread/pyspinel](https://github.com/openthread/pyspinel).  Do you have a proposal on the 'quasi installed' location?

DuaneEllis-TI, 2017-05-09:
Hmm - I see that this was moved, but never *removed* from openthread...

I guess there are two issues then:

a) Just how stable is the SPINEL protocol? How often do we expect new packets to be added to the protocol, and how do we coordinate updating *both* SPINEL and Openthread at the same time.  I am concerned about the corner case of requiring an update to *both* pyspinel and openthread in order to get an acceptance test to pass.

b) The other issue is a test/build environment issue and the way the environment works.

When using Travis - it always gives you a "clean machine" and an "isolated machine".

In contrast, other test platforms (ie: Jenkins) generally give you a shared user - with separate build directories. 

To solve the above problem (the specific version problem) - we could move the "pip install" from the travis setup to the Makefiles that use them.

And install a specific/specified version into the ${openthread}/build directory, instead of:  ```pip install --user```, use  ```pip install --target ${openthread}/build``` then set the PYTHONPATH variable for the tests that require the package. All of that could be done in the test specific top level Makefile

Another problem that a semi-shared environment like Jenkins has is this:  Often Jenkins kicks off multiple jobs - and that means you can easily have two tests running at the same time.  If each job installs its own tools, and they install to a SHARED location  JOB (A) may reinstall the shared tool that JOB(B) is currently using... 

Again - I think installing into the local build directory solves that.

jwhui, 2017-05-09:
The intent of #961 was to remove from this repository once the new repository was up.  No sense to maintain multiple copies of the same code.  #1742 completes the migration process.

FWIW, looking at the history, it seems that spinel-cli has mostly been a second class citizen.  Historically, changes/additions to the Spinel specification were immediately reflected in [src/ncp](https://github.com/openthread/openthread/tree/master/src/ncp) and [wpantund](https://github.com/openthread/wpantund/).  spinel-cli significantly lags in implementing the full Spinel specification.  Also, I'm tempted to question the value of spinel-cli's function as a part of Continuous Integration.  #1748 was a fix to a breaking change in #1433 nearly two weeks prior that went fully undetected by Travis.

Given the differences between Travis and Jenkins, it may be that the Travis scripts are not applicable to Jenkins environments.

All that said, I'm fine with installing into a local build directory.

wbober, 2017-05-18:
@DuaneEllis-TI

I agree that using pyspinel master branch might not be the best idea. However, this could be solved by referencing a specific commit, branch, or tag in the setup.sh, i.e.:

git+https://github.com/openthread/pyspinel@COMMIT-SHA

Regarding the installation issue I would suggest to use python virtualenv. As a part of the job a CI server would have to create a virtual env and activate it. This way all packages would be installed in that env. 

What do you think?

DuaneEllis-TI, 2017-05-18:
The PY virtenv - if it is local to the build would be a good solution.

jwhui, 2017-10-26:
@DuaneEllis-TI, is this still an open issue for you?

closed 2017-10-26 by DuaneEllis-TI