openthread/openthread #1780: otDatasetSendMgmtActiveGet

thcu-gp, 2017-05-17:
Shouldn't otDatasetSendMgmtActiveGet be available for mtd devices as well?
Any device should be able to request the Active dataset from another device.

chshu, 2017-05-17:
@thcu-gp, As far as I see, only MGMT_ACTIVE_GET.rsp is mandatory for all devices, as described in section 8.7.4.4:
> The Leader and other Thread Devices MUST send a MGMT_ACTIVE_GET.rsp in response to receiving a MGMT_ACTIVE_GET.req message.

This feature is supported by MTD with PR #1360. 

But regarding to MGMT_ACTIVE_GET.req, I didn't notice that it is mandatory for all Thread devices, could you please help to refer some words from spec? From section 8.7.4.3, it only says:
> The Commissioner and Thread Devices may retrieve Active Operational Dataset values from the Leader by sending a MGMT_ACTIVE_GET.req.

It is may, so I think it is not mandatory for all Thread devices.

thcu-gp, 2017-05-17:
Let me quote the spec on `MGMT_ACTIVE_SET.req` because it has similar phrasing with respect to the 'may'
p5-56 L32
> The Commissioner and Thread devices may provide a new Active Operational Dataset to the Leader by sending a MGMT_ACTIVE_SET.req.

This part has, similarly to MGMT_ACTIVE_GET.req, a 'may' in the text (note this is not a 'MAY'). However, in section 8.7.4.2 the spec says:
> A Thread device that has a newer Active Operational Dataset, as determined by the Active Timestamp included in Thread Network Data, MUST provide its newer Active Operational Dataset to the Leader by sending a MGMT_ACTIVE_SET.req to the Leader. 

This text has a clear 'MUST'. In my opinion the 'may' more means _when it is required_ rather than _it is up to the device to decide if it sends or doesn't send this message_.

With respect to `MGMT_ACTIVE_SET.req` p8-20 L37 of the spec says:
> If the Parent advertises an Active Timestamp that is different that the Thread device’s local Active Timestamp, the Thread device MUST retrieve and install the Active Operational Dataset from its Parent.

How else would it retrieve the dataset than by using `MGMT_ACTIVE_SET.req`

That being said. It is not because the functionality exists it has to be made available through the API. 
This brings me back to my first post. In my opinion the `otSendMgmtActiveGet` is something that should be available through the API for all devices, not excluding MTDs. On the other hand I don't think `otSendMgmtActiveSet` should be available through the API, because a stack event will trigger a `MGMT_ACTIVE_SET.req`, while any  other (user) attempt of changing the active dataset should pass through the commissioner.

chshu, 2017-05-17:
@thcu-gp, Thread device could use MLE Data Request/Response to retrieve Datasets from its parent (OT are using this way), that's why I think `MGMT_ACTIVE_GET.req` could be optional.

I agree with you that the `may` is not `MAY`, and I'm open to support MGMT_ACTIVE_GET.req in MTD, my only concern is that it will increase some code size to MTD build while the feature may be not actually used.

thcu-gp, 2017-05-18:
```
ThreadError otDatasetSendMgmtActiveGet(otInstance *aInstance, const uint8_t *aTlvTypes, uint8_t aLength, const otIp6Address *aAddress)
{
    return aInstance->mThreadNetif.GetActiveDataset().SendGetRequest(aTlvTypes, aLength, aAddress);
}
```
The function body is already part of an mtd build. So the only increase in code size comes from adding an additional function. So, unless one wants to shave off every superfluous byte. I'm not sure if code size is the most important ground to include or not include the api call. (One could turn the observation around and claim, the code size could probably be reduced further)

Use case is a different story. At this time I cannot thing of a real life use case for this function unless one assumes the local dataset is out of date and the application cannot wait for the update of the dataset by the regular Thread mechanisms. It thus boils down to: "Do we support it because it is allowed" or "we don't support it because it is not used"

chshu, 2017-05-18:
`otDatasetSendMgmtActiveGet()` is located in dataset_ftd_api.cpp which is not part of MTD now.

The code size was an issue before, we could not fit OT into some real products, that's the reason we need a MTD build, and want to keep MTD build smaller.

To support this functionality, we need to add at least following functions into MTD build (estimated hundreds bytes):
- [ProcessMgmtGetCommand()](https://github.com/openthread/openthread/blob/master/src/cli/cli_dataset.cpp#L579)
- [otDatasetSendMgmtActiveGet()](https://github.com/openthread/openthread/blob/master/src/core/api/dataset_ftd_api.cpp#L66)
- [DatasetManager::SendGetReques()](https://github.com/openthread/openthread/blob/master/src/core/meshcop/dataset_manager.cpp#L752)

I think it is easy to support this functionality once there are some real use cases or feature requests, so we could add it by then, do you agree?

thcu-gp, 2017-05-22:
I agree with your final conclusion.

closed 2018-01-31 by jwhui