openthread/openthread #1786: Enhance otPlatRadioTransmitDone() callback by passing up the ACK frame

chshu, 2017-05-18:
This is a proposal to enhance otPlatRadioTransmitDone() callback by **passing up the ACK frame** as part of it.

Current otPlatRadioTransmitDone() callback only pass up the ACK's `aFramPending` information:
```
extern void otPlatRadioTransmitDone(otInstance *aInstance, RadioPacket *aPacket, bool aFramePending,
                                    ThreadError aError);
```
The new otPlatRadioTransmitDone() callback will pass up the whole ACK frame instead:
```
extern void otPlatRadioTransmitDone(otInstance *aInstance, RadioPacket *aPacket, RadioPacket *aAckPacket,
                                    ThreadError aError);
```

The benefits of the enhancement:
- The ACK messages could also be used for link quality evaluation (short term);
- To support the Enhanced ACK which may be used in some new features (long term).

Please give comments if you have any suggestions or concerns, thanks for that.

pvanhorn, 2017-05-18:
I think my biggest concern with this is that, we currently find ourselves running out of radio buffers during periods of heavy throughput.  Using a RadioPacket to report every ACK might further tax that resource.  Understand that the packet is received via an ISR but the RadioPacket is delivered via a task thread.  This means a pool of packets is needed so that the ISR can continue to function while the thread is delayed.

chshu, 2017-05-19:
@pvanhorn, Thanks for sharing your concern.

Here we are talking about passing up ACK frame instead of a single `aFramPending` flag, there will be no changes to how radio receive packets, and no extra packets pool needed.

The radio could transmit only one packet at a time, it will not transmit the next packet until it receive the ACK or timeout. So only a single ACK packet will be used, but not a pool.

pvanhorn, 2017-05-19:
Your last point is a good one. I hadn't thought that through to completion. So we are fine as long as we later don't create a transmit queue in an effort to increase throughput.

chshu, 2017-05-27:
Filed separate issues (#1839, #1840, #1841, #1842) for the remaining work on different platforms, so I think this issue could be closed.

closed 2017-05-27 by chshu