openthread/openthread #1798: NCP-UART: assert when used with CDC

GiedriusM, 2017-05-22:
One of my devices uses USB CDC protocol as a UART channel. This means that before the VCOM is opened on the host, no data can be sent and as such otPlatUartSend returns an error, which causes an assert in `ncp_uart.cpp`, when sending an initial status update (even before wpantund sends any information).

I've changed otPlatUartSend code to always return no error and this seems to work, however I have few questions.
1) Are there any potentials problems (NCP-wise) with otPlatUartSend always returning success, even though if the actual data is not sent (at least initially)?
2) What's the purpose of sending the initial status update before any messages are received from the host? In real UART case there is no way of knowing that the host is listening.

Also this does not look like such a critical area to cause an assert. Here's few ideas how it could be changed:
1) Ignore the returned status of otPlatUartSend.
2) Add flag which enables sending messages only after information is received from host.
3) Add a forth `UartTxState` state `kSendingFrame`, which would retry the otPlatUartSend until success.

Any thoughts?

LuDuda, 2017-05-22:
We have encountered similar problem with USB CDC. Please look at #1563

GiedriusM, 2017-05-22:
Thank you, @LuDuda. That's exactly the same problem. Is it correct to say that the consensus and the proposed solution for this issue is that UART driver should never return en error, but instead delay calling otPlatUartSendDone until it is ready to send new data?

rlubos, 2017-05-23:
Exactly, this is how it works on our side. When USB CDC driver is not able to send data due to closed port, it  stores the data, sends it once the port is opened and then calls otPlatUartSendDone. No data should be lost in this particular case, as NCP shall not send any more data before otPlatUartSendDone is called.

GiedriusM, 2017-05-23:
Applied your solution. Seems to work.

closed 2017-05-23 by GiedriusM