openthread/openthread #1803: FR: Arduino support

beriberikix, 2017-05-22:
Arduino has been a long-time ask as a supported framework. Most users just want a solution that allows them to hang a radio off an Uno, form a simple mesh and write some Processing code to blink lights/control motors/etc. I propose that an initial port would effectively be an NCP connected to an Arduino via serial and Arduino-inspired interface that maps to the CLI commands. Any OT-supported platform should work with the vanilla NCP build. For the actual API there is plenty of prior art in "Arduino-esque" networking: there's an official [Xbee shield](https://www.arduino.cc/en/Main/ArduinoWirelessShield) with 802.15.4 support that "just works." There's also an official WiFi library ([docs](https://www.arduino.cc/en/Reference/WiFi), [code](https://github.com/arduino/Arduino/tree/master/libraries/WiFi)) with a richer interface. With the [porting guide](https://github.com/openthread/openthread/wiki/Porting-Guide) we should be in a better position to develop this port. Also, Arduino now supports [CI](https://github.com/arduino/arduino-builder/wiki/Doing-continuous-integration-with-arduino-builder) so we can run tests against the port as OT evolves.

FWIW, the [FRDM-KW41Z](http://www.nxp.com/products/software-and-tools/hardware-development-tools/freedom-development-boards/nxp-freedom-development-kit-for-kinetis-kw41z-31z-21z-mcus:FRDM-KW41Z) would make for an ideal reference platform since it is in the form factor of a Shield.

aeliot, 2017-05-23:
I think that the initial port, with just the NCP model, shouldn't involve too many (if any) changes to OT, it should just need an Arduino library to handle spinel, and probably a bit of management so that the user doesn't need to know too much about the inner workings of OT.

aeliot, 2017-05-23:
I spoke too soon.  I had assumed that since there were other network technologies that supported Arduino, there would be a light weight IP stack for it already.  It looks like most shields actually bring their own IP stacks, running them NCP side instead of host side.  I'm looking into if it would be worth it to port a super light weight IP stack (probably uIP since it looks like it could fit on an UNO) onto Arduino for use by OT.

jwhui, 2017-05-23:
OT brings just enough of an IPv6 stack to support Thread.  Another possibility is to extend Spinel to export another data stream that is more friendly to application developers (e.g. UDP).

aeliot, 2017-05-23:
Yea, thats what I was just thinking about.  This also could make porting to other minimal development platforms easier down the road too.

Gr3yh0und, 2017-07-26:
That would be a cool feature. It would also be nice to get support for the [FRDM-K22F](http://www.nxp.com/products/software-and-tools/hardware-development-tools/freedom-development-boards/nxp-freedom-development-platform-for-kinetis-k22-mcus:FRDM-K22F) as it's also Arduino R3 compatible and with using a NRF24L01 as IEEE 802.15.4 radio that would create an OT platform available for less than 30$.

beriberikix, 2017-07-26:
Unfortunately, AFAICT the NRF24L01(+) can function as an IEEE 802.15.4. However, the [FRDM-KW41Z](http://www.nxp.com/products/software-and-tools/hardware-development-tools/freedom-development-boards/nxp-freedom-development-kit-for-kinetis-kw41z-31z-21z-mcus:FRDM-KW41Z) has R3 headers, as well as the Nordic [nRF52840-DK](https://www.nordicsemi.com/eng/Products/nRF52840-Preview-DK). That said, there are a good number of low-cost transceivers worth investigating once Arduino support becomes available.

Humancell, 2018-12-13:
Hello,

I wanted to let you know that we are *very* close to having this complete.  Not ready to announce *yet* ... but extremely close.  We will obviously by making our repo and project completely open very soon.

My question is:  Is there an estimate on when the next "release" is coming?

We have had to make several updates to the code for compatibility, and will work towards this being an official pull request eventually.  In the mean time, we are working with the current "release" from 2018-09-26.  Depending on timing, we want to see if there is a next release that is close?

BTW ... our current work is based on the KW41Z chipset, and we are creating both Arduino Uno and Arduino Pro Mini (or Feather) form factor boards.  :-)

jwhui, 2018-12-14:
@Humancell, great to hear about your efforts! Our existing releases mirror what is provided with the Thread Test Harness. We do not currently have a newer release in the pipeline. Look forward to hearing more updates!

PTS93, 2019-01-22:
@Humancell Will that be specific to the KW41Z or is it abstracted to work with all openthread supported plattforms that also support Arduino? The nRF52840 and SAMR21 comes to mind.

CraigZych, 2019-05-05:
Is this almost done?
I'd be very interested to try this out

charly37, 2019-09-11:
What about the new NANO 33 BLE series based on nrf52840 ?

soburi, 2019-11-13:
Hi, all.

I also created experimental arduino package that integrated with openthread. (support nrf52840)

https://github.com/soburi/openthread_nrf52_arduino
Direct url for set arduino preferences .
https://soburi.github.io/openthread_nrf52_arduino/package_soburi_otnrf52_index.json

One thing I noticed on create a package.

- ISR implementation will conflicts with arduino's. one

   Arduino implementation has their own UART ISR handler.
   I think it is more good that ISR function is defined as weak symbol.

(This package is contains OpenThread sources without modification.
  I wish to use OpenThread like a library.)

hpssjellis, 2019-11-15:
What about the new NANO 33 BLE series based on nrf52840 ? I would very much like to see this series supported. Is there any way we could make a test openthread library for importing into the arduino IDE?

IncPlusPlus, 2020-06-15:
@Humancell Is this still in the works? I was really hoping I could use this on an Uno.

kasperkamperman, 2020-12-03:
Curious to hear if someone got this working on an Arduino.

andoks, 2021-03-19:
Echoing Kasper above me, I too am curios if anyone has this working on arduino.

Going back to the roots of this issue itself, would it perhaps be possible to flash an [Arduino nano 33 BLE](https://store.arduino.cc/arduino-nano-33-ble) (based on the nRF52840  chip) with a thread-enabled firmware?

hpssjellis, 2021-03-19:
@andoks I have moved on as it is much too difficult, but yes with the Particle Debugger it is fairly easy to flash nrf52840 bootloader to the Nano 33 BLE and then use nordic software. However that has several pitfalls, such as losing a few pins. See link below.

Getting openthread nicely running on the Nano 33 Ble Sense with the Arduino environment would be interesting, but I have moved on to LoRaWan networks.

My rather long discussion is here. Good luck.

https://community.platformio.org/t/like-if-you-have-an-nrf52840-nordic-usb-dongle-arduino-nano-33-ble-adafruit-sparkfun-april-makerdairy-fanstel-particle-or-any-other/11112

Wiz-IO, 2021-08-07:
Hi
you can make CC2652 Arduino ( PlatformIO ) experiments
https://github.com/Wiz-IO/wizio-cc

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui

andoks, 2023-06-28:
@jwhui: could you point me in the direction of how to use openthread on an arduino? I am unable to find it mentioned anywhere at [openthread.io](https://openthread.io/s/results?q=arduino)