openthread/openthread #1814: Invalid MPL cache entry lifetime value

LuDuda, 2017-05-23:
I've just spotted that OT sets `OPENTHREAD_CONFIG_MPL_CACHE_ENTRY_LIFETIME` to **5** seconds.

Current specification defines it to **30** minutes (according to RFC7731), but this time may be reduced to **90** seconds (due to errata [SPEC-679](https://threadgroup.atlassian.net/browse/SPEC-679))

If we adjust our cache lifetime, I believe there is a need to store our MPL Sequence value in a persistent storage, as it is done for MLE and MAC frame counters (e.g. store after each 32 MPL transactions or so..).

Do you think we should implement it, and if yes, where should we store MPL sequence value:
	- add a new key of settings to store MPL sequence only e.g. `kKeyMplInfo` inside the module
	- put MPL sequence value into `kKeyNetworkInfo`  (may be hard to implement)
	- add a new key of settings to store IP6 protocol data (currently only MPL) e.g. `kKeyIp6Info` with function ` Ip6::Store()` similar to `Mle::Store()`

jwhui, 2017-05-24:
Regarding the `SEED_SET_ENTRY_LIFETIME` value, I believe even 90 seconds is much longer than we need.  With relatively small values of `DATA_MESSAGE_IMAX` and `DATA_MESSAGE_TIMER_EXPIRATIONS` should mean that something on the order of 5 seconds should be sufficient.  Thoughts?

I'd like to avoid non-volatile storage of the MPL Sequence value, if possible.

LuDuda, 2017-05-25:
@jwhui I'm fine with the current solution.

But maybe we should propose this change to be included in the specification.

I can easily imagine situation where node with OpenThread stack may have problem to send MPL messages to nodes with other Thread stack. e.g. after reset (in worst case other nodes will not hear up to 27 MPL messages)

closed 2022-08-01 by LuDuda