openthread/openthread #1818: Router floods network with Child Update Requests after reset

wbober, 2017-05-24:
We see a suspicious behavior of a router/leader after reset. If the router had any children then, after reset, it will send a child Update Request to each of them. The issue is that it keeps doing so every second, for about 240 seconds (I guess until all children are removed from the child table). Is that an intended?

Steps to reproduce:
- start a device and wait until it promotes to a leader
- start another device and wait until it attaches as a child
- switch off the second device
- reset the first device and observe the issue

jwhui, 2017-05-24:
The mechanism you are observing is intended to implement the child restoration feature after a reset, as discussed in Thread 1.1 Section 4.7.3.  I'll review the implementation again to see if/how we can improve.

wbober, 2017-05-24:
@jwhui Yes, I'm aware of that. The issue is that the current implementation causes quite a lot of traffic (4 frames / child) every second. My understanding of Section 4.7.3 is that it's sufficient to send a single Child Update Request after a reset.

jwhui, 2017-05-24:
If the implementation is sending 4 frames/child every second, I agree that is a problem.  At the same time, I'm not sure one is that robust since messages can get lost (albeit with low probability).  I think some number between 2 and 4 is sufficient.

wbober, 2017-05-24:
From a brief code inspection and info from my colleagues it looks like there is one Child Update Request / per child / per second generated in mle_router. 4 frames per request are due to MAC retransmissions.

I think it should be sufficient to generate a single Child Update Request / per child sometime after reset and leave to MAC to deal with potential frame loss.

jwhui, 2017-05-24:
Ah, I see what is going on.  I believe the current logic was written with the SED in mind.  For SEDs, the desire was to keep trying to send Child Update Request messages for as long as the SED's timeout.  We felt this was OK given that transmissions are naturally rate limited by the SED's polling.

However, the current logic fails to change that behavior for rx-on-when-idle Children.  I'm fine with adjusting the logic such that it tries only once or twice for rx-on-when-idle Children.  However, I see value in continued attempts with SEDs.

One of the challenges with IEEE 802.15.4 is that received ACKs can lead to false positives (the ACK only includes an 8-bit DSN), especially in dense environments.  That is why retransmissions are not always sufficient.  False positives occur with low probability, but we do see them in practice.

PR #1821 (2017-05-25): Only send one Child Update Request when restoring rx-on-when-idle child.
Resolves #1818.
closed 2017-05-25 by jwhui