openthread/openthread #1828: Multi-Hop Multicast Packets via NCP

d-bahr, 2017-05-25:
I have a network with 2 routers and 1 leader. The leader is within range of router 1, and router 1 is within range of router 2, but the leader is out of range of router 2 and must go through an extra hop to get there.

When I send a multicast packet from the leader to the all-nodes link-local address "FF02::1", I would expect both routers to receive the multicast packet, but it appears that router 1 swallows the multicast and router 2 never receives the packet. I have debugged the outgoing packet to ensure the hop limit is > 1.

I am using three CC2538 processors as NCPs via wpantund. This issue is similar to #238. (maybe a regression?)

When I get a spare moment, I will try reproducing with the CLI.

jwhui, 2017-05-25:
Link-local scope in Thread is defined by receivers within direct transmission range.  If you want to reach receivers multiple hops away using multicast, you need to use Realm-Local Scope (e.g. FF03::1).

d-bahr, 2017-05-25:
Thanks, that seems to do the trick. In the CLI, pinging FF02::1 only returned a ping from router 1, but pinging FF03::1 returned a ping from both routers. I'll go ahead and close this issue.

closed 2017-05-25 by d-bahr