openthread/openthread #1869: Test harness FED 9.2.13 fails

thcu-gp, 2017-06-02:
**Observation:** The parent of the dut sends a icmp destination unreachable.

**Cause:** the ping destination is the mleid of the dut, but the mesh header final destination is the dut's parent's rloc16. Because the dut is a FED, the parent does not know the dut's mleid, therefore sending the icmp error (it can not ip forward the message).

**Down the rabbit hole:** the Final destination in the mesh header is the dut's parent's rloc16. I believe this should be the dut's rloc16. The preceding energy request is supposed to have the same source and destination as the ping, but it hasn't. On inspection, the mesh header of the energy request has as final destination the dut's rloc16. So why is the addressing of the energy request and the ping request different? I believe the answer lies in the energy response. The dut sends this response to its parent (_ip route over_). The parent then forwards (_mesh under_) it to the commissioner (*). The commissioner sees in the mesh header the dut's parent's rloc16 associated with the dut's mleid. I think the commissioner updates its eid-to-rloc cache with this (incorrect) information. Consequentially, it uses this incorrect information to send the ping request.
Attached are the wireshark and test harness logs
[logging.zip](https://github.com/openthread/openthread/files/1047563/logging.zip)

(*) I think this behaviour is incorrect: the dut should use a mesh header with O=dut's rloc16 and F=commissioner's rloc16

jwhui, 2017-06-02:
@thcu-gp, good catch!  I agree that the FED should be the originator for the mesh header.  The existing implementation only checked whether the device was in the child state.  We really should be checking whether or not the device is acting as an MED.  I have submitted #1870 to address this issue.  Please review.

librasungirl, 2017-06-04:
Validated #1870 on the mixed testbed with TH v45
[GitHub_Issue1869.zip](https://github.com/openthread/openthread/files/1050439/GitHub_Issue1869.zip)

PR #1870 (2017-06-05): Full End Devices should initiate mesh-under forwarding.
Resolves #1869.
closed 2017-06-05 by jwhui