openthread/openthread #1873: Consider making OpenThread available with the Conan C/C++ Package Manager

jhwoodyatt, 2017-06-03:
The [Conan](https://conan.io/) C/C++ Package Manager seems to be the emerging convention.

beriberikix, 2017-06-03:
Ya, Conan seems to be growing in popularity and will likely grow now that jfrog acquired them.

@jhwoodyatt are you familiar with their system and syntax at all? Adding a config to /etc seems like a easy first step.

jhwoodyatt, 2018-01-19:
My forecast is growing less optimistic...

beriberikix, 2018-01-19:
Agreed. http://pfultz2.com/blog/2017/10/27/universal-package-manager/

jwhui, 2018-01-31:
@jhwoodyatt, do you want to keep this issue open?

jhwoodyatt, 2018-02-01:
I'm not sure I like PlatformIO any better than Conan or any of the other obvious alternatives, but I'm sympathetic to at least the general view expressed in that Paul Futz blog post: that the embedded C++ community needs to settle on a decent application package manager. Either that, or my current preferred alternative: abandon C++ for a language with a community that already has one (and I can think of several good candidates).

I'll close this problem.

closed 2018-02-01 by jhwoodyatt