openthread/openthread #1902: Illegal Network Name causes crashes

DuaneEllis-TI, 2017-06-14:
This actually crashes wpantund, but has roots in the mis-behaving remote device.

Not sure where this should be filed, as it is an architectural decision.

To reproduce:

From the 'wpanctl' application, use the "scan" command to scan for networks.  The code in wpantund (dbus, specifically) assumes, rightly or wrongly, that the network name is a UTF8 string. 

the NCP transmits a beacon request.

In this scenario, the remote device answering the beacon request has a bug and is sending a bad or invalid string as the network name. That device should transmit: "OpenThread" - instead it transmits: "O\377\377\377\377hread" (Where: \377 = Octal for an 0xFF byte).  It looks like something wrote a 32bit "-1" value over the top of something it should not have.

This goes over the air with the "0xff" bytes in place of a portion of the network name, up through the NCP and finally into Wpantund - where an assertion triggers this error:
```
process 592: arguments to dbus_message_iter_append_basic() were incorrect, assertion "_dbus_check_is_valid_utf8 (*string_p)" failed in file ../../dbus/dbus-message.c line 2676.
```
of course this should be fixed in the transmitting device, but the robustness principle states either the NCP, or WPANTUND ... something, somewhere shall not die a horrible death as it is right now.

-Duane.

jwhui, 2017-06-14:
My personal preference is to have the NCP simply pass through the Network Name as it does today and have wpantund updated to handle such cases.  It gives the host more visibility into what's going on in the network.

Do you believe the "O\377\377\377\377hread" network name something caused by the OpenThread library itself?  If so, can you also provide more information to reproduce?

DuaneEllis-TI, 2017-06-14:
I'd agree - let the host see this. But that means the host, in this case, wpantund needs to firewall this type of issue. 

I'll file an issue on the wpantund side.

As to what is causing it? Don't know  yet. We  have several people experimenting with things.

closed 2017-06-14 by DuaneEllis-TI