openthread/openthread #1915: Building cc2650 examples is there a bug or have I done something wrong?

fizz-bang, 2017-06-19:
Hi, I am jumping straight in and trying to build the cc2650 example to use with the launchpad hardware.

I "think" I have everything in place as per the instructions (usually use IDE's under Windows) but when I follow the instructions using a terminal window in Ubuntu 16.04 LTS to build the example I get errors:

make -f examples/Makefile-cc2650
  BUILD    cc2650
make -j10 -C build/cc2650 --no-print-directory all
  CHECK    .local-version
make --no-print-directory all-recursive
Making all in include
make  all-recursive
Making all in openthread
Making all in platform
make[6]: Nothing to be done for 'all'.
make[6]: Nothing to be done for 'all-am'.
Making all in src
Making all in core
make[4]: Nothing to be done for 'all'.
Making all in cli
make[4]: Nothing to be done for 'all'.
Making all in ncp
make[4]: Nothing to be done for 'all'.
make[4]: Nothing to be done for 'all-am'.
Making all in third_party
Making all in mbedtls
make[4]: Nothing to be done for 'all'.
make[4]: Nothing to be done for 'all-am'.
Making all in examples
Making all in platforms
Making all in utils
make[5]: Nothing to be done for 'all'.
Making all in cc2650
  CC       libopenthread_cc2650_a-cc2650_ccfg.o
In file included from /home/gareth/openthread-master/examples/../examples/platforms/cc2650/cc2650_ccfg.c:48:0:
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:372:42: error: initializer element is not a constant expression [-Wpedantic]
 #define DEFAULT_CCFG_MODE_CONF           ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:464:5: note: in expansion of macro 'DEFAULT_CCFG_MODE_CONF'
     DEFAULT_CCFG_MODE_CONF          , // 0x50003FB4
     ^~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:372:42: note: (near initialization for '__ccfg.CCFG_MODE_CONF')
 #define DEFAULT_CCFG_MODE_CONF           ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:464:5: note: in expansion of macro 'DEFAULT_CCFG_MODE_CONF'
     DEFAULT_CCFG_MODE_CONF          , // 0x50003FB4
     ^~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:387:42: error: initializer element is not a constant expression [-Wpedantic]
 #define DEFAULT_CCFG_VOLT_LOAD_0         ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:465:5: note: in expansion of macro 'DEFAULT_CCFG_VOLT_LOAD_0'
     DEFAULT_CCFG_VOLT_LOAD_0        , // 0x50003FB8
     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:387:42: note: (near initialization for '__ccfg.CCFG_VOLT_LOAD_0')
 #define DEFAULT_CCFG_VOLT_LOAD_0         ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:465:5: note: in expansion of macro 'DEFAULT_CCFG_VOLT_LOAD_0'
     DEFAULT_CCFG_VOLT_LOAD_0        , // 0x50003FB8
     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:393:42: error: initializer element is not a constant expression [-Wpedantic]
 #define DEFAULT_CCFG_VOLT_LOAD_1         ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:466:5: note: in expansion of macro 'DEFAULT_CCFG_VOLT_LOAD_1'
     DEFAULT_CCFG_VOLT_LOAD_1        , // 0x50003FBC
     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:393:42: note: (near initialization for '__ccfg.CCFG_VOLT_LOAD_1')
 #define DEFAULT_CCFG_VOLT_LOAD_1         ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:466:5: note: in expansion of macro 'DEFAULT_CCFG_VOLT_LOAD_1'
     DEFAULT_CCFG_VOLT_LOAD_1        , // 0x50003FBC
     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:399:42: error: initializer element is not a constant expression [-Wpedantic]
 #define DEFAULT_CCFG_RTC_OFFSET          ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:467:5: note: in expansion of macro 'DEFAULT_CCFG_RTC_OFFSET'
     DEFAULT_CCFG_RTC_OFFSET         , // 0x50003FC0
     ^~~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:399:42: note: (near initialization for '__ccfg.CCFG_RTC_OFFSET')
 #define DEFAULT_CCFG_RTC_OFFSET          ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:467:5: note: in expansion of macro 'DEFAULT_CCFG_RTC_OFFSET'
     DEFAULT_CCFG_RTC_OFFSET         , // 0x50003FC0
     ^~~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:404:42: error: initializer element is not a constant expression [-Wpedantic]
 #define DEFAULT_CCFG_FREQ_OFFSET         ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:468:5: note: in expansion of macro 'DEFAULT_CCFG_FREQ_OFFSET'
     DEFAULT_CCFG_FREQ_OFFSET        , // 0x50003FC4
     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:404:42: note: (near initialization for '__ccfg.CCFG_FREQ_OFFSET')
 #define DEFAULT_CCFG_FREQ_OFFSET         ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:468:5: note: in expansion of macro 'DEFAULT_CCFG_FREQ_OFFSET'
     DEFAULT_CCFG_FREQ_OFFSET        , // 0x50003FC4
     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:414:42: error: initializer element is not a constant expression [-Wpedantic]
 #define DEFAULT_CCFG_BL_CONFIG           ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:473:5: note: in expansion of macro 'DEFAULT_CCFG_BL_CONFIG'
     DEFAULT_CCFG_BL_CONFIG          , // 0x50003FD8
     ^~~~~~~~~~~~~~~~~~~~~~
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:414:42: note: (near initialization for '__ccfg.CCFG_BL_CONFIG')
 #define DEFAULT_CCFG_BL_CONFIG           ( \
                                          ^
/home/gareth/openthread-master/examples/../third_party/ti/cc26xxware/startup_files/ccfg.c:473:5: note: in expansion of macro 'DEFAULT_CCFG_BL_CONFIG'
     DEFAULT_CCFG_BL_CONFIG          , // 0x50003FD8
     ^~~~~~~~~~~~~~~~~~~~~~
Makefile:707: recipe for target 'libopenthread_cc2650_a-cc2650_ccfg.o' failed
make[5]: *** [libopenthread_cc2650_a-cc2650_ccfg.o] Error 1
Makefile:467: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
Makefile:451: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:613: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:538: recipe for target 'all' failed
make[1]: *** [all] Error 2
examples/Makefile-cc2650:232: recipe for target 'do-build-cortex-m3' failed
make: *** [do-build-cortex-m3] Error 2

Have I not set something correctly with GCC?

Thanks in advance.

beriberikix, 2017-06-19:
What happens when you enter `arm-none-eabi-gcc --version`?

fizz-bang, 2017-06-19:
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

fizz-bang, 2017-06-19:
gcc --version
gcc (Ubuntu 5.4.1-2ubuntu1~16.04) 5.4.1 20160904
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

beriberikix, 2017-06-19:
Looks like you have version of the ARM GCC compiler (standard GCC won't work.) To level set, have you tried the [codelab](https://codelabs.developers.google.com/codelabs/openthread-simulation/) first to see if you have everything else set up correctly?

jwhui, 2017-06-19:
There is an issue with compiling the cc2650 example using gcc6.  As noted in [examples/platforms/cc2650/README.md](https://github.com/openthread/openthread/blob/master/examples/platforms/cc2650/README.md), the cc2650 platform is currently tested against `arm-none-eabi-gcc 5.4.1 20160609 (release)`.  Try using gcc 5.4.1 instead.

raryanpur, 2017-06-20:
I had the same issue as @fizz-bang building the cc26xx examples using gcc6 and resolved by rolling back to `arm-none-eabi-gcc 5.4.1 20160609 (release) `as mentioned by @jwhui

fizz-bang, 2017-06-20:
Indeed, that seems to be the problem.  I do not get errors using 4.9.3 20150529  which is easy to install in Ubuntu.  I will upgrade to 5.4.1. when I get a bit further on.

Thanks everyone!

closed 2017-06-20 by fizz-bang

leslieguyo, 2017-09-04:
How do I install a specific version 5.4.1 using apt-get?

beriberikix, 2017-09-04:
The ARM team doesn't maintain older packages on apt-get. I'm not on a Linux machine but I'm pretty sure you can find older packages maintained by third parties. See the update from [ARM](https://launchpad.net/gcc-arm-embedded/+announcement/13824).

jwhui, 2017-09-05:
FWIW, you can grab older versions directly from https://launchpad.net/gcc-arm-embedded

leslieguyo, 2017-10-06:
guys I am new to linux. How do I install a previous version of gcc-arm-none-eabi.  sudo apt-get install will always get the latest version and using this version to build openthread is giving errors. The bootstrap script has this command and I am stuck. Please help. I need step-by-step instructions if possible

jwhui, 2017-10-06:
@leslieguyo 

1. Download pre-built binaries from: https://launchpad.net/gcc-arm-embedded
2. Extract the downloaded file.
3. Update your `PATH` to point to the bin directory.
```bash
export PATH=$HOME/gcc-arm-none-eabi-5_4-2016q3/bin:$PATH
```

leslieguyo, 2017-10-06:
if I try to run arm-none-eabi-gcc  --version I get this message $HOME/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: No such file or directory

jwhui, 2017-10-06:
Can you copy-paste the terminal output here?

leslieguyo, 2017-10-06:
leslie@leslie-VM:~$ export PATH=$HOME/gcc-arm-none-eabi-5_4-2016q3/bin:$PATH
leslie@leslie-VM:~$ arm-none-eabi-gcc --version
bash: /home/leslie/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: No such file or directory
leslie@leslie-VM:~$

jwhui, 2017-10-06:
@leslieguyo, as mentioned in my comment above, can you validate that `/home/leslie/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc` actually exists?

leslieguyo, 2017-10-06:
leslie@leslie-VM:~$ ls -l /home/leslie/gcc-arm-none-eabi-5_4-2016q3/bin
total 27132
-rwxr-xr-x 1 leslie leslie  772740 Sep 26  2016 arm-none-eabi-addr2line
-rwxr-xr-x 2 leslie leslie  800200 Sep 26  2016 arm-none-eabi-ar
-rwxr-xr-x 2 leslie leslie 1306664 Sep 26  2016 arm-none-eabi-as
-rwxr-xr-x 2 leslie leslie  780464 Sep 26  2016 arm-none-eabi-c++
-rwxr-xr-x 1 leslie leslie  771332 Sep 26  2016 arm-none-eabi-c++filt
-rwxr-xr-x 1 leslie leslie  776368 Sep 26  2016 arm-none-eabi-cpp
-rwxr-xr-x 1 leslie leslie   25520 Sep 26  2016 arm-none-eabi-elfedit
-rwxr-xr-x 2 leslie leslie  780464 Sep 26  2016 arm-none-eabi-g++
-rwxr-xr-x 2 leslie leslie  776368 Sep 26  2016 arm-none-eabi-gcc
-rwxr-xr-x 2 leslie leslie  776368 Sep 26  2016 arm-none-eabi-gcc-5.4.1
-rwxr-xr-x 1 leslie leslie   24324 Sep 26  2016 arm-none-eabi-gcc-ar
-rwxr-xr-x 1 leslie leslie   24292 Sep 26  2016 arm-none-eabi-gcc-nm
-rwxr-xr-x 1 leslie leslie   24324 Sep 26  2016 arm-none-eabi-gcc-ranlib
-rwxr-xr-x 1 leslie leslie  447252 Sep 26  2016 arm-none-eabi-gcov
-rwxr-xr-x 1 leslie leslie  418572 Sep 26  2016 arm-none-eabi-gcov-tool
-rwxr-xr-x 1 leslie leslie 4522888 Sep 26  2016 arm-none-eabi-gdb
-rwxr-xr-x 1 leslie leslie 5004552 Sep 26  2016 arm-none-eabi-gdb-py
-rwxr-xr-x 1 leslie leslie  837764 Sep 26  2016 arm-none-eabi-gprof
-rwxr-xr-x 4 leslie leslie 1091400 Sep 26  2016 arm-none-eabi-ld
-rwxr-xr-x 4 leslie leslie 1091400 Sep 26  2016 arm-none-eabi-ld.bfd
-rwxr-xr-x 2 leslie leslie  784196 Sep 26  2016 arm-none-eabi-nm
-rwxr-xr-x 2 leslie leslie  953668 Sep 26  2016 arm-none-eabi-objcopy
-rwxr-xr-x 2 leslie leslie 1173092 Sep 26  2016 arm-none-eabi-objdump
-rwxr-xr-x 2 leslie leslie  800200 Sep 26  2016 arm-none-eabi-ranlib
-rwxr-xr-x 2 leslie leslie  464724 Sep 26  2016 arm-none-eabi-readelf
-rwxr-xr-x 1 leslie leslie  773732 Sep 26  2016 arm-none-eabi-size
-rwxr-xr-x 1 leslie leslie  773380 Sep 26  2016 arm-none-eabi-strings
-rwxr-xr-x 2 leslie leslie  953668 Sep 26  2016 arm-none-eabi-strip

jwhui, 2017-10-06:
How about simply executing:
```bash
/home/leslie/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc --version
```

leslieguyo, 2017-10-06:
leslie@leslie-VM:~$ /home/leslie/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc --version
bash: /home/leslie/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: No such file or directory
leslie@leslie-VM:~$

jwhui, 2017-10-06:
It appears your system is 64-bit - you need to install the 32-bit libraries.  See more [here](https://ioprog.com/2015/06/27/setting-up-arm-none-eabi-gcc-on-ubuntu-15-04/).

leslieguyo, 2017-10-06:
yes it is. Forgot about that

leslieguyo, 2017-10-06:
okay working now
Thanks a lot.