openthread/openthread #1928: openthread seems like

Daparrag, 2017-06-22:
**Hi, everyone** 

maybe you have earned about @contiki operating system  which has the same network architecture present in @openthread ; Maybe I'm wrong but what is the principal difference between  @openthread  and the competitors like @contiki  or @RTOS operating systems. 

**What is Contiki?**

- Contiki is an open source operating system for the Internet of Things. 
- Contiki connects tiny low-cost, low-power microcontrollers to the Internet. 
- Contiki is a powerful toolbox for building complex wireless systems.

aeliot, 2017-06-22:
From my understanding, Contiki is an OS, where as OpenThread is a networking stack.  Although OT can be run bare metal (as is common in our example code), it can also be run with an OS like Contiki or FreeRTOS.

jwhui, 2017-06-22:
Adding to @aeliot's comments, OpenThread is designed to be OS and platform agnostic.  You can read more in a recent [blog post](https://nest.tech/an-open-foundation-for-the-connected-home-1070a8e41e34).

beriberikix, 2017-06-22:
And as an example of how OpenThread runs *in* an OS and is not a competitor to something like Contiki, check out the integration of OpenThread with [RIOT](https://github.com/RIOT-OS/RIOT/pull/7082), another RTOS.

Daparrag, 2017-06-23:
**Hi everyone**
Thanks you so much for clarify my doubts.

I'm understood  that this **openthread** is quite interesting and soon I will try to get involved. 
 
again thanks you so much.

closed 2017-06-23 by jwhui