openthread/openthread #1939: bootstrap kw41z

khn123, 2017-06-27:
GitHub Issues are for bugs and feature requests.  To make bugs and feature requests more easy to find and organize, we close issues that are deemed out of scope for GitHub Issues.

Usage questions? Post questions to [Stack Overflow](http://stackoverflow.com/) using the [[openthread] tag](http://stackoverflow.com/questions/tagged/openthread). We also use Google Groups for discussion and announcements:

* [openthread-announce](https://groups.google.com/forum/#!forum/openthread-announce) - subscribe for release notes and new updates on OpenThread
* [openthread-users](https://groups.google.com/forum/#!forum/openthread-users) - the best place for users to discuss OpenThread and interact with the OpenThread team
* [openthread-devel](https://groups.google.com/forum/#!forum/openthread-devel) - team members discuss the on-going development of OpenThread

khn123, 2017-06-27:
I just compiled the example and I got the following ...
[machine@ubuntu:~/openthread$ ./bootstrap 
./third_party/nlbuild-autotools/repo/scripts/bootstrap: 280: ./third_party/nlbuild-autotools/repo/scripts/bootstrap: --automake: not found
machine@ubuntu:~/openthread$ make -f examples/Makefile-kw41z 
  CONFIG   kw41z...
(cd build/kw41z && /home/machine/openthread/examples/../configure INSTALL="/usr/bin/install -p" CPP="arm-none-eabi-cpp" CC="arm-none-eabi-gcc" CXX="arm-none-eabi-g++" OBJC="cc" OBJCXX="" AR="arm-none-eabi-ar" RANLIB="arm-none-eabi-ranlib" NM="arm-none-eabi-nm" STRIP="arm-none-eabi-strip" CPPFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-kw41z-config.h\"' -I/home/machine/openthread/examples/../examples/platforms/kw41z/  -mcpu=cortex-m0plus -mfloat-abi=soft -mthumb " CFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-kw41z-config.h\"' -I/home/machine/openthread/examples/../examples/platforms/kw41z/  -mcpu=cortex-m0plus -mfloat-abi=soft -mthumb " CXXFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-kw41z-config.h\"' -I/home/machine/openthread/examples/../examples/platforms/kw41z/  -mcpu=cortex-m0plus -mfloat-abi=soft -mthumb -fno-exceptions -fno-rtti " LDFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-kw41z-config.h\"' -I/home/machine/openthread/examples/../examples/platforms/kw41z/  -mcpu=cortex-m0plus -mfloat-abi=soft -mthumb -specs=nano.specs -specs=nosys.specs -Wl,--gc-sections -Wl,-Map=map.map " --host=arm-none-eabi --prefix=/ --exec-prefix=/kw41z --enable-cli-app=all --enable-ncp-app=all --with-ncp-bus=uart --enable-diag --with-examples=kw41z --with-platform-info=KW41Z )
/bin/sh: 1: /home/machine/openthread/examples/../configure: not found
examples/Makefile-kw41z:226: recipe for target 'build/kw41z/config.status' failed
make: *** [build/kw41z/config.status] Error 127
](url)
Anyone can help me?

hubertmis, 2017-06-27:
I had similar problem. I installed `libtoolize` to solve it.

jwhui, 2017-06-29:
@khn123, did the suggestion @hubertmis help resolve your issue?

bydottck13, 2017-07-05:
@jwhui, I had the similar problem, and this works for me:
# sudo apt-get install libtool

My OS is Ubuntu 14.04 32 bits.

khn123, 2017-07-17:
@jwhui  this works for me. Thx @hubertmis .

closed 2017-07-17 by khn123