openthread/openthread #1978: MLE Announce processing corner case

thcu-gp, 2017-07-11:
Currently the Thread spec only specifies what to do if `local_timestamp > announce_timestamp` and what to do if `local_timestamp < announce_timestamp` The case where both timestamps are equal is unspecified.

OpenThread acts as if `local_timestamp > announce_timestamp` when both timestamps are equal. This causes a huge number of MLE Annouce Messages to be sent in rapid succession (see 
[InfiniteAnnounce.pcapng.gz](https://github.com/openthread/openthread/files/1138878/InfiniteAnnounce.pcapng.gz))

There is no behaviour specified in the Thread spec (raised the issue [SPEC-730](https://threadgroup.atlassian.net/browse/SPEC-730) to address this). I my opinion in this case it is not too bad to do nothing (no not apply the information from the MLE Announce and do not send any MLE announces yourself)

This would mean to split the `if ... else` into an `if ... elif ... else` construct in mle.cpp:3036. (I know `elif` does not exist in c++, but it explains it more clearly)
The above would address the issue in the short term (and is compliant with the spec), until the Thread group figures out what to do with the issue

jwhui, 2017-07-11:
Thanks for identifying this, @thcu-gp!  I agree with your suggested fix.  Would you like to submit a PR?

jwhui, 2017-07-13:
Resolved by #1984.

closed 2017-07-13 by jwhui