openthread/openthread #1979: uint8_t* vs otExtAddress

thcu-gp, 2017-07-11:
Sometimes the OpenThread api uses uint8_t* (of length 8) and sometimes it uses otExtAddress to represent an extended address.
*  `otLinkSetAssignLinkQuality` has argument `uint8_t* aExtAddr`
* `gpOpenThreadApi_ThreadSetSteeringData` has argument `otExtAddress_t aExtAddress`

I believe a unified approach would be better

jwhui, 2017-07-11:
Thanks for flagging this issue.  I agree we should unify around `otExtAddress` rather than the generic `uint8_t*`.

Note that `otLinkSetAssignLinkQuality()` is being replaced as a part of #1967.  Are there other instances that need changing that you are aware of?

thcu-gp, 2017-07-12:
| file | function |
|--|--|
| src/core/api/link_raw_api.cpp:202 | otError otLinkRawSrcMatchAddExtEntry(otInstance *aInstance, const uint8_t *aExtAddress) |
| src/core/api/link_raw_api.cpp:226 | otError otLinkRawSrcMatchClearExtEntry(otInstance *aInstance, const uint8_t *aExtAddress) |
| src/core/api/link_api.cpp:65 | const uint8_t *otLinkGetExtendedAddress(otInstance *aInstance)

jwhui, 2017-07-13:
Thanks for identifying those other instances.  Are you planning to submit a PR for this?

PR #2029 (2017-07-28): [radio] unify around otExtAddress type
Resolves #1979.
closed 2017-07-28 by jwhui