openthread/openthread #2037: CoAP - Duplicate message detection & Commissioning Problems

DuaneEllis-TI, 2017-07-27:
This is the embedded side of:  https://github.com/openthread/borderrouter/issues/81

At a general level, the public CoAP API should provide a means to detect duplicate CoAP requests of a server,  this currently does not exist.

At a specific level, in the Leader when a commissioning petition is received there is no 'duplicate message detection' - there needs to be.

For example, this time line:

1) Using the Android commissioning App as an example - it sends a Commission Petition.
2) The otbr-agent (on the e border router) receives a commissioning petition.
3) This is forwarded to the NCP, which then forwards to the current LEADER

Note: the key variable here is: 'mSessionId", this is the commissioning session id.
As a concrete example, we'll use the mSessionId = 42

4) The leader grants the request and sends back a reply, with SessionID = 42

See: https://github.com/openthread/openthread/blob/master/src/core/meshcop/leader.cpp#L102

5) HOWEVER - if the Android app sends a DUPLICATE petition (and it does)
6) The leader receives the DUPLICATE petition - there is no duplicate detection

7) The Leader code does "++mSessionId" and generates a new session id, and sends the response.
The Leader now has mSessionID = 43

8) Later - (about 20 to 30 seconds later) - the Android App sends KEEP ALIVE, the android app however sends back the FIRST session ID  ie:  ID=42, however the Leader has "43"

Then here (below) when the KeepAlive is managed the session IDs do not match..
   https://github.com/openthread/openthread/blob/master/src/core/meshcop/leader.cpp#L189

And as a result a "REJECT" message is sent to the Android App.

9) Sadly - there is a 3rd bug - in the Android App - it seems to ignore the REJECT message and continues to say that it is trying to find the device ... The android app sits there ... displaying searching.... endlessly, meanwhile every 30 seconds the Commissioner sends REJECT in response to keep alive messages.

jwhui, 2017-07-27:
OpenThread's CoAP implementation should already include duplicate detection and response retransmission ([relevant code](https://github.com/openthread/openthread/blob/master/src/core/coap/coap.cpp#L656)).

Does the "duplicate" CoAP request from the Android app contain the same CoAP Message ID?  If so, then I believe this is more of an issue in the Border Agent.  If not, these are really distinct CoAP Requests.  The question is then what to do with distinct CoAP Request from an already-accepted Commissioner.  Unfortunately, it is not possible to know if Commissioner has maintained any state since the prior request.  As a result, OpenThread takes the conservative approach and reinitializes the Commissioner state.

DuaneEllis-TI, 2017-07-27:
I did not see the "coap" level duplicate detection - thanks for pointing that out.

The android app sends the *SAME* message ID.

The AGENT does not recognize the duplicate, and creates a *NEW* message ID.

In fact, I think the AGENT probably should *NEVER* do duplicate detection.  Instead *only* the end point should perform duplicate detection. Rational:  Lets say the forwarded/proxied message (Agent->NCP->Leader) got lost because some intermediary did not receive the message, or did not have buffers for the message.  The Proxy *COULD* - perform a retry operation... but I don't think *both* the Android App & the Agent should perform retries.

jwhui, 2017-07-27:
In general, I agree with your comments around the Border Agent.  However, unless there is a stateless 1:1 mapping between the received COMM_PET and sent LEAD_PET message, the Border Agent probably needs to perform some kind of duplicate detection.  Because the Border Router may generate CoAP requests for other purposes, we may need to maintain some kind of mapping between CoAP message IDs.

DuaneEllis-TI, 2017-07-27:
Is there an absolute hard requirement that the proxy always sends (to the network/leader) from a specific port?  I believe the answer is no, because the spec talks about the source as a Ephemera port number (reference spec 8.4.7.3.1 sequence diagram in figure 8-16)

Today, the agent supports/requires 1 proxy - for commissioning and only commissioning.

If in the future more proxies are required ( I think that is what you are suggesting) - those other proxies can choose any *other* port other then the currently active Proxy port.  If so then all proxies on that interface must coordinate each others port usage.   And we can cross that bridge when we require more then 1 active proxy. When that happens, we probably need some type of LRU on the proxy port number allocation with an aging or timeout scheme.

Because each proxy now has a distinct source port, the one-to-one mapping for message IDs is easy, just copy the message ID from (external-message) to (proxied-message)

jwhui, 2017-07-27:
Makes sense to me.  There is no hard requirement on the source port.  However, we currently have a single CoAP instance (performing both client and server roles) bound to the TMF port on the NCP.  One path forward is to push the Border Agent's CoAP Client functionality that interacts with a new UDP socket on the NCP side.

DuaneEllis-TI, 2017-07-27:
Not sure I completely understand your "however..."

can you say that another way?

jwhui, 2017-07-27:
OpenThread's CoAP implementation currently combines Client and Server functionality on a single UDP socket.  The TMF Proxy on the NCP utilizes the same CoAP instance on the NCP.  To utilize separate UDP source ports between TMF Client interactions originated by the NCP itself and those provided via the TMF Proxy, we could either (i) create a second CoAP instance on the NCP to support the TMF Proxy or (ii) push the CoAP Client logic into the Border Agent on the Linux side and have the TMF Proxy sit directly on top of a UDP socket on the NCP (rather than the CoAP instance).

DuaneEllis-TI, 2017-07-27:
I think that is a *future* problem that we must solve when we have to support more then 1 proxy on the agent and it is not a current issue that must be addressed as part of this idea/method to fix.

jwhui, 2017-07-27:
Not sure I follow.  There is a single CoAP instance on the NCP for handling all TMF messages.  IIf you want to send CoAP Requests with different source ports, we need to have more than one CoAP instance.  For example, Address Queries are sent on one source port and Petitions are sent with a different source port.  What am I missing?

DuaneEllis-TI, 2017-07-27:
I am saying this: The Agent only preforms proxies on behalf of a the commissioning app, and since there can only be 1 commissioner at a time ... then it is legal for the agent/proxy to copy the message IDs from the (public-side) to the (proxy'ed-side) and visa-versa.

You seem to be saying the Agent will, or could do other things in the future.  My answer is this: If/when that happens we solve the multiple proxies then, but until that bridge arrives we take advantage of a the singleness of the commissioning process.

jwhui, 2017-07-27:
What I'm saying is that the Border Agent currently shares the same CoAP endpoint as all other TMF CoAP interactions on the NCP.

jwhui, 2017-07-27:
Let me try to pull this discussion back in.  The TMF Proxy code on the NCP currently sits on top of the NCP's CoAP instance.  If we simply changed that to sit directly on top of a UDP socket bound to an ephemeral port, that should satisfy the proposal here, right?

In other words, the host-side Border Agent would generate the LEAD_PET.req message (with the CoAP Message ID copied from the received COMM_PET.req message) and send it via a UDP socket (which happens to be resident on the NCP).

DuaneEllis-TI, 2017-07-28:
@jwhui 

> > In other words, the host-side Border Agent would generate the LEAD_PET.req message (with the CoAP Message ID copied from the received COMM_PET.req message) and send it via a UDP socket (which happens to be resident on the NCP).

partial answer  - yes - however - I think we need also need to do that for the other proxied messages (keep alive, commissioner set, and relay tx and tx - the agent just uses/copies the same message id). 

I say partial answer because I don't have the entire design in my head as well as you do, I'm not sure about how the port numbers are allocated/managed and matched up/reconnected

jwhui, 2017-07-28:
@DuaneEllis-TI 

OK, I think I have a good understanding of what needs to happen.

> partial answer  - yes - however - I think we need also need to do that for the other proxied messages (keep alive, commissioner set, and relay tx and tx - the agent just uses/copies the same message id). 

Yes, I provided COMM_PET/LEAD_PET as just a single example of how it would work and did not mean to limit the solution to those messages.

jwhui, 2017-10-26:
@DuaneEllis-TI, is this still an open issue?

DuaneEllis-TI, 2017-10-26:
resolved

closed 2017-10-26 by DuaneEllis-TI