openthread/openthread #2040: cc2538 makefile, configure file not found

jaimebohigas, 2017-07-28:
GitHub Issues are for bugs and feature requests.  To make bugs and feature requests more easy to find and organize, we close issues that are deemed out of scope for GitHub Issues.

Usage questions? Post questions to [Stack Overflow](http://stackoverflow.com/) using the [[openthread] tag](http://stackoverflow.com/questions/tagged/openthread). We also use Google Groups for discussion and announcements:

* [openthread-announce](https://groups.google.com/forum/#!forum/openthread-announce) - subscribe for release notes and new updates on OpenThread
* [openthread-users](https://groups.google.com/forum/#!forum/openthread-users) - the best place for users to discuss OpenThread and interact with the OpenThread team
* [openthread-devel](https://groups.google.com/forum/#!forum/openthread-devel) - team members discuss the on-going development of OpenThread

I just begin to follow the first steps to compile the examples for the cc2538 chipset, and I found these errors:

/bin/sh: 1: /mnt/c/Users/JaimeBohigas/openthread/examples/../configure: not found

I looked for configure file but on the repository is missed. How can I fix the problem??

Thank you very much.

Kind regards

jwhui, 2017-07-28:
It would help if you referenced the steps that you actually tried and their output.  In particular, did you follow the steps in this [README.md](https://github.com/openthread/openthread/blob/master/examples/platforms/cc2538/README.md)
?  Also, make sure you have the GNU toolchain installed (autoconf, automake, libtool)

jaimebohigas, 2017-07-31:
Dear all, specially Mr. jwhui,

I have followed the steps on the README.md, and also I use the bash on windows toolchain.

I verified I have installed the autoconf, automake and libtool too.

The libtool version I have got is 2.4.6-0.1
The automake version I have got is 1:1.15-4ubuntu1
the autoconf version I have gor is 2.69

This is the report I have on compile time

jaimegm@DESKTOP-95UCL84:/mnt/c/Users/JaimeBohigas/openthread$ make -f examples/Makefile-cc2538
  CONFIG   cc2538...
(cd build/cc2538 && /mnt/c/Users/JaimeBohigas/openthread/examples/../configure INSTALL="/usr/bin/install -p" CPP="arm-none-eabi-cpp" CC="arm-none-eabi-gcc" CXX="arm-none-eabi-g++" OBJC="cc" OBJCXX="" AR="arm-none-eabi-ar" RANLIB="arm-none-eabi-ranlib" NM="arm-none-eabi-nm" STRIP="arm-none-eabi-strip" CPPFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-cc2538-config.h\"' -I/mnt/c/Users/JaimeBohigas/openthread/examples/../examples/platforms/cc2538/  -mcpu=cortex-m3 -mfloat-abi=soft -mthumb " CFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-cc2538-config.h\"' -I/mnt/c/Users/JaimeBohigas/openthread/examples/../examples/platforms/cc2538/  -mcpu=cortex-m3 -mfloat-abi=soft -mthumb " CXXFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-cc2538-config.h\"' -I/mnt/c/Users/JaimeBohigas/openthread/examples/../examples/platforms/cc2538/  -mcpu=cortex-m3 -mfloat-abi=soft -mthumb -fno-exceptions -fno-rtti " LDFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-cc2538-config.h\"' -I/mnt/c/Users/JaimeBohigas/openthread/examples/../examples/platforms/cc2538/  -mcpu=cortex-m3 -mfloat-abi=soft -mthumb -nostartfiles -specs=nano.specs -specs=nosys.specs -Wl,--gc-sections " --host=arm-none-eabi --prefix=/ --exec-prefix=/cc2538 --enable-cli-app=all --enable-ncp-app=all --with-ncp-bus=uart --enable-diag --with-examples=cc2538 --with-platform-info=CC2538 --enable-linker-map )
/bin/sh: 1: /mnt/c/Users/JaimeBohigas/openthread/examples/../configure: not found

I how It can show the problem.

Thank you very much.

Kind regards.

jwhui, 2017-07-31:
Did you run `./bootstrap` then `make -f examples/Makefile-cc2538`?  What was the output, if any, from `./bootstrap`?

jaimebohigas, 2017-08-01:
Dear all,

Finally I have finished the make procedure.

I made a mistake because I tried to execute the file bootstrap before I had all the necesary programs.
 
Thank you very much.

closed 2017-08-01 by jaimebohigas

leblond113, 2021-09-23:
@jaimebohigas whitch files did you mean ?
I have the same problem and i can't solve the issue.
Please how did you exactly manage to figure it out  ?

jwhui, 2021-09-23:
> @jaimebohigas whitch files did you mean ?
> I have the same problem and i can't solve the issue.
> Please how did you exactly manage to figure it out ?

Did you follow the directions in [examples/platforms/cc2538/README.md](https://github.com/openthread/openthread/blob/main/examples/platforms/cc2538/README.md)?