openthread/openthread #2050: coap start : OpenThread Command is not working for KW41z

Embedded-1, 2017-08-01:
Hello,

I had built the OpenThread application for KW41z Board, flashed the binary and joined with another Node.

Here I observed that "coap start" command is not working fine as per expectation. 

Kindly see the logs on its terminal:
******************************************************************************************
> ipaddr
fdf4:e9a9:c246:b131:0:ff:fe00:400
fd01:0:0:3ead:ae81:371b:79a8:1859
fdf4:e9a9:c246:b131:dfd6:ae1f:d86d:70ba
fe80:0:0:0:fcd5:4214:1c68:4183
> state
router
Done
> coap start
Error 6: Parse
******************************************************************************************

Here I worked on this git commit :-
-----------------------------------------------------------------------------------------------
commit 31b89311babc8ad1876a4038857f0d4f11e12f22
Author: Buke Po <bukepo@gmail.com> 
Date:   Tue Aug 1 05:54:59 2017 +0800

[spinel] support in place unpack. (#2025)
-----------------------------------------------------------------------------------------------

Let me know if any change needed to enable coap functionality into the source code or is it known issue ?

Thanks.

LuDuda, 2017-08-01:
Hello,

To make a use of OpenThread CoAP implementation in the application layer, you need to explicitly enable it.

Please add extra argument when you build your project, I guess for your platform it would look like:
`make -f examples/Makefile-kw41z COAP=1`

Internally it will add `--enable-application-coap` option to `configure`.

Embedded-1, 2017-08-02:
Thanks, @LuDuda . It is working fine.

closed 2017-08-02 by jwhui