openthread/openthread #2068: DEBUG UART

DuaneEllis-TI, 2017-08-05:
Observation: often physical devices have more then 1 physical UART

1) The NCP does have support to push the DEBUG LOG upto the host platform over a "reverse channel" 
This is nice but depending upon log volume - the rate of increased traffic causes other problems.

2) In the GRL test harness, it took talks to the CLI and does not expect log messages.
Using a second UART for LOGs will make it easier to debug certain issues.

3) In the POSIX platform, when debugging problems that go wrong in self-test-automation-scripts the LOG feature interferes with and messes up the UNIT test, in the CLI the easiest solution is to provide a CLI command where you can specify a filename to capture the debug log output.

We have this working internal at TI for CC2538 and CC2652 and POSIX platforms and are looking at pushing this up.

jwhui, 2017-08-05:
@DuaneEllis-TI, separating the log output stream from the CLI is very useful.  OpenThread can actually be configured to use the platform-provided `otPlatLog()` implementation or to stream them through the CLI or Spinel.  See [`OPENTHREAD_CONFIG_ENABLE_DEFAULT_LOG_OUTPUT`](https://github.com/openthread/openthread/blob/master/src/core/openthread-core-default-config.h#L618).  Having a separate log output stream on embedded platforms would be a great enhancement.

Note that in the posix example platform, [`otPlatLog()`](https://github.com/openthread/openthread/blob/master/examples/platforms/posix/logging.c#L73) utilizes `syslog`, which enables a number of nice logging facilities in addition to separating the log output stream from the CLI.

DuaneEllis-TI, 2017-08-05:
yes, well aware of these things, and the Macro to set the log function.

jwhui, 2017-08-05:
For the posix use case, what tradeoffs do you see with using syslog vs. writing directly to a file?

DuaneEllis-TI, 2017-08-05:
I am not against using SYSLOG, the feature should remain and not be removed.
I am only proposing an alternate for POSIX - this also serves as POSIX test vehicle for the embedded side of the "otPlatDebugUart_xxx()" functions.

But you asked, and I'm sure that there are many "linux types" that would ask the same question.
All of the below is from an 'non-linux' point of view

Currently the POSIX use case is nearly 100% totally a test environment, perhaps there are other plans in the future where this may change, hence by no means do I want to suggest *not* supporting syslog, or not using a "syslog style" of logging - the syslog model is a very good, and well designed model.

**The TL;DR version is this:**  Creating separate log files in a simple and easy way is the main advantage right now for the POSIX-CLI and POSIX-NCP applications. Because it closely mirrors the way physical devices work that you compare things against.

**The LONG VERSION...**

* Why do I want my "test logs" to be intermixed with USB plug & unplug event messages, or NTP time update notifications? How are these relevant to the test or problem I am debugging? In a test environment what benefit does including those extraneous events give me?

* If on TRAVIS machine 
   * all tests log into the SYSLOG, and there are well over 50 different tests in `${src}/tests/scripts`, how would you extract all 50 different tests with (3 to 5) different devices all writing data into one common text file called: SYSLOG?

   * If nothing goes wrong, you don't care. The log is deleted.

   * if something does go wrong... it will be a PITA to figure out, which test used which process ID so that I can grep the syslog for that specific process ID of that specific test element, and create separate files

* For example if TEST 1234 has gone south, and has 3 components - why not create at the on set:
   * test-1234-leader-log.txt
   * test-1234-router-log.txt
   * test-1234-device-log.txt

* The log  file /var/log/syslog - is a SYSTEM artifact, not a test artifact
    * On a Travis system, you get a brand new, completely reset machine every time.
    * My desktop Linux machine, or the VM (vagrant or otherwise) instance I use is not reset
        * Thus, every test just floods /var/log/syslog with needless noise.
    * It's simpler if all test artifacts are stored within a single directory structure.
        *  If I run tests N times today, I have N directories, the results of which are in those directories.
         * Self containment is good thing, I can pack things up ... as needed
         * or delete things if I don't need them any more.

* SYSLOG generally logs with YMD and HMS 

   * When reviewing test logs the first 25% of the 80 column screen (YMD & H) is just useless 
       * Some of us cannot read 6 pixel font that shows us 200 columns.
       * And when reviewing stuff side by side (ie: Pair Debugging) you are farther back from the screen

   * For the purposes of a test, I don't care that it is August 2017, 
       * The epoch is really when the test started, that is time zero.
           * Yes there may be 3 to 4 applications involved in the test...
       * my compromise is, "time zero" is when the application started.
       * this matches the embedded system, where timezero is at the reset vector.
       * All test times are relative relative to that point in time, thus that is "time zero"

   * Where SYSLOG came from - things on Unix machines LOGS are generally thought of "this morning there was activity" - or "about 10:30 lastnight the syslog shows.." The timespan is hours, days, if not weeks. Hence "log rotate" is a common thing.

   * In contrast - for THREAD tests, millisecond resolution is important here *because* timeouts, and events in the thread network are measured on millisecond values not  seconds or minutes.  SYSLOG does not support such a level.  And believe me I've been dealing with little bastard problems where timing is everything.

    * Another way of looking at the problem is:  "dmesg" vrs "syslog" - they exist for different purposes.

* Live verses Post-Mortem logs, There are times when it is helpful watch things happen live... 
   * You might wait for an event to occur then break execution and examine variables.
   * This is often helpful to compare: a known good vrs failure
   * In the automated tests scheme (where you are running a batch of 50 tests) you don't monitor the output of something - you just log all out put into files and do some type of "post-mortem" examination.
   * In contrast, when debugging an issue - being able to  on a POSIX system "tail -f  leader.log" is very similar to having a TERATERM window running on a physical LEADER device.

    * To my earlier point, it is easier to type:  "tail -f  test-1234-leader.log" 
    * Whats the command I would use to monitor that single test from the SYSLOG?
          *   Tail SYSLOG,  GREP for the application name, GREP for the specific PROCESS ID?
          *  Oh... it just crashed ... Did you remember to "tee   SOMELOGFILE" so you have a copy of the log?

* Human factors also play a giant role here, Team Members, and Customer Skill sets vary greatly
   * some are absolute Linux Gurus, but don't exactly get Embedded Systems.
        * And the reverse exists.
   * Some are very experienced, some are fresh out of school
   * Or they know their domain - their product, but not this side of the equation.
        * you need to explain to somebody how to create the log file so they can capture the log.
   * Telling them to add the command:   "logfilename FOOBAR" is easily understood by all.

For all of the above reasons, sometimes writing directly to a log file is just simpler.

jwhui, 2017-08-05:
Adding an option for specifying a log-output file sounds like a great enhancement to me.  I wasn't clear from your earlier comments whether you were proposing to replace the existing syslog-based implementation or to provide another option.  Thanks for providing your detailed thoughts.

I can see both syslog and dedicated log file output being useful depending on the use case and user.  FWIW, as an embedded developer, I used to do everything with separate raw log-output files.  At the same time, I've also found syslog to be useful in a number of scenarios.  In short, I see value in both models.

A few brief comments:
- We do have product use cases where OpenThread runs on the Linux host, which allows for reducing resource requirements on the IEEE 802.15.4 peripheral.  See this [blog post](https://nest.tech/an-open-foundation-for-the-connected-home-1070a8e41e34) and the [Windows README](https://github.com/openthread/openthread/tree/master/examples/drivers/windows).
- Syslog implementations vary in their capabilities (my machine, for example, provides microsecond-level timestamps).
- There are some great tools for viewing and analyzing syslog data, which help address many of the issues outlined above.

jwhui, 2017-08-24:
Resolved by #2082.

closed 2017-08-24 by jwhui