openthread/openthread #2081: COMM_KA vrs RELAY_{tx|rx} spec issue/question

DuaneEllis-TI, 2017-08-08:
Hmm - There seems to be a large timing issue that manifests in multiple ways, no one way is clearly the problem.  This could also be due to the way the Android Commissioning app is structured :-(

For the purposes of this discussion, I want to divide the commissioning process into 3 phases.

Phase 1 - is prior to the DISCOVERY REQUEST
Phase 2 - is during the time the RLY_{TX|RX} is active
Phase 3 - is after JOIN_FIN

During Phase 1 - the COMM_KA(keep alive) is a very useful message to determine if the application should shut down, or give up - etc - this make sense.

During Phase 2 - (when RLY_{TX|RX} is active) the COMM_KA(keepalive) becomes very problematic, and often does not occur at an actual rate that it needs to.

This also has to do with TIMEOUT_COMM_PET (50secs) and TIMEOUT_LEAD_PET(50secs), THR_DTLS_INIT_RETRANSMIT_TIMEOUT(8sec start), note the RE-TRANSMIT timeout increases exponentially.

In the android app (current design), there is a thread that does the keep alive process. This thread does not actually send the COMM_KA over the air (wifi), instead it queues the request into a work queue, another thread takes the COMM_KA  {and the RLY_TXRX messages} and sends these via the DTLS session to the AGENT.

The actual transmission time of the COMM_KA can be upwards of 20 seconds after the request is queued. Why so long? Because the under powered joiner the DTLS session with the JOINER is takes more time, in addition if there are *ANY* over the air re-transmissions of RF packets, it takes longer for the *LARGE* packets to be reassembled. 

In effect, the COMM_KA and RELAY_{TX|RX} are serialized and msg(A) slows down msg(B) because the messages are processed in order.  This often brings us just past the TIMEOUTs,  in any of the 4 parties involved (AndroidApp,   NCP,  JOINING_ROUTER,  and JOINING_DEVICE) it is not consistent, and not reliable.

One PROPOSED change - is to treat the RELAY_{TX|RX} activity as an alternate COMM_KA, the rational is: If the RELAY_TX & RX is occurring, this is occurring with the Border Agent, and is being sent by the COMMISSIONER - if those messages are succeeding (making forward progress in any form) then the commissioner has not walked away.

jwhui, 2017-08-08:
@DuaneEllis-TI, thanks for raising this issue.  This seems to be an implementation-specific issue in the Thread Group's Commissioning application.  The DTLS session from an External Commissioner terminates at the Border Agent.  As a result, the timing of sending DTLS messages between the Commissioner and Border Agent should be independent from the DTLS messages between the Commissioner and Joiner.

I would suggest filing a JIRA in the [Thread Group's COMMAPP project](https://threadgroup.atlassian.net/projects/COMMAPP/issues).

jwhui, 2018-06-06:
Closing due to inactivity.

closed 2018-06-06 by jwhui