openthread/openthread #2110: NCP add joiner PSK does not work without OPENTHREAD_ENABLE_BORDER_ROUTER=1

GiedriusM, 2017-08-17:
I am not aware if OPENTHREAD_ENABLE_BORDER_ROUTER=1 is required for a NCP device, but `wpanctl commissioner -a XXXXXXXXXXXXXXX` gives an error 16 (PropertyNotFound), which is caused by combination of 
```C
factory.set_lock_property(SPINEL_PROP_THREAD_ALLOW_LOCAL_NET_DATA_CHANGE);
```
in wpantund codebase and 
```C
#if OPENTHREAD_ENABLE_BORDER_ROUTER
    NCP_SET_PROP_HANDLER_ENTRY(THREAD_ALLOW_LOCAL_NET_DATA_CHANGE),
#endif
```
in the openthread codebase.

For me personally, not all NCP devices should/could be border routers, for example in cases there they are used just to give connectivity to other devices or act like a repeater/extender, and as such border router functionality could be omitted (?).

jwhui, 2017-08-22:
Thanks for pointing this out!  Off hand, I'm not sure why wpantund needs to lock the local network data - adding entries to the joiner list does not affect the local network data.

GiedriusM, 2017-08-22:
I haven't had time to take a closer look, but I've noticed that each time `wpanctl commissioner -a XXXXXXXXXXXXXXX` is executed, an MLE is sent. Attached pcap might give you some insight.

[image]

jwhui, 2017-12-04:
@abtink, can you take a look?

abtink, 2017-12-04:
We are not changing the network data so not sure why the `THREAD_ALLOW_LOCAL_NET_DATA_CHANGE` is being used. Probably copy-paste from another part. I can change this.

abtink, 2017-12-06:
Submitted https://github.com/openthread/wpantund/pull/256 which should address this.

abtink, 2017-12-06:
Above PR merged in wpantund.

closed 2017-12-06 by abtink