openthread/openthread #2120: Missing SRV implementation

Nopik, 2017-08-18:
It seems, that Service TLV and Server TLV along with all remaining implementation (like distribution of that in the Network Data) is missing, although Thread spec v1.1.1 mentions it and OpenThread claims compatibility with Thread 1.1.1

Nopik, 2017-08-18:
Can I get some basic pointers in order how to implement it properly? Consider me a beginner in the areas of OpenThread internals and Thread specification itself, please.

If I understand correctly, the implementation would consist of following parts:
a) leader when receiving Service / Server TLVs, add them to network data and broadcast them as usual. Some service ALOC assignment might be involved, not sure about that part yet. 
b) new API call need to be added to allow iterating the TLVs from network data. E.g. `otServerGetNextEntry()` taking some iterator, very much like  `otBorderRouterGetNextOnMeshPrefix()` work), so client might iterate across all entries and choose the one it needs.
c) some API for easily registering new service data (`otServerRegister()`) and send them to the leader.

Is there anything else?

Specifically, how the closest server will be used when client will try to send some packets?

jwhui, 2017-08-18:
Your proposal looks pretty good overall.  Some points that may help:

1. The Service ALOC is derived from the Service Type ID.  The Service Type ID is allocated by the Leader.
2. The API to register new Server information can be modeled after the existing APIs for adding/removing Border Router and External Route information.  See [include/openthread/border_router.h](https://github.com/openthread/openthread/blob/master/include/openthread/border_router.h).
3. The API to read out Server Information can also be modeled after the existing iterator-based APIs, as you noted.
4. As for forwarding messages with ALOC destination, there is a nice TODO waiting for you in [src/core/thread/mesh_forwarder.cpp](https://github.com/openthread/openthread/blob/master/src/core/thread/mesh_forwarder.cpp#L848) :)
5. If possible, add some tests that can be run as part of [continuous integration](https://travis-ci.org/openthread/openthread).

Nopik, 2017-08-18:
Thanks. I will probably need some help with that TODO part, though ;)

jwhui, 2017-08-18:
Sounds good.  Let's get through the Network Data registration and access first.  Then we can strap in the message forwarding logic.

Nopik, 2017-08-19:
@jwhui I did some preparation work, re-read the Thread spec to understand all nuances, also got the existing testsuite running and green. The testsuite seems to run the cli examples in the background and check the results. So, it seems, that the best way for me to move forward would be to extend the cli by commands like `server register` and `dump networkdata` (the latter I was already missing a lot). Then by proper coap invocations I can check the ALOC routing as well. 

Hopefully adding more commands to cli will be beneficial for other use cases as well.

Do you think that is the right way forward?

Nopik, 2017-08-19:
Actually, `netdataregister` should probably become `netdata register` + `netdata show` + `netdata unregister` + `netdata add srv ...`, etc. Probably keeping `netdataregister` around for compatibility reasons. Probably with some local & leader flavors (i.e. leader has its netdata, but locally we might have different data, so we need commands to show & manipulate both).

Nopik, 2017-08-19:
Also, there is another detail. The `ThreadNetif` class has this code:

```
#if OPENTHREAD_ENABLE_BORDER_ROUTER
    NetworkData::Local mNetworkDataLocal;
#endif  // OPENTHREAD_ENABLE_BORDER_ROUTER
```
and, obviously, few (5-10?) places, where code like

```
#if OPENTHREAD_ENABLE_BORDER_ROUTER
    mNetif.GetNetworkDataLocal().SendServerDataNotification();
#endif
```
or
```
#if OPENTHREAD_ENABLE_BORDER_ROUTER
    mNetif.GetNetworkDataLocal().ClearResubmitDelayTimer();
#endif
```
is used. If I'm reading this, those are exactly the places where network data local should be handled (cleared, sent, retransmitted, etc.) - so I'm thinking about removing `#if OPENTHREAD_ENABLE_BORDER_ROUTER` from those places. The `ot::NetworkData::Local` class seems to be awfully dependent on the border router, though. One of the obvious choices would be to `#ifdef` the hell out of it. Another, would be to create some class like `ot::NetworkData::SrvLocal` or something - but I dont like that duplication, the more complex `Local` class seems to be better after all. With a bit of luck the conditional logic will not be hard.

What do you think?

jwhui, 2017-08-21:
@Nopik, thanks for sharing your thoughts and proposals.

1. For the CLI commands, I think we should simply add commands to add/remove server data.  `netdataregister` can be used to trigger the `SRV_DATA.ntf` message.  I understand the desire for viewing the raw contents of the network data.  In practice, this is only useful to core Thread/OpenThread developers.  The APIs should provide information useful to applications.  For example, the Border Router and External Route APIs present information in a useful way to applications.  For the raw byte contents, I would suggest simply turning on logging.  Search for the `otDumpDebgNetData()` logging calls in the `network_data_*.cpp` sources.

2. I agree that we should simply modify the existing network data classes, rather than create a new one.  For the `#if` preprocessor statements, I believe those are still useful - but could be renamed.  Not all devices care to publish their own network data.  For example, devices that do not serve as border routers or do not care to offer services to other devices.  I would be okay with leaving the `OPENTHREAD_ENABLE_BORDER_ROUTER` as is for now, then consider to renaming something like `OPENTHREAD_ENABLE_LOCAL_NETWORK_DATA` at a later time.

Nopik, 2017-08-21:
Right. I'll go as you described.

For the record, I was thinking about doing the API layer just as you described, and maybe extend CLI by some display of network data - not about exposing API to get the raw bytes.

As a person who've been learning Thread/OpenThread details, I really found it annoying not being able to see the network data in the CLI - just for the learning purposes. 

As for the `#if`, I think the best way forward will be to create `OPENTHREAD_ENABLE_LOCAL_NETWORK_DATA` right away and add SRV-related stuff there, and  have code
```
#if OPENTHREAD_ENABLE_BORDER_ROUTER
#undef OPENTHREAD_ENABLE_LOCAL_NETWORK_DATA
#define OPENTHREAD_ENABLE_LOCAL_NETWORK_DATA 1
#endif
```
so the compatibility will be there, and it will not be possible to shoot yourself in the foot (by enabling BR but disabling LND). Yet also give the options to use LND without BR, or ditch them both. After this it makes sense to refactor few of the current `#if BR` uses to LND already (when I went through sources, it seems to be about ~10 occurences in the code which would be relevant).

Nopik, 2017-08-21:
@jwhui 2 questions:

1) It seems that Border Router API has 2 methods which will be 100% identical in my Server API, namely `otBorderRouterGetNetData()` and `otBorderRouterRegister()`. They just seem to operate on the underlying Network Data Local structure directly. I would strongly recommend getting some public API to do that, instead of duplicating the code between Border Router and Server implementations. I.e. having 2 identical methods like this:

```
otError otBorderRouterGetNetData(otInstance *aInstance, bool aStable, uint8_t *aData, uint8_t *aDataLength)
{
    otError error = OT_ERROR_NONE;

    VerifyOrExit(aData != NULL && aDataLength != NULL, error = OT_ERROR_INVALID_ARGS);

    aInstance->mThreadNetif.GetNetworkDataLocal().GetNetworkData(aStable, aData, *aDataLength);

exit:
    return error;
}
```
doesn't seem a good idea. I would even go a step further and mark `otBorderRouterGetNetData` as deprecated in favour of `otNetworkDataLocalGetData`, same for `otBorderRouterRegister()` to be replaced by `otNetworkDataLocalRegister`.

What do you think?

2) The implementation of `otBorderRouterGetNetData` seem to have subtle bug - it doesn't check the length of passed buffer. I.e. if caller will have too short buffer, `otBorderRouterGetNetData` will not check the length and call `aInstance->mThreadNetif.GetNetworkDataLocal().GetNetworkData()`, which in turn will not check the length neither, and happily proceed with `memcpy` - leading to memory corruption. Was that a bug or intentional optimization? If so, then https://github.com/Nopik/openthread/blob/master/include/openthread/border_router.h#L60-L61 is heavily misleading and should be changed to `out` and something like "You need to provide at least 256 bytes." or something like that.

jwhui, 2017-08-22:
@Nopik 

1. I think it makes sense to rename the APIs to `otNetDataLocalGet()` and `otNetDataLocalRegister()`.  Note that we already have a bunch of `otNetData*` APIs, but those are for the Thread Network Data published by the Leader.

2. The documentation of `otBorderRouterGetNetData()` indicates that the passed in value indicates the size of the output buffer.  As a result, I would say this is a bug.

Nopik, 2017-08-26:
I think there is some problem in the Thread 1.1.1 specification, regarding Service TLV. Generally, when some server want to register itself, it sends `SVR_DATA.ntf` to the leader. That request should contain Service TLV, but it in turn contains `S_id` field, which is assigned by the leader. What value local server should assign to the `S_id` field, when registering new service?

Most likely this field should be ignored by the leader, though it looks like mistake in the Thread specifications, actually.

Nopik, 2017-08-26:
The nesting of the structures (list of services, each having a list of server), where both service and server TLV have their own data blobs is pretty annoying when it needs to be defined as a static C arrays. Namely, it suggests such structures:

```
struct ServerConfig {
   uint8_t mServerData[ 255 ];
   // ...
} 

struct ServiceConfig {
   uint8_t mServiceData[ 255 ];
   ServerConfig mServers[ 62 ];
   // ...
} 
``` 
Which causes the structure to have like 16 kilobytes. Since network data is 255 bytes max, Service TLV has minimum of 4 bytes of header, it is at most of 251 bytes per all Server TLVs, each having at least 4 bytes. So, it is not possible to have more than 62 Server TLVs. Otherwise, the whole structure would be much longer.

When used as a local network data, it is not a problem, we deal with max 1 server (current) at a time. When iterating through network data from leader, the API can't use structs like the above.

jwhui, 2017-08-26:
@Nopik 

1. Regarding the `S_id` field, I agree that the Thread Specification is underspecified here.  I would assume `S_id` field is ignored in a `SRV_DATA.ntf` message.

2. I agree that translating the nested structure of Service and Server TLVs into static structures is undesirable.  When iterating through the network data from the leader, I would have each iteration through the Server Set return a single (Service, Server) tuple - similar to what we do with the Valid Prefix Set (single Prefix, Border Router tuple) and External Route Set (single Prefix, Has Route tuple).

Nopik, 2017-10-11:
@jwhui I added PR #2256 to showcase my current progress and hopefully get some feedback. The work there is not finished yet. It currently only handles managing local network data and sending it to leader. Next steps are to handle the SRV TLVs on leader and then use them in mesh forwarder.

Please let me know if I'm going in the right direction.

jwhui, 2017-10-26:
Resolved by #2256.

closed 2017-10-26 by jwhui