openthread/openthread #2136: Joiner does not give correct priority to available networks

CiaranWoodward, 2017-08-25:
Currently, the joiner (joiner.cpp) gives priority to networks on higher channels. This has the effect that a joiner can accidentally fail to join the correct network if there is another network on a higher channel which is listing 'accept any' steering data. The Thread spec 1.1.1 section 8.4.4.1.2 specifies a priority system for choosing a network to try and join.

I will shortly submit a pull request to make sure that the highest priority network is selected for joining rather than the highest channel.

I'm not sure how to implement the queue for networks that is described in a way that fits with the rest of openthread. Statically allocating a space in memory seems messy and wasteful given that joining is rare and only when not any other thread role. Message allocation is obviosly not intended for something like this. Any advice?

jwhui, 2017-08-25:
Thanks for raising this issue and working to address it.

As for handling multiple responses, you point out the right issues.  One nit is that the Thread spec does not specify how many responses a device must be able to support.  A max queue length of 1 is still a valid.  That said, OT could provide a compile-time configurable option the number of responses to support.  If you want to pursue this path, static allocation seems best.  In the future, we could better segment OT's "pre-commissioned" and "post-commissioned" memory and have them share the same memory region.

One other nit on the Thread spec is its use of the term "exact match" - Bloom filters are probabilistic data structures and do not provide "exact matches".

jwhui, 2017-09-07:
Resolved by #2152.

closed 2017-09-07 by jwhui