openthread/openthread #2138: Cannot build for Silabs Ef3r2 device

carloscal, 2017-08-25:
I'm getting some build casting warnings (treated as errors) when building for the Silabs Efr32 device, I'm using the lastest build (Aug 25). Simplicity installed Flex 1.2.1.0.

The file in question is the /examples/platforms/efr32/radio.c, the fix is to explicity cast the to the correct pointer type.

This is the error in question:
=====
/opt/openthread-repo/openthread/examples/../examples/platforms/efr32/radio.c: In function 'otPlatRadioSetExtendedAddress':
/opt/openthread-repo/openthread/examples/../examples/platforms/efr32/radio.c:212:36: error: passing argument 1 of 'RAIL_IEEE802154_SetLongAddress' discards 'const' qualifier from pointer target type [-Werror=discarded-qualifiers]
RAIL_IEEE802154_SetLongAddress(aAddress->m8);
^~~~~~~~
In file included from /opt/openthread-repo/openthread/examples/../examples/platforms/efr32/radio.c:50:0:
/opt/openthread-repo/openthread/examples/../third_party/silabs/gecko_sdk_suite/v1.1/platform/radio/rail_lib/protocol/ieee802154/rail_ieee802154.h:275:6: note: expected 'uint8_t * {aka unsigned char *}' but argument is of type 'const uint8_t * {aka const unsigned char *}'
bool RAIL_IEEE802154_SetLongAddress(uint8_t *longAddr);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/openthread-repo/openthread/examples/../examples/platforms/efr32/radio.c: In function 'RAILCb_IEEE802154_DataRequestCommand':
/opt/openthread-repo/openthread/examples/../examples/platforms/efr32/radio.c:616:35: error: passing argument 1 of 'findSrcMatchExtEntry' from incompatible pointer type [-Werror=incompatible-pointer-types]
findSrcMatchExtEntry(aAddress->longAddress) >= 0) ||
^~~~~~~~
/opt/openthread-repo/openthread/examples/../examples/platforms/efr32/radio.c:436:8: note: expected 'const otExtAddress * {aka const struct otExtAddress *}' but argument is of type 'uint8_t * {aka unsigned char *}'
int8_t findSrcMatchExtEntry(const otExtAddress *aExtAddress)
^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:820: recipe for target 'libopenthread_efr32_a-radio.o' failed
make[5]: *** [libopenthread_efr32_a-radio.o] Error 1
make[5]: *** Waiting for unfinished jobs....
Makefile:502: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
Makefile:479: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:640: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:566: recipe for target 'all' failed
make[1]: *** [all] Error 2
examples/Makefile-efr32:240: recipe for target 'do-build-cortex-m4' failed
make: *** [do-build-cortex-m4] Error 2

xiaom-GitHub, 2017-08-26:
Thanks for notifying this compile error. I will submit a PR to address this soon.

Xiao

xiaom-GitHub, 2017-08-28:
Please have a review for #2143 . Thanks.

PR #2143 (2017-08-29): [efr32] fix build casting warnings on pointer type
Fix #2138 building efr32 platform error.
closed 2017-08-29 by jwhui