openthread/openthread #2156: Confusion with Joining Device Credential vs PSKd and their usage

GiedriusM, 2017-09-01:
As far as I understand, the joiner-commissioner DTLS session should be established using base32-thread encoded string (Joining Device Credential), however in the Thread 1.1.1 specification there's a small distinction of Joiner Device Credential vs PSKd: 
`This credential when specifically encoded in binary is referred to as the PSKd (Pre-Shared Key for the Device).`
and `otJoinerStart` has a param  ` aPSKd             A pointer to the PSKd.`.

My questions are:
1) Is the PKSd in otJoinerStart (also in otCommissionerAddJoiner) actually a Joining Device Credential as specified in Thread 1.1.1?
2) Is there/what is an actual usage of [binary] PSKd?

turon, 2017-09-01:
Hi Giedrius,

The Joiner Device Credential and PSKd are essentially the same thing, the
only difference is the encoding.

The PSKd is in raw binary form, and is what the underlying OpenThread code
uses.

The specification section on base32-thread encoding of the Joiner Device
Credential is a presentation layer advisory.  The PSKd needs to be
presented to the user in a standard way across the entire user experience:

   - on the device label
   - when typed into a phone
   - within the QR code

If one vendor renders the PSKd in ASCII HEX, another in base-64, and a
third in base32-thread, there would be interoperability problems.  For
example, the standard Commissioning app wouldn't know how to convert what
the user types in into binary without extra information.

Martin

_____________________________
Martin Turon  |  Nest Labs

jwhui, 2017-09-01:
@GiedriusM, let me answer your questions directly (your original post indicates you already fully understand the differences between the Joiner Device Credential and PSKd).

> 1. Is the PKSd in otJoinerStart (also in otCommissionerAddJoiner) actually a Joining Device Credential as specified in Thread 1.1.1?

`otJoinerStart()` and `otCommissionerAddJoiner()` take the PSKd (i.e. the binary form).

> 2. Is there/what is an actual usage of [binary] PSKd?

The short answer is that the Thread Certification Harness also passes the PSKd rather than the Joiner Device Credential, so the API needs to support this use case until the Harness is updated.

I think it would be great for OpenThread to provide a utility function for converting between the Joiner Device Credential and PSKd.  Would you like to contribute one?

GiedriusM, 2017-09-01:
> otJoinerStart() and otCommissionerAddJoiner() take the PSKd (i.e. the binary form).

I was pretty sure that these functions take Joining Device Credential (JDC). There are few reasons for that:
1) Thread 1.1.1 spec, section 8.4.4.5 mentions JDC instead of PSKd 
```The Commissioner, upon reception of a DTLS-ClientHello message with a cookie from one of its scanned Joiners, inspects the Joiner IID, and looks up the Joining Device Credential to use to continue the DTLS handshake for that Joiner.```
2) `otJoinerStart` and all other related functions use `const char *aPKSd` type with no length indication (`strlen(aPSKd)` is used), which does not allow a null-character inside of a PKSd.
3) JDC has a maximum length of 32 symbols and given the base32 (2^5), the maximum binary (base-256) length would be 32*5/8=160bits or 20 bytes. However PSKd in OpenThread is `kPskMaxLength = 32`.

This begs further questions:
1) Should the OpenThread's joiner API take JDC string (and convert it internally) or a pointer to PSKd byte array and it's length as the parameters?
2) Is the Thread spec misleading and the maximum PSKd length is 32 bytes and 32*8/5=51.2 bytes for the JDC.
3) Is there a base32-thread encoding format reference? For example, what is the padding symbol? Or is the PSKd length limited to multiples of 40 bits, so that it could be expressed with whole-bytes in both base-32 and base-256 (binary) formats.

jwhui, 2017-09-01:
@GiedriusM, the existing OpenThread APIs should be changed to taking byte-array rather than char-array to avoid the confusion you are raising.  FWIW, the char-array is passed directly as a byte-array to [`mbedtls_ssl_set_hs_ecjpake_password()`](https://github.com/openthread/openthread/blob/master/third_party/mbedtls/repo.patched/include/mbedtls/ssl.h#L1892), which takes the EC J-PAKE password as binary.

> 1. Should the OpenThread's joiner API take JDC string (and convert it internally) or a pointer to PSKd byte array and it's length as the parameters?

OpenThread's API should take the PSKd - the Thread Certification Harness requires it.  I think OpenThread could provide a utility function that converts from base32-thread to PSKd.

> 2. Is the Thread spec misleading and the maximum PSKd length is 32 bytes and 32*8/5=51.2 bytes for the JDC.

The Thread Spec may be inconsistent.  But I believe the real limit is that the base32-thread encoding is limited to 32 characters.  As a result, the PSKd maximum length should be 20 bytes.

> 3. Is there a base32-thread encoding format reference? For example, what is the padding symbol? Or is the PSKd length limited to multiples of 40 bits, so that it could be expressed with whole-bytes in both base-32 and base-256 (binary) formats.

I'm not aware of a reference, but I agree it would be useful to have.  I also agree that padding needs to be addressed.  We should submit a Thread Group JIRA to address this gap.

GiedriusM, 2017-09-12:
@jwhui  Are there any news about base32-thread spec? I've submitted an early PR for this #2185

jwhui, 2017-10-12:
@GiedriusM, given the recent discussions, I think it may make sense to change the API and CLI surface to accept the Joiner Credential (i.e. base32-encoded string).

Regarding the Thread Test Harness, the harness is indeed passing the PSKd directly, and this is something we need to resolve with the Thread Group.

Thoughts? (I know this causes extra work for you).

GiedriusM, 2017-10-12:
Then I guess it's not worth merging the current PR, but do a fresh one once the details are sorted.

jwhui, 2017-10-12:
@GiedriusM, I would suggest moving forward with making the change to Joiner Device Credential now.  We'll resolve the Thread Test Harness discrepancy separately (and provide an interim workaround to the Thread Test Harness, if needed).

GiedriusM, 2017-10-12:
@jwhui, could you expand on what you mean by "JDC change"? As I understand, decision is to have JDC string passed inside of the API, instead of binary PSKd, which is what is currently done in OpenThread.

jwhui, 2017-10-12:
@GiedriusM, couple necessary items:
1. Change argument/variable names to clearly indicate when JDC is being used.
2. Integrate base32 conversion of JDC to PSKd where necessary.

jwhui, 2017-10-13:
@GiedriusM, on further thought, it might be best to hold off on any changes for now.  We'll try to get some quick feedback from the Thread Group on this issue.  Thanks again for raising this issue.

GiedriusM, 2017-10-13:
Alright, let me know when there's any news. I would also suggest reconsider using binary PSKd representation internally, instead of JDC. For one, it makes more sense to work with an actual value rather than a representation of it, as less [core] code is needed to verify it's validity and it leaves a slightly lower program and RAM memory footprint. It is also simpler when PSKd is stored inside EEPROM or is displayed in other formats such as hex, QR or whatever is the most convenient for the manufacturer - that way at most only one decode operation is necessary.

jwhui, 2017-11-27:
@GiedriusM, just to follow up on this, I have filed [SPEC-742](https://threadgroup.atlassian.net/browse/SPEC-742).  There seems to be agreement within the Thread Group that the JDC itself should be used as the J-PAKE shared secret (i.e. the string itself and no base32 conversion).

GiedriusM, 2017-11-27:
Thanks, @jwhui! So I assume that the answer to my first question is "yes"? If so, then the second one becomes kind of unanswered :grin:

I take it, that at the time of writing the whole binary/encoded representation was not fully conceptualized and later the ASCII variant became "de facto". Either way the cited section could be rephrased and if it is decided that JDC is the PSKd, then one or the other term could be dropped altogether. IMO it would make more sense and be less confusing.

Also, just as a clarification, does this means that the shared secret (PSKd/JDC) can only contain the base32-thread symbols? What about lower/upper case? On top of that, there comes another issue - OpenThread should validate the passed string contents... :smirk:

jwhui, 2019-06-25:
This has been addressed in Thread 1.2 with [SPEC-742](https://threadgroup.atlassian.net/browse/SPEC-742).

closed 2019-06-25 by jwhui