openthread/openthread #2164: [netdata] issues when multiple BRs are configured the same prefixes (or routes) with different stable flag

librasungirl, 2017-09-03:
Multiple BRs MAY be configured with the same prefixes (or routes) of different stable flag, different priorities etc. That is, in theory, one prefix may include both stable and unstable borderroutertlv, each of which may also include multiple borderrouterentries.

Current `GetNextOnMeshPrefix()` would always get the BRs in the first `BorderRouterTlv` of the Prefix.  This would also impact `IsOnMeshPrefixConsistent() ` and may cause repeating SRV_DATA.ntf. Similar impact on external routes.

abtink, 2017-09-05:
With the change from https://github.com/openthread/openthread/pull/2073, the two methods `GetNextOnMeshPrefix()` and `GetNextExternalRoute()` should now iterate over all entries.

librasungirl, 2017-09-06:
@abtink, I think #2073 could iterate all entries in the **first** `BorderRouterTlv` of one Prefix. However as stated above, one `PrefixTlv` may have two `BorderRouterTlv` (one stable, and one unstable).  Existing code does try to iterate the second `BorderRouterTlv`.

abtink, 2017-09-06:
Ah, I see... Got it. Thank you.

abtink, 2017-09-11:
PR: https://github.com/openthread/openthread/pull/2183

closed 2017-09-14 by abtink