openthread/openthread #2176: MbedTls failure during commissioning on slow platforms.

erja-gp, 2017-09-07:
We're seeing issues when running the OpenThread stack on an older platform, where the crypto computations are done on the CPU and are not fast enough to avoid DTLS timeouts on the other side.

The issues are similar to, but not the same as, what is reported in: https://github.com/openthread/openthread/pull/1207

In our case the code (including the fix "0001-Fix-commissioning-problem-with-retransmissions.patch") behaves as follows: 

Note that in my setup the Joiner is the slow platform which takes more than 8 seconds to perform it’s computations, whereas the Commissioner is on a faster platform (therefor there are no repeats of the 'Client Hello' messages).

```
   Joiner                               Commissioner
(1)  |              Client Hello              |
     |--------------------------------------->|
     |                                        |
(2)  |          Hello Verify Request          |
     |<---------------------------------------|
     |                                        |
(3)  |         Client Hello + cookie          |
     |--------------------------------------->|
     |                                        |
(4)  |              Server Hello              |
     |          Server Key Exchange           |
     |           Server Hello Done            |
     |<---------------------------------------|
     |                                        | |
(5)  |              Server Hello              | |  DTLS timeout
     |          Server Key Exchange           | |
     |           Server Hello Done            | v
     |<---------------------------------------| 
     |                                        |
(6)  |          Client Key Exchange           |
     |           Change Cipher Spec           |
     |      Encrypted Handshake Message       |
     |        Client Key Exchange (DUP)       |
     |        Change Cipher Spec (DUP)        |
     |   Encrypted Handshake Message (DUP)    |
     |--------------------------------------->|
     |                                        |
(7)  |           Change Cipher Spec           |
     |      Encrypted Handshake Message       |
     |<---------------------------------------|
     |                                        |
(8)  |            Application Data            |
     |--------------------------------------->|
     |                                        |
(9)  |        Change Cipher Spec (DUP)        |
     |   Encrypted Handshake Message (DUP)    |
     |         Application Data (DUP)         |
     |<---------------------------------------|
     |                                        |

```

Because the Joiner takes a long time to compute the data for (6), one sees that on the Commissioner side the 8s timeout gets triggered, and the previous message (4) is retransmitted as (5).

After creating the data for (6) the TLS code notices that there are repeat messages,
and decides (as per spec) to reply to both (4) and (5). In our case both replies end up in the same message (6) !

Now, the Commissioner replies with (7), but only removes the first half of (6) out of its rx queue.

When the Commissioner receives (8), it notices the remaining part of (6) is still in its rx queue, and replies to this first with the "Change Cipher Spec (DUP)". Additionally,  it also replies to the recently received Application Data (8). Both replies end up in (9).

At this point, the Joiner code has moved to a state where it's looking for "Application Data", but there is unexpected data (Change Cipher Spec (DUP)) in front of it.
The current DTLS code doesn't handle this situation: it sees an incorrect record and discards the complete UDP packet (and not only the offending record).

That's the current situation. It is clear that the current patch does only fixes a single variation of the problem (again, see PR1207), and not the bigger problem of the DTLS state machine ending up in an incorrect state.

There are multiple paths to fix the underlying problem:
* one could adapt the code to also handle the "Change Cipher Spec" by trplying to it (I've tried it and it works). But this requires again multiple fixes, and I don't think this is the correct way forward: in fact it's like handling multiple parallel handshakes. The sniffer logs end up looking very messy with all those (unnecessary) retransmission. Also, it spins out of control, with multiple retransmissions on both sides when two older platforms start a DTLS session between one another.

* we should reconsider sending the initial retransmit of the "Client Key Exchange".
The spec clearly states it should be done:

>     "2. The implementation reads a retransmitted flight from the peer: the
>       implementation transitions to the SENDING state, where it
>       retransmits the flight, resets the retransmit timer, and returns
>       to the WAITING state.  The rationale here is that the receipt of a
>       duplicate message is the likely result of timer expiry on the peer
>       and therefore suggests that part of one's previous flight was
>       lost."
> 

But, it also mentions that the reasoning behind this rule is to retransmit a flight that was previously lost. In the outlined case there is no previous flight. When one runs a slow platform, it's obvious that the retransmits are NOT caused by a lost previous flight, so why send multiple replies in that case ?

The code that triggers this type of retransmit is located in third_party/mbedtls/repo/library/ssl_tls.c:

```
int mbedtls_ssl_prepare_handshake_record( mbedtls_ssl_context *ssl ) {
...
            if( recv_msg_seq == ssl->handshake->in_flight_start_seq - 1 &&
                ssl->in_msg[0] != MBEDTLS_SSL_HS_HELLO_VERIFY_REQUEST )

```
To still benefit from the speed improvements, it is probably not required to entirely remove this retransmission mechanism. Only remove the retransmissions 
* @ compile time: only for slow platforms
* @ run time: only for the state(s) with known long computational delays.

Note: disabling this code will not create issues in case were a message really gets lost, because there is always the local timer which will timeout when no reply is received. This, in turn will (locally) trigger a retransmit.

hubertmis, 2017-09-08:
I would not completely disable retransmission mechanism even if following set of conditions is met:
- slow platform,
- state with known delay,
- retransmission is caused by retransmitted message from the peer.
I think it would be spec violation. RFC requires to retransmit last flight if previous flight from the peer is received.

I agree we don't have to retransmit a flight multiple times in case there is a few received flights in the buffer. Moreover, mbedtls should process Application Data that is embedded in the handshake message.

BTW There are a few PRs in the mbedtls repository referenced from #1207: https://github.com/ARMmbed/mbedtls/pull/942 https://github.com/ARMmbed/mbedtls/pull/946 . Perhaps it addresses our problems.

erja-gp, 2017-09-08:
I understand your doubt, but perhaps this scenario was not taken into account while writing the DTLS RFC. I cannot believe that we should intentionally start duplicate transmissions.
Should I also create a ticket in the mbedtls group to raise that question there?

jwhui, 2017-09-14:
@erja-gp, thanks for raising this issue.  Given that this is an issue that is primarily dealt with in mbedTLS, I think it would be worthwhile to raise the issue within the mbedTLS community.

hanno-arm, 2017-10-11:
Hi everyone,

I have been working on a related fix https://github.com/ARMmbed/mbedtls/pull/946 for Mbed TLS and am trying to understand whether it might also resolve the present issue. From @erja-gp's description, it seems that the problem is that there's a datagram including a `ChangeCipherSpec` followed by application data, and which gets completely dropped because the unexpected `ChangeCipherSpec` is treated as an invalid record leading to entire datagram dismissal. 

Is that understanding correct?

If yes, then this should hopefully be fixed with the PR. The problem is that the previous code checks the epoch-depending length constraints for records (leading to entire datagram dismissal when violated) _before_ checking if the message actually belongs to the current epoch (leading to record-only dismissal if not). The first check isn't sensible before knowing that the epoch is the expected one, and that's the problem for the `ChangeCipherSpec`: With respect to the newly negotiated epoch - to which however it doesn't belong - it appears to be too short to be a proper record, leading to an entire datagram dismissal. Instead, the right way would be to silently drop it because it's from an old epoch, and go on with the rest of the datagram. See https://github.com/ARMmbed/mbedtls/pull/946/commits/52c6dc64c675d24b1be65e06d43dd67dc111e762. 

Let me know if the PR solves your issue or if I misunderstood your description.
Thanks,
Hanno

hubertmis, 2017-10-11:
@hanno-arm, thanks for your input. It looks that this commit should fix one of two reported problems. I'll try to apply it and check on nRF52 platform.

The second problem is connected with slow ECC calculations. Some platforms calculate ECC in software so slow, that peer times out and retransmits last flight (check sequence diagram in #1207). This retransmission is stored in rx buffer and it is not processed until ECC calculation ends. In this case the node, that was performing ECC calculation, sends duplicated response (the first one is freshly calculated, the second one is triggered by retransmission in receive buffer).
It would be great if mbedtls could send only one response in case there are is a few retransmissions of last flight in rx buffer.

erja-gp, 2017-10-11:
I will try to verify it before the end of this week.

hanno-arm, 2017-10-11:
@hubertmis I agree that this leads to unnecessary network load, but does it prevent the handshake from succeeding? 
The only time-agnostic remedy I could conceive on the level of Mbed TLS is to never issue multiple retransmissions in response to what the library finds within a single datagram, but this wouldn't solve the problem when multiple datagrams have accumulated during a longer calculation. 
In that regard, I also wanted to ask you and @erja-gp if your code performs some merging of datagrams in the receive buffer? Otherwise, I don't see how it could happen (using Mbed TLS) that a retransmitted `ChangeCipherSpec` and an `ApplicationData` message share the same datagram.

hubertmis, 2017-10-11:
@hanno-arm I've never seen a problem with successful handshake on nRF52 caused by duplicated responses. Only unnecessary traffic. From my perspective preventing duplicated responses is 'nice to have' but not priority.

hubertmis, 2017-10-11:
@hanno-arm I've performed smoke test with provided patch. I could not finish DTLS handshake. There was some assertion fault in OpenThread. I don't have time to debug it right now.

hanno-arm, 2017-10-11:
@hubertmis Thanks for giving it a try so quickly. Did you apply the commit only or the entire PR? It might be that the problem triggering #1207 is occurring here, too, in which case the entire PR would be needed. But of course it might be something entirely different, or a fault introduced by the commit - let me know once you had time to debug.

hubertmis, 2017-10-11:
@hanno-arm I've used patched version from OpenThread (fix for #1207 is present) and added your commit as another patch. It seems that after handshake first (or 2nd) `mbedtls_ssl_write()` returns `MBEDTLS_ERR_SSL_WANT_READ`. I didn't check why.

hanno-arm, 2017-10-11:
@hubertmis Ok, so the failure happens after `mbedtls_ssl_handshake` has completed? Is it happening for the joiner or the commissioner? Is the scenario you tested the one described in this issue or something different?

hubertmis, 2017-10-11:
@hanno-arm It seems that handshake is completed. I can see first Application Data sent by Joiner, followed by `Change Cipher Spec, Encrypted Handshake, Change Cipher Spec, Encrypted Handshake` combo sent by Commissioner. After this message Joiner reports `MBEDTLS_ERR_SSL_WANT_READ`

erja-gp, 2017-10-11:
This sounds very much like the original problem which I described in https://github.com/openthread/openthread/issues/2176.

Each retransmitted frame (due to a slow device) causes its own separate answer. This means later on in the flow, a frame not related to the current state of the state machine is received and this causes the DTLS code to return errors (the state machine got confused). The 'duplicate' answers not only cause overhead, they make the state machine much more complicated.

In my opinion filtering the duplicate frames, will result in a simpler, easier to maintain solution.

hanno-arm, 2017-10-11:
@erja-gp @hubertmis Could you point me to the code where you're performing the handshake and sending the application data afterwards, so that I can have a closer look?
Also, is it possible for you to give the whole PR a try and see if you still have problems? The use of `WANT_READ` is quite incompatible with event-driven IO in the current codebase. Note also that the PR will be in conflict with the fix for #1207 you've already merged, so that would need to be undone for the testing.

hubertmis, 2017-10-11:
@hanno-arm I believe handshake and application data is handled in `src/core/meshcop/dtls.cpp` file. `Start()` and `Send()` methods.
I've tried to perform commissioning procedure with mbedtls replaced with ARMmbed/mbedtls#946. The result is exactly the same. During sending application data Joiner stops in assertion `WANT_READ`.

hanno-arm, 2017-10-11:
@hubertmis Thank you. In general, note that you must be prepared to receive a `WANT_READ` or `WANT_WRITE`, even if you call the library in response to an incoming message event or for the purpose of writing. The reason is exactly what we have been discussing, namely that incoming datagrams might entirely consist of unexpected or stale records that are silently discarded. For example, if the Joiner receives the `Change Cipher Spec, Encrypted Handshake, Change Cipher Spec, Encrypted Handshake` combo after the handshake is completed, the subsequent call of `ssl_read` will return `WANT_READ`  because none of the received information is of any use anymore.
Concerning the concrete problem at hand, though, you seem to get the failure from `ssl_write`, which only performs read operation if the handshake is not yet complete - it would therefore be beneficial to understand first if `ssl_write` resp. `Dtls::Send` is indeed only called once the handshake completes. From looking at https://github.com/openthread/openthread/blob/master/src/core/meshcop/dtls.cpp#L141 and https://github.com/openthread/openthread/blob/master/src/core/meshcop/dtls.cpp#L414, I don't know which part of the code would guarantee that, as `Dtls::Process` and therefore also `Dtls::Start` would return without error if the handshake in `Dtls::Process` was paused with a `WANT_READ` or a `WANT_WRITE`. I'm new to your code and don't know where the `Dtls` class is instantiated for use by the joiner, and in particular, where `Dtls::Send` is invoked first after the handshake. Could you point me to that?

hubertmis, 2017-10-11:
@hanno-arm I'm not sure how `dtls.cpp` works.
@jwhui Could you help us with this problem?

jwhui, 2017-10-11:
@hanno-arm, thanks for working with us to address this issue.

Application data is only sent in response to calling `mbedtls_ssl_handshake()` and the ssl context `state` member is set to `MBEDTLS_SSL_HANDSHAKE_OVER` after the call.  The corresponding code is here: https://github.com/openthread/openthread/blob/master/src/core/meshcop/dtls.cpp#L402

hanno-arm, 2017-10-11:
@jwhui Thanks. So it looks like it is ensured that `Dtls::Send` is only called once `Dtls::Process` has passed https://github.com/openthread/openthread/blob/master/src/core/meshcop/dtls.cpp#L402 - that's very important, because if `Dtls::Send` was called beforehand, the subsequent invocation of `ssl_write` would would complete the handshake silently, and `mConnectedHandler` would be skipped this way.

Regarding the behavior you observe, I'm still puzzled, so please don't mind me recollecting my understanding to be entirely sure we're on the same page: 
1. Joiner sends `ChangeCipherSpec` flight as first part of (6) and awaits Commissioner's `ChangeCipherSpec` flight.
2. Joiner instead receives duplicated `ServerHello, ..., ServerHelloDone` flight from Commissioner, and responds by retransmitting his `ChangeCipherSpec` flight. 
3. Concurrently, Commissioner responds to each of these `ChangeCipherSpec` flights with his `ChangeCipherSpec` flight.
4. Joiner receives and processes the first of the two Commissioner's `ChangeCipherSpec` flights. After that, the handshake is complete, so `mConnectedHandler` is called, which as far as I see goes down to `Joiner::SendJoinerFinalize` being called (?). This in turn leads to `Dtls::Send` and therefore `ssl_write` being called, which finds the handshake completed and hence delivers the data without any attempt to read. So this first invocation of `Dtls::Send` and `ssl_write` should be fine - right?
5. What happens on the Joiner side afterwards? Does he idle on the underlying transport, or immediately attempt to send more data? From my understanding, the duplicated `ChangeCipherSpec` flight from the Commissioner should at this point still be pending in the Joiners receive buffers, but I can't conceive how they would lead to a `WANT_READ` in response to a `ssl_write`. Any subsequent `ssl_read` would lead to a `WANT_READ`, though, because these messages will be discarded, but as far as I understand it's a write-call that runs into the `WANT_READ` assertion that @hubertmis mentioned.

The above is not entirely in sync with @hubertmis's description, which mentioned _two_ `ChangeCipherSpec` flights _after_ the Joiner's `ApplicationData` message; perhaps that's what happens if things are so slow that the Commissioner retransmits his `ServerHello` flight even twice?

hubertmis, 2017-10-12:
@hanno-arm When I disable all hardware acceleration I get following handshake exchange that leads to the assertion fault:
```
    Joiner                        Commissioner
(1) |      Client Hello           |
    |---------------------------->|
    |                             |
(2) |      Hello Verify Request   |
    |<----------------------------|
    |                             |
(3) |      Client Hello           |
    |---------------------------->|
8 s |                             |
(4) |      Client Hello           |  retransmission
    |---------------------------->|
    |                             |
(5) |      Server Hello           |
    |      Server Key Exchange    |
    |      Server Hello Done      |
    |<----------------------------|
    |                             |
(6) |      Server Hello           |  response to retransmission
    |      Server Key Exchange    |
    |      Server Hello Done      |
    |<----------------------------|
8 s |                             |
(7) |      Server Hello           |  retransmission
    |      Server Key Exchange    |
    |      Server Hello Done      |
    |<----------------------------|
    |                             |
(8) |      Client Key Exchange    |
    |      Change Cipher Spec     |
    | Encrypted Handshake Message |
    |      Client Key Exchange    |
    |      Change Cipher Spec     |
    | Encrypted Handshake Message |
    |      Client Key Exchange    |
    |      Change Cipher Spec     |
    | Encrypted Handshake Message |
    |---------------------------->|
    |                             |
(9) |      Change Cipher Spec     |
    | Encrypted Handshake Message |
    |<----------------------------|
    |                             |
(10)|    Application Data (CoAP)  |
    |---------------------------->|
    |                             |
(11)|      Change Cipher Spec     |
    | Encrypted Handshake Message |
    |      Change Cipher Spec     |
    | Encrypted Handshake Message |
    |      Application Data       |
    |<----------------------------|
    |                             |
(12)|    Application Data (CoAP)  |  retransmission
    |
    | Joiner assertion fault on mbedtls_ssl_write(CoAP data): WANT_READ
```

Check backtrace that reports WANT_READ error code:
```
#0  mbedtls_ssl_fetch_input (ssl=ssl@entry=0x200070a0 <sInstanceRaw+24136>, nb_want=<optimized out>) at /home/humi/git/openthread_github/examples/../third_party/mbedtls/repo.patched/library/ssl_tls.c:2360
#1  0x000247a4 in mbedtls_ssl_read_record_layer (ssl=ssl@entry=0x200070a0 <sInstanceRaw+24136>) at /home/humi/git/openthread_github/examples/../third_party/mbedtls/repo.patched/library/ssl_tls.c:3883
#2  0x00024e06 in mbedtls_ssl_read_record (ssl=ssl@entry=0x200070a0 <sInstanceRaw+24136>) at /home/humi/git/openthread_github/examples/../third_party/mbedtls/repo.patched/library/ssl_tls.c:3739
#3  0x00029480 in ssl_parse_server_hello (ssl=0x200070a0 <sInstanceRaw+24136>) at /home/humi/git/openthread_github/examples/../third_party/mbedtls/repo.patched/library/ssl_cli.c:1451
#4  mbedtls_ssl_handshake_client_step (ssl=ssl@entry=0x200070a0 <sInstanceRaw+24136>) at /home/humi/git/openthread_github/examples/../third_party/mbedtls/repo.patched/library/ssl_cli.c:3408
#5  0x00022d92 in mbedtls_ssl_handshake_step (ssl=ssl@entry=0x200070a0 <sInstanceRaw+24136>) at /home/humi/git/openthread_github/examples/../third_party/mbedtls/repo.patched/library/ssl_tls.c:6583
#6  0x00022de2 in mbedtls_ssl_handshake (ssl=0x200070a0 <sInstanceRaw+24136>) at /home/humi/git/openthread_github/examples/../third_party/mbedtls/repo.patched/library/ssl_tls.c:6607
#7  0x00024624 in mbedtls_ssl_write (ssl=ssl@entry=0x200070a0 <sInstanceRaw+24136>, buf=buf@entry=0x2003fea8 "@\002퀱c\002jf\377\020\001\001!\nOPENTHREAD\"\bNRF52840#\017g6933a1c6-dirty%\006\030\264\060", 
    len=len@entry=60) at /home/humi/git/openthread_github/examples/../third_party/mbedtls/repo.patched/library/ssl_tls.c:7215
#8  0x0000a6f8 in ot::MeshCoP::Dtls::Send (this=0x20006c68 <sInstanceRaw+23056>, aMessage=..., aLength=<optimized out>) at /home/humi/git/openthread_github/examples/../src/core/meshcop/dtls.cpp:220
#9  0x0001bf18 in ot::Coap::Coap::SendCopy (this=this@entry=0x20007210 <sInstanceRaw+24504>, aMessage=..., aMessageInfo=...) at /home/humi/git/openthread_github/examples/../src/core/coap/coap.cpp:473
```

hanno-arm, 2017-10-12:
@hubertmis This is very helpful. So it is indeed the case that (at least) the second `Dtls::Send` is called when a handshake is in progress (either the initial or a new one), and the return of `WANT_READ` is to be expected and desired in this case.
Still, after our discussion above it is unclear to me at moment how it can happen that `Dtls::Send` is called while a handshake is in progress. @hubertmis Is it possible to obtain an entire debug trace of Mbed TLS for the joiner? The assertion backtrace you provided indicates that a fresh handshake is being started at some point (because of `client_step` calling `ssl_parse_server_hello`), and it would be good to understand when exactly that happens and why.

hubertmis, 2017-10-12:
@hanno-arm , I've enabled logging to get some information.
Here is log of processing flight (11): https://pastebin.com/bQ1uEiM2
It seems that problem is in line `[0000080258] -MESH-CP-: bad application data message`. It is followed by error `MBEDTLS_ERR_SSL_UNEXPECTED_MESSAGE` that closes connection in https://github.com/openthread/openthread/blob/master/src/core/meshcop/dtls.cpp#L450. Because of that following write start handshake and leads to assertion fault.

hanno-arm, 2017-10-12:
@hubertmis Is renegotiation disabled for in your Mbed TLS build?

hubertmis, 2017-10-12:
@hanno-arm I've found following test in `mbedtls-config.h`:
```
/**
 * (...)
 *
 * Comment this to disable support for renegotiation.
 */
//#define MBEDTLS_SSL_RENEGOTIATIO
```
Because of that I think renegotiation is disabled. Can I verify it run-time?

hanno-arm, 2017-10-12:
@hubertmis Ok, great, we've found the issue. It's a bug in Mbed TLS that has already been fixed but not yet merged: As you can see looking at [this part of `ssl_read`](https://github.com/ARMmbed/mbedtls/blob/development/library/ssl_tls.c#L6881), the code-path handling handshake when expecting application data is only included if `MBEDTLS_SSL_RENEGOTIATION` is set, and otherwise handshake messages are always treated as fatal. This is wrong - they must always be considered, and only the path which [accepts renegotiation](https://github.com/ARMmbed/mbedtls/blob/development/library/ssl_tls.c#L6966) should be conditional on `MBEDTLS_SSL_RENEGOTIATION`. I will check our internal ticket for that and see if I can push it.

hanno-arm, 2017-10-12:
@hubertmis Could you apply [this patch](https://github.com/mpg/mbedtls/commit/2694afbb56639f2135f1dd48b3ef903df6ac7ecd) and retest?

hubertmis, 2017-10-12:
@hanno-arm Tested. Works perfectly on nRF52840 :)

hanno-arm, 2017-10-12:
@hubertmis That's great news. So let me summarize the outcome of this:
1. PR https://github.com/ARMmbed/mbedtls/pull/946 fixes that sometimes datagrams are unrightfully dropped when they contain a stale `ChangeCipherSpec`. That's what initially triggered this issue.
2. Patch https://github.com/mpg/mbedtls/commit/2694afbb56639f2135f1dd48b3ef903df6ac7ecd fixes that handshake messages are treated as fatal in `ssl_read` if `MBEDTLS_SSL_RENEGOTIATION` is disabled, a bug that makes the scenario in this issue continue to fail even after (1) has been fixed. I will follow up on this and give you a note once a decision for the proper long-term solution has been made and a PR has potentially been opened. Please note that the patch shouldn't yet be used in production code as it hasn't undergone thorough study and testing.
3. As a general comment, `WANT_READ` must be expected even when calling `mbedtls_ssl_read` in response to an incoming message. It must also be expected in response to `mbedtls_ssl_write` _provided renegotiation is used_. As the latter doesn't seem to be the case for your code, at the moment it seems that asserting `res != WANT_READ` should be fine.

hanno-arm, 2017-10-12:
@hubertmis Added a slightly different patch as part of [this PR](https://github.com/ARMmbed/mbedtls/pull/1141). If you could give that a test run in conjunction with either https://github.com/ARMmbed/mbedtls/commit/52c6dc64c675d24b1be65e06d43dd67dc111e762 or https://github.com/ARMmbed/mbedtls/pull/946, that would be great.

hubertmis, 2017-10-13:
@hanno-arm I've just tried https://github.com/ARMmbed/mbedtls/pull/1141 with ARMmbed/mbedtls@52c6dc6 patch. It works correclty.
Also tried to apply https://github.com/ARMmbed/mbedtls/pull/1141 as patch on top of ARMmbed/mbedtls#946, but patching failed and I did not perform this test.

hanno-arm, 2017-10-13:
@hubertmis Ok, the merge conflicts will be resolved eventually. Knowing that [#1141](https://github.com/ARMmbed/mbedtls/pull/1141) together with https://github.com/ARMmbed/mbedtls/commit/52c6dc6 works fine provides enough confidence to go forward with it for now. Thanks again for all the input and the report. Let me know if you encounter any other problems.

erja-gp, 2017-10-13:
@hanno-arm I've also tests ARMmbed/mbedtls#1141 with ARMmbed/mbedtls@52c6dc6 patch and it also is working on my platform.

hubertmis, 2017-10-13:
@hanno-arm Do you plan a release that includes  ARMmbed/mbedtls#946 and ARMmbed/mbedtls#1141? When can we expected it?

hanno-arm, 2017-10-13:
@hubertmis I'm sorry, but that's nothing I decide or can safely comment on.

erja-gp, 2017-10-17:
@hanno-arm A slightly different question: when do you think the above fixes will me merged to the main branch ?

hanno-arm, 2017-10-17:
@erja-gp I hope and try to get them reviewed and merged soon, but I cannot make any definite statements - I'm sorry.

jwhui, 2017-12-04:
Status update:
- https://github.com/ARMmbed/mbedtls/pull/946 appears close to merging.
- https://github.com/ARMmbed/mbedtls/pull/1141 has been merged.

hanno-arm, 2018-04-05:
Update: https://github.com/ARMmbed/mbedtls/pull/946 has been merged and will be included in the next release of Mbed TLS.

thcu-gp, 2018-04-17:
It would be great to have this next version of mbedtls in OpenThread

jwhui, 2018-04-17:
Yes, we'll sync with the next release of mbedtls when that is available.

thcu-gp, 2018-05-28:
just wanted to mention there is a new tagged version 2.9.0 of mbedtls
https://github.com/ARMmbed/mbedtls/tree/mbedtls-2.9.0

jwhui, 2018-05-31:
Submitted #2756 for upgrade to v2.9.0.  Please review.

PR #2756 (2018-06-08): [mbedtls] upgrade to 2.9.0
Resolves #2176.

~~The windows build currently fails due to a "potentially uninitialized local variable" warning.  The mbedtls project acknowledged this issue in: https://github.com/ARMmbed/mbedtls/issues/1662~~

Updated the mbedtls windows build project to not treat warnings as errors.
closed 2018-06-08 by jwhui