openthread/openthread #2191: Unstable test results for thread-cert test case 9-2-14 commissioner with TI-2538 devices.

bot-zjlu, 2017-09-13:
We run the test cases under '\tests\scripts\thread-cert' with 7 TI-2538 devices by adding some device manager and our own sniffer scripts.

Now all the test cases except 'Cert_9_2_14_PanIdQuery' when Commissioner as DuT are passed.  We stress-test the  test case of 'Cert_9_2_14_PanIdQuey_Commisisoner_As_DuT.py" many times and found about 60-70% changes passed and 30-40% failed.  

The error message is always:
"
017.09.13 03:45:01:INFO:BroadOT.Test> 1: channel
2017.09.13 03:45:01:INFO:BroadOT.Test> The TI device on port = COM17, current Channel= 11
2017.09.13 03:45:02:ERROR:BroadOT.Test> Traceback (most recent call last):
  File "C:\Python35\lib\site-packages\pexpect\expect.py", line 97, in expect_loop
    return self.timeout()
  File "C:\Python35\lib\site-packages\pexpect\expect.py", line 70, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: <pexpect_serial.serial_spawn.SerialSpawn object at 0x0000000004B46BA8>
searcher: searcher_re:
    0: re.compile("b'Conflict:'")

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\BroadOT_DRT\Code\BroadOT\tests\scripts\thread-cert\Cert_9_2_14_PanIdQuery.py", line 101, in test
    self.nodes[COMMISSIONER].panid_query(0xdead, 0xffffffff, 'ff33:0040:fdde:ad00:beef:0:0:1')
  File "C:\BroadOT_DRT\Code\BroadOT\tests\scripts\thread-cert\node.py", line 226, in panid_query
    self.interface.panid_query(panid, mask, ipaddr)
  File "C:\BroadOT_DRT\Code\BroadOT\tests\scripts\thread-cert\node_cli.py", line 425, in panid_query
    self.pexpect.expect('Conflict:', timeout=60)  #BRCM changed timeout=8 to timeout=60
  File "C:\Python35\lib\site-packages\pexpect\spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "C:\Python35\lib\site-packages\pexpect\spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "C:\Python35\lib\site-packages\pexpect\expect.py", line 107, in expect_loop
    return self.timeout(e)
  File "C:\Python35\lib\site-packages\pexpect\expect.py", line 70, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: <pexpect_serial.serial_spawn.SerialSpawn object at 0x0000000004B46BA8>
searcher: searcher_re:
    0: re.compile("b'Conflict:'")
<pexpect_serial.serial_spawn.SerialSpawn object at 0x0000000004B46BA8>
searcher: searcher_re:
    0: re.compile("b'Conflict:'")
"

From the sniffer log, we can find the differences between passed and failed logs.

Do you guys have similar experience  or  any clues?

Thanks,

zj

jwhui, 2017-09-14:
@bot-zjlu, I have not seen this kind of failure rate for this test scenario.

Are you running the `ot-cli-ftd` example image directly from [examples/apps/cli](https://github.com/openthread/openthread/tree/master/examples/apps/cli)?  Or are you running some custom code on the CC2538?

bot-zjlu, 2017-09-14:
Hi Jonathan,

Thank you for quick response!

We make the build with "Makefile-cc2538" and use the binary file
 'output/cc2538/bin/ot-cli-ftd'

thanks,

zj

jwhui, 2017-09-18:
@bot-zjlu, in your original posts, I see that two exceptions are occurring.  The first seems to be something related directly to the serial.  Do you know what `pexpect` is expecting that is causing the timeout?  If I'm understanding properly, the second exception seems to be a direct result of the first.

bot-zjlu, 2017-09-18:
@jwhui ,

Thanks Jonathan for your investigation and reply on this.

Do I know what the 'expect' command expects?   It expects 'Conflict:' from the serial port, but didn't get it, and the timeout exception has occurred.  Again, this failure happens about 30~40% rate on the CC2538 boards, but not always.

When I have moved to the latest OpenThread code (git hash: 782328a6c2955b441c4d256d72dad7ffd02fb9ee) on 09/13 for this test, I got new problems.  Could you help me on this new questions as well?
 
1.  Based on the "travis/scirpt.py" file, I have used the command as below to build the code for TI CC2538;
         > COMMISSIONER=1 JOINER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 DNS_CLIENT=1 make -f examples/Makefile-cc2538:

When I have downloaded the output image into CC-2538 board and tested the tests/scripts/thread-cert/Cert_9_2_14_PanIdQuery.py test script, it failed for the "macfilter addr clear" command. I have confirmed that there is no 'macfilter' CLI command included on this build from the command list. Why does the Makefile-cc2538 disable the 'macfilter' CLI command?

2. Based on my own investigation, I have modified two things to enable the macfilter as below; 
2-1. Added  "MAC_FILTER=1" in the build command, as below;
           > COMMISSIONER=1 JOINER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 DNS_CLIENT=1 make -f examples/Makefile-cc2538  clean all  MAC_FILTER=1
2-2. Added "--enable-mac-filter" on the "Makefile-cc2538", as below;
 configure_OPTIONS = \
    --enable-cli-app=all          \
    --enable-ncp-app=all          \
    --with-ncp-bus=uart           \
    --enable-diag                 \
    --with-examples=cc2538        \
    --enable-linker-map           \
    --enable-mac-filter           \
    $(NULL)
After these two changes, the test could run for the 'macfilter' command.  Am I doing properly on these changes to test "tests/scripts/thread-cert/Cert_9_2_14_PanIdQuery.py" test script?

3.  Even after these, still the "tests/scripts/thread-cert/Cert_9_2_14_PanIdQuery.py" test script fails on the CC-2538 boards, as shown below;   Is this failure expected one or should something more be modified or improved?

2017.09.15 14:28:35:INFO:BroadOT.Test> 1: commissioner panid 57005 4294967295 fdde:ad00:beef:0:0:ff:fe00:2800
2017.09.15 14:28:43:INFO:BroadOT.Test> 1: thread stop
2017.09.15 14:28:43:INFO:BroadOT.Test> 1: ifconfig down
2017.09.15 14:28:43:INFO:BroadOT.Test> 2: thread stop
2017.09.15 14:28:43:INFO:BroadOT.Test> 2: ifconfig down
2017.09.15 14:28:43:INFO:BroadOT.Test> 3: thread stop
2017.09.15 14:28:43:INFO:BroadOT.Test> 3: ifconfig down
2017.09.15 14:28:43:INFO:BroadOT.Test> 4: thread stop
2017.09.15 14:28:43:INFO:BroadOT.Test> 4: ifconfig down
2017.09.15 14:28:43:INFO:BroadOT.Test> 2: exit
2017.09.15 14:28:43:INFO:BroadOT.Test> factory reset: factoryreset
2017.09.15 14:28:44:INFO:BroadOT.Test> 2: channel
2017.09.15 14:28:44:INFO:BroadOT.Test> The TI device on port = COM19, current Channel= 11
2017.09.15 14:28:45:INFO:BroadOT.Test> 3: exit
2017.09.15 14:28:45:INFO:BroadOT.Test> factory reset: factoryreset
2017.09.15 14:28:45:INFO:BroadOT.Test> 3: channel
2017.09.15 14:28:45:INFO:BroadOT.Test> The TI device on port = COM15, current Channel= 11
2017.09.15 14:28:46:INFO:BroadOT.Test> 4: exit
2017.09.15 14:28:46:INFO:BroadOT.Test> factory reset: factoryreset
2017.09.15 14:28:46:INFO:BroadOT.Test> 4: channel
2017.09.15 14:28:47:INFO:BroadOT.Test> The TI device on port = COM21, current Channel= 11
2017.09.15 14:28:47:INFO:BroadOT.Test> 1: exit
2017.09.15 14:28:47:INFO:BroadOT.Test> factory reset: factoryreset
2017.09.15 14:28:48:INFO:BroadOT.Test> 1: channel
2017.09.15 14:28:48:INFO:BroadOT.Test> The TI device on port = COM17, current Channel= 11
2017.09.15 14:28:48:ERROR:BroadOT.Test> Traceback (most recent call last):
  File "C:\Python35\lib\site-packages\pexpect\expect.py", line 97, in expect_loop
    return self.timeout()
  File "C:\Python35\lib\site-packages\pexpect\expect.py", line 70, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: <pexpect_serial.serial_spawn.SerialSpawn object at 0x00000000043B2080>
searcher: searcher_re:
    0: re.compile("b'Conflict:'")

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\work\BroadOT_0825\BroadOT\tests\scripts\thread-cert\Cert_9_2_14_PanIdQuery.py", line 99, in test
    self.nodes[COMMISSIONER].panid_query(0xdead, 0xffffffff, ipaddr)
  File "C:\work\BroadOT_0825\BroadOT\tests\scripts\thread-cert\node.py", line 230, in panid_query
    self.interface.panid_query(panid, mask, ipaddr)
  File "C:\work\BroadOT_0825\BroadOT\tests\scripts\thread-cert\node_cli.py", line 495, in panid_query
    self.pexpect.expect('Conflict:', timeout=8)  #BRCM changed timeout=8 to timeout=60
  File "C:\Python35\lib\site-packages\pexpect\spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "C:\Python35\lib\site-packages\pexpect\spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "C:\Python35\lib\site-packages\pexpect\expect.py", line 107, in expect_loop
    return self.timeout(e)
  File "C:\Python35\lib\site-packages\pexpect\expect.py", line 70, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: <pexpect_serial.serial_spawn.SerialSpawn object at 0x00000000043B2080>
searcher: searcher_re:
    0: re.compile("b'Conflict:'")
<pexpect_serial.serial_spawn.SerialSpawn object at 0x00000000043B2080>
searcher: searcher_re:
    0: re.compile("b'Conflict:'")

jwhui, 2017-09-18:
Thanks for the additional detail.

Yes, when the `commissioner panid` command is run, it expects an output starting with `Conflict:`.  Unfortunately, it is difficult to determine the issue without some additional detail.  You mentioned that you have sniffer traces for both the passing and failing runs.  Can you attach those to this issue?

bot-zjlu, 2017-09-18:
double-attached twice for same log file to prevent file crash.

[9_2_14_commissioner_failed_sniffer_log.zip](https://github.com/openthread/openthread/files/1312401/9_2_14_commissioner_failed_sniffer_log.zip)

[9_2_4_sniffer_log.zip](https://github.com/openthread/openthread/files/1312399/9_2_4_sniffer_log.zip)

jwhui, 2017-09-21:
@bot-zjlu, thanks for attaching the sniffer traces.  AFAICT, the IEEE 802.15.4 Beacon with the conflicting PAN ID is being sent.  However, it is difficult to tell whether or not ROUTER1 actually receives the IEEE 802.15.4 Beacon message.  Can you debug further to see if ROUTER1 is receiving the Beacon?

bot-zjlu, 2017-09-22:
@Jhui, we are working on..

bot-zjlu, 2017-10-29:
> @bot-zjlu <https://github.com/bot-zjlu>, any updates on this issue?

Hi Jonthan,

Sorry for late reply!  I have other urgent task to do, so no time on this
project.  We still have this problem and will work in it if get time.

Thanks,

zj

jwhui, 2018-01-29:
@bot-zjlu, I'm going to close this stale issue.  Please reopen if/when you want to move it forward.

closed 2018-01-29 by jwhui