openthread/openthread #2225: Case 7.1.3 is failed because the expected packet "unicast data response" is not found (refer to Issue #2107)

xinnnxinnn, 2017-09-27:
Hi,
When I test case 7.1.3  using openthread according to test plan V1.1, the test procedure 2  "
The DUT MUST send a unicast MLE Data Response with the updatedNetwork Data to SED_1." is failed.
I find that in function "SynchronizeChildNetworkData" , leader sends child  update request to SED instead of data response (refer to Issue #2107 "[mle] enable retransmission of network data to SEDs"). So the expected packet “data response” is not found and the case failed.

jwhui, 2017-09-27:
This is a known issue with the Thread Test Harness and is being tracked within the Thread Group as [TESTPLAN-380](https://threadgroup.atlassian.net/browse/TESTPLAN-380).

Sending a Child Update Request is allowed and Thread 1.1.1 Section 5.15.4 states:
```
Routers use the Child Version Number Set to determine when new Thread Network Data
needs to be sent to a Child. For each Child whose network version is out-of-date, the Router
MUST send an MLE Network Data TLV and an MLE Leader Data TLV using either MLE Child
Update Request or MLE Data Response unicast messages containing the TLVs until it
receives a Leader Data TLV with the updated VN_version or VN_stable_version from the
Child.
```

xinnnxinnn, 2017-09-27:
Hi @jwhui,
Thank you for your reply, I will also track this issue.

jwhui, 2018-10-11:
This has been resolved in Thread Test Harness v49.3.

closed 2018-10-11 by jwhui