openthread/openthread #2237: message buffer peak usage - verses "normality" & running out of buffers.

DuaneEllis-TI, 2017-09-29:
This is more of a question about openthreads message buffer implementation, and how buffer management is done.  It seems the default message buffer count is minimal at best and possibly far to small on the NCP device.

I have what I believe to be a simple test case -  NCP + Border Router  + One joining device - Android APP  - During the joining process, the peak buffer usage is 31 (of 40) buffers in use (75% are busy) leaving only 9 buffers, at 5 minutes into the process about 50% of the buffers are still busy - it takes a full 7 minutes to return to what I would call a normal level of activity (ie: 2 busy, 38 free)

it is important to note that in my example test network there is zero "application traffic" - it is only thread network self induced activity.

My question is along these lines - sadly it is not a distinct single question

* For the NCP - is the "40 buffers" reasonable?  Should this be larger?
* Or is this really small bare (spec) minimum, but not enough to be practical?

* If the NCP is a router, and in this example it is AND if there where other devices on the network (ie: sleep devices) and routing between devices - the other network traffic would there would be, and there is an amount of other traffic (there would be several) or other activity (ie: messages routed up stream to the public internet or to the local in-home network)) the NCP will probably run out of buffers during the commissioning process.

Below is some detail and background (data) to my question. 

-Duane.

COMMISSIONING TIME LINE
===================

At time 0 - the NCP is reset, and is not part of a network.

Within about 10 to 15 seconds - I have used the WEB browser and created a network, and I have used the Android App to scan a barcode to commission a new device.

IMPORTANT - Before hand - 

In the "message.cpp" file - where messages are allocated & freed, insert a bit of code to print out the total number of FREE messages each time a message is allocated or free. It is this number we are tracking from here on out.

In this simple use case - the result is a peak usage of 31 of the 40 (75%) leaving 9 free buffers.

If there is ANY other activity (ie:  The NCP might be a router that needs to forward messages) - it often runs out of messages.

Commissioning completes in about 1 minute 
Actual time in my test case is timestamp = 74 seconds.
But I'll use the 1 minute as a nice round number for this discussion.

I'm not - purposefully stopping the android app - I am being a "an average user who would not know to do this"

At +2 minutes (120 seconds)  - there are 18 to 19 free messages (22 are in use)
(there has been a little activity, a few messages allocated and freed)

At +3 minutes (180 seconds) - same: 18 to 19 free messages
(there has been a little activity, a few messages allocated and freed)

Note: at 234 seconds - the NCP announces "commissioner in active" caused by a system level commissioning timeout - I did not 'cancel' the process.

At +4 minutes (240 seconds) -  same: 18 to 19 free messages (20 free)

MOST of these messages are, and have been stuck/lingering in the coap response queue.

At +5 minutes (300 seconds) - the curve begins to bend... 22 free messages (50%)

At +6 minutes (360 seconds) - 31 free messages 
  20 seconds later - there are 32 free messages

At +7 minutes (420 seconds) - 37 free messages - (3 busy messages)

In my example network - a single sample mtd end device does not transmit, it emulates a binary 1/0 output (think lightbulb) - it can locally be turned on/off - or it can remotely (via  coap) change state. But to be clear: There is ZERO application layer network traffic. The numbers here are the network self induced message buffer usage.

From here on out - there are always 36 to 37 free messages - and I don't think there's much to say.

SIMPLE LINUX TEST APP - Polling Devices
===========================

Taking the completed above network (4 devices)
   2 of the devices are "paired" - a temperature sensor and a thermostat, they talk to each other
   2 of the devices are simple actuator type devices (button/light type) they do not talk to each other

On a linux box - I run a simple test script that does the following:

Loop Forever
     For device A, B, C, D
         using "libcoap example app: "coap-client" fetch data from the device.
         NOTE: this is "coap" (non-secure, not secure)

         Using "ping6" - ping the device.
     Sleep 5 seconds

Depending upon what is going on  - this peaks at about 10 buffers in use.  Meaning - commissioning has the possibility of having problems.  I do understand that UDP is lossy - and retries should occur...  but this example test case is small, it seems the this might be very "borderline" at best. 

Possible Bug?
===================

Lastly - I think there is a bug - this is why I started this investigation - but I'm not sure yet.

After running my simple forever loop test (described above) several hours (12 or so), to overnight (24hrs, or over the weekend) - the devices often loose (leak?) a message buffer

I think it would be helpful - on the posix simulation - to introduce some level of "lossy ness" to the network channel.  Meaning today - the network SEND operation is nearly perfect, for simulation purposes - I think it would be very helpful to add a bit of "random loss"  - where in the transmission process maybe  10 to 20% of the packets transmitted are thrown away and not transmitted (simulating a transmission jammer) and likewise in the simulated receiver, 10 to 20% of the packets are also thrown away  (simulating a receiver jammer)

Then - have some means to run the test process overnight for a very long time in both MTD and FTD modes.

-END-

jwhui, 2017-10-02:
The relatively long lifetimes for messages in the CoAP response cache is expected - RFC 7252 specifies a default `EXCHANGE_LIFETIME` value of 247 seconds.  The current OpenThread default limits the CoAP response cache to 10 messages.  Of course, the maximum number of message buffers held in the CoAP response cache depends on the size of the messages.

RFC 7252 allows for bypassing the response cache if the request is idempotent or the overhead/side-effect of performing the action on duplicate requests is preferred compared to caching the response.  DTLS has its own duplicate detection mechanism, so we could probably avoid caching responses for DTLS messages.  At the same time, there are a number of Thread-related CoAP interactions that we could bypass the response cache for.  That should alleviate much of the message buffer pressure.

I am not aware of a message buffer leak.  Let us know if you can confirm the issue and find the root cause.

Finally, 40 message buffers is probably on the smaller side for the Thread router/leader role.  At the same time, the number of message buffers needed depends a lot on application requirements and number of children supported.  This is the primary reason why the Thread Specification also allows a Thread router to advertise its buffering capabilities.

DuaneEllis-TI, 2017-10-04:
An update - there is *NO* buffer leak that we can find

What we do find, is that if/when commissioning is active *AND* other data activity is happening (ie: pings, and coap requests to other nodes) the commissioning process is more prone to failure.

that said, it is often hard to tell if it is the "process it self" the quality level in the Thread Commissioning App

This issue can be closed.

closed 2017-10-04 by DuaneEllis-TI