openthread/openthread #2248: Application coap retransmission problem

GiedriusM, 2017-10-05:
This is identical situation to #2128, except instead of secure coap, the application coap is affected.

GiedriusM, 2017-10-05:
The current solution of GetOwner feels quite awkward especially if yet another exception for application coap would be added. Re-implementing context variable in (single instance) timers could greatly simplify and unify the code.

abtink, 2017-10-10:
The reason behind removing context from `Timer` was to reduce memory (specially for MTD build). Almost all of classes/instance within OT are singletons and we can save RAM by not keeping track of  context pointer for callbacks.  We have few cases where this may not be the case (like `Coap` or `DatasetManager`). Here are some approaches we can consider:

1. Short-term we can use the same solution as in #2128.
2. I think the better (longer-term) approach is to change the class model for  `Coap`. We can define a base class `CoapBase` which practically contains everything in current `Coap` except it does not define the `Timer` member variable, and instead keeps track of a `Timer` reference (or pointer) which its sub-classes should define and pass to base class in its constructor.  We can define different sub-classes of `CoapBase` for different purposes (and ensure each sub-classes is a singleton). 

Thought?

GiedriusM, 2017-10-11:
1. I agree that such workaround is possible as it would be the least disturbing change.
2. I'm not quite sure I follow how this would help to get from a static timer handler context into the class context without a similar mechanism to 1.

--- OFF-TOPIC ---
As a general thought, some things about the Timer do not feel right. First the dependency on the otInstance (with a separate timer scheduler) and then another dependency on a Context, which on one hand (single instance case) has no context and on another has multiple contexts. This comes from my personal view that a timer is (should) be a simple special-case system event which calls your handler with a provided context. I would like to see otTimer API, where an application could use OpenThread timers to schedule it's own events OR the whole Timer, TimerScheduler and Alarm reduced to one static (i.e. single instance) platform-provided timer API. This way there wouldn't be a need for duplicate Timer scheduler implementations - OpenThread timer schedulers for networking and OS or platform timer scheduler for the application. Pros for this are reduced program memory due to duplicate scheduler implementations and simplified code (a lot less GetOwner), which would even further reduce program memory footprint. Cons - like you've mentioned increased RAM usage. I wonder if there's a way to get the best of two.
--- ON-TOPIC---

3. What about creating `CoapTimer`/`ContextTimer`/`EventTimer` (sub)class which would just have a handler coming together with a context just like it's usually done in other timer/callback event APIs?

abtink, 2017-10-11:
On (2) this is what I am thinking: We can have the Coap sub-classes define the timer and its static callback. From the static class method in the sub-class we can then  invoke the base class `HandleTimer()` method. The sub-classes would be singleton so `GetOwner()` would work ok. The base class can still start/stop the timer since it gets a pointer/reference to it in the constructor... (I think this all will work). 
I think this solution probably would have least RAM impact (a `Timer` pointer in base class) and small code change.

(3) is also a good solution. We would need to remember the context and callback pointer in the new `EventTimer/ContextTimer` subclass so probably more RAM impact.

GiedriusM, 2017-10-11:
What about ResponsesQueue::HandleTimer? I think it should also have it's context fixed.

abtink, 2017-10-11:
Good point! It does indeed...

abtink, 2017-10-13:
Submitted https://github.com/openthread/openthread/pull/2264 which should address this. 
Ended up going for a hybrid of (1) and (2) approaches, i.e., defining a Cap base class and 3 sub-classes and passing the handlers from sub-classes.

jwhui, 2017-10-19:
Resolved by #2264.

closed 2017-10-19 by jwhui