openthread/openthread #2254: EUI64 ? SHA256 of EUI64 : while adding joiner

vinvinodk, 2017-10-09:
As per the section 8.10.1.9 of thread specification, User will feed the EUI64 of the device to the Commissioner for adding a thread product. But in native commissioner code of OT, we are mem-comparing the user's EUI64 address against SHA256 of EUI64. 

Code @ Commissioner::HandleRelayReceive API of commissioner.cpp file.

If my understanding is wrong, then user should feed SHA256 of Device's EUI64. What's the right one?

jwhui, 2017-10-10:
At the user-interface level, the input should be the factory-assigned IEEE EUI-64.

I see that the `commissioner joiner add` CLI command does expect the Joiner ID (i.e. SHA-256 of the factory-assigned IEEE EUI-64).

I can look into changing the CLI to support the factory-assigned IEEE EUI-64 instead.

jwhui, 2017-11-02:
@vinvinodk, submitted #2311.

PR #2311 (2017-11-07): [meshcop] use IEEE EUI-64 rather than Joiner ID on user input
Resolves #2254.
closed 2017-11-07 by jwhui