openthread/openthread #2279: [travis] increased failures on Cert_5_3_03_AddressQuery

jwhui, 2017-10-19:
Example log from Travis: [log.txt](https://github.com/openthread/openthread/files/1399705/log.txt)

```
======================================================================
FAIL: test (__main__.Cert_5_3_3_AddressQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "../../../../tests/scripts/thread-cert/Cert_5_3_03_AddressQuery.py", line 151, in test
    msg = dut_messages.next_coap_message('0.02', '/a/aq')
  File "/home/travis/build/jwhui/openthread/build/x86_64-unknown-linux-gnu/openthread-ge4d5cbf-dirty/tests/scripts/thread-cert/message.py", line 317, in next_coap_message
    assert message is not None, "Could not find CoapMessage with code: {}".format(code)
AssertionError: Could not find CoapMessage with code: 0.02

----------------------------------------------------------------------
Ran 1 test in 571.060s
```

hjian2017, 2017-10-20:
Here we power off Router_3 and wait for the DUT to update its address cache and removes all entries based on Router_3’s Router ID.
I discussed this with RongLi before. We thought the average time is :
MAX_NEIGHBOR_AGE + INFINITE_COST_TIMEOUT + ID_REUSE_DELAY + propagation time + transmission time ~ 550s.

We find Harness wait for 580s here. So I will try 580s.

librasungirl, 2017-10-25:
Closed as #2291 already merged

closed 2017-10-25 by librasungirl