openthread/openthread #2299: BUG: Failed Commissioning results in "Frame rx failed, error:UnknownNeighbor" on second try

DuaneEllis-TI, 2017-10-24:
Scenario:
    NCP + Border Router
    Joining Device

During first attempt at commissioning, something goes wrong - commissioning fails.

During the second attempt, the joiner reports:
 [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:78, seqnum:228, type:Data, src:0x6c00, dst:ffeeb4c865ab8cdf, sec:n

Steps to reproduce:
    Start the commissioning process (attempt 1)
    during the process, at some point before completion cause the join to fail.
    Example: remove power from joining device
    Example: Put either Joiner or NCP into an RF shield box
    Harder : Introduce a very noisy "RF jammer"

    Start second commissioning attempt 

During the second attempt - during the "CoAP-Secure" transfer the Joining device reports:

 [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:78, seqnum:228, type:Data, src:0x6c00, dst:ffeeb4c865ab8cdf, sec:n

======
ideas to fix
======

At the discovery request - clean up the tables in the NCP (joiner, leader, etc)
This could be a denial of service attack, an attacker could snoop - and send random random discovery requests causing tables to clean when they should not.

Another would be to clean up at the end of a failed joining operation.

Another approach would be to do this only at the Relay Tx/Rx stage
Rational:  The Commissioner is present, and has agreed/authenticated the join request via cryptographic means "this is the device", and is willing to commission this specific device.

Something else??

Suggestions? We don't have a fix yet.

jwhui, 2017-10-24:
@DuaneEllis-TI, the log message you provided indicates that the Joiner is receiving messages that include a short source address in the IEEE 802.15.4 header.  All messages sent from the Joiner Router to the Joiner should include an extended source address.

Can you provide the log output (INFO level) for both the Joiner Router and Joiner devices during this process?

jwhui, 2017-11-27:
@DuaneEllis-TI, any updates on this issue?

DuaneEllis-TI, 2017-12-04:
This still happens - it is very hard to reproduce, and when we do reproduce - we don't always have all the logging and sniffers going that we wish.

jwhui, 2018-06-06:
Closing due to inactivity.

closed 2018-06-06 by jwhui