openthread/openthread #2324: Silicon Labs Thunderboard 2 support?

beriberikix, 2017-11-07:
https://www.silabs.com/products/development-tools/thunderboard/thunderboard-sense-two-kit

Silicon Labs recently announced a new, low-cost development kit that supports their 802.15.4 hardware. Has anyone been able to test the existing `efr32` platform example?

tomas-c, 2017-11-29:
I have tested MGM111 and MGM12P modules. They all seem to work. Thunderboard uses the EFR32MG12 family chip (same as MGM12P module) so I don't expect it to major any issues. 

You might need to modify the makefiles to compile for the specific chip and depending on whether that module has a FEM might need to port [this](https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2017/08/16/configuring_the_mgm1-1wpa).

jobroe, 2017-12-16:
You can try my [fork.](https://github.com/jobroe/openthread/tree/thunderboard-sense-2) I  changed the make-files according to the Thunderboard Sense 2 platform. Unfortunately there is currently no board config for version 2 included in the SDK. Therefore I used the board config of version 1. I can not give information if everything works but on my desk, the simply getting started ping example is working.

jobroe, 2018-01-30:
Is there any interest in supporting the Silabs Thunderboard Sense 2 platform in general ?
If I would do a pull request , how would you like to add the platform support ?
As additional folder under /examples/platforms/:
```
├── cc2538
├── cc2650
├── cc2652
├── da15000
├── efr32
├── efr32-tbsense2
├── emsk
 ....
```
Or other suggestions ?

jwhui, 2018-01-30:
Would be great to add support for the Thunderboard Sense 2 platform.

From your [fork](https://github.com/jobroe/openthread/tree/thunderboard-sense-2), it appears that the only changes are a CPP define and a board path.  Does it make sense to add conditionals in [examples/Makefile-efr32](https://github.com/openthread/openthread/blob/master/examples/Makefile-efr32) and [examples/platforms/efr32/Makefile.am](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/Makefile.am) based on an environment variable?

_Side note: I realize you ran into some UART issues, but I think we should address those as well.  Moving to use the UARTDRV provided by the SiLabs SDK has the advantage of using a SiLabs-supported driver._

jobroe, 2018-02-08:
I agree to your suggestion and also started to create a proposal, see [here](https://github.com/jobroe/openthread/tree/efr32-configurable-board). With this version it is possible to configure the MCU and board, for example
 `make -f examples/Makefile-efr32 COMMISSIONER=1 BOARD=BRD4161A MCU=EFR32MG12P432F1024GL125`
But I think the selection should be limited to supported boards and the MCU should be selected automatically. I tried this as well, but my problem is that I'm not familiar with automake/autoconf buildsystem. 
At least for the makefile-efr32 something like this would work:
```
# Differentiate between boards
# - BRD4161A / SLWSTK6000B / Wireless Starter Kit
# - BRD4160A / SLTB004A / Thunderboard Sense 2
#
ifeq ($(BOARD),BRD4161A)
MCU = EFR32MG12P432F1024GL125
else
ifeq ($(BOARD),BRD4160A)
MCU = EFR32MG12P332F1024GL125
else
$(info BOARD undefined)
endif
endif
```
The problem is that this won't work for the examples/platforms/efr32/Makefile.am. If my understanding is correct, this should be done with the help of configure.ac. But so far it doesn't contain platform specific parametrization. 
What do you think ? Maybe someone can advise me what to do.

jwhui, 2018-02-08:
@jobroe, I believe all you need is to have a board-specific define (e.g. `-DEFR32MG12P332F1024GL125`) and a board-specific include path (e.g. `-I$(EFR32MG_SDK_SRCDIR)/hardware/kit/EFR32MG12_BRD4166A/config`).  In your example above, how about appending to `COMMONCFLAGS` and `EFR32_MBEDTLS_CPPFLAGS` instead?

jobroe, 2018-02-11:
@jwhui  Unfortunately I'm not sure what is the different between my proposal and your last suggestion. Could you please explain more detailed ?

jwhui, 2018-02-12:
Maybe I misunderstood your previous comment, but I thought you were concerned about having to modify `configure.ac`.  What I'm saying is that the platform-specific Makefile can simply setup `COMMONCFLAGS` and `EFR32_MBEDTLS_CPPFLAGS` without any hooks in `configure.ac`.

jobroe, 2018-02-13:
Okay now I got it, I'm sorry. I updated my proposal, see [here](https://github.com/jobroe/openthread/commit/bf8226502e9352280880dce5572f58ec410a4f7f)
The user can now just provide the board for example for WSTK:
```
make -f examples/Makefile-efr32 COMMISSIONER=1 BOARD=BRD4161A
```
or for Thunderboard Sense 2
```
make -f examples/Makefile-efr32 COMMISSIONER=1 BOARD=BRD4166A
```
What do you think ?

jwhui, 2018-02-14:
@jobroe, your approach looks good to me.

jwhui, 2018-02-27:
Resolved by #2566.

closed 2018-02-27 by jwhui