openthread/openthread #2328: FR: Openthread Platform Abstraction for 802.15.4 MLME-SAP and MCPS-SAP

CiaranWoodward, 2017-11-09:
Hi, I am trying to investigate the possibility of running openthread on top of an external 802.15.4 MAC layer, using the 802.15.4 defined MLME-SAP and MCPS-SAP.

So far, I have successfully got a mostly-working port by:
-Modifying Mac.cpp to use SAP primitives rather than raw PHY frames and manage the PiB security tables as necessary
-Completely changing radio.h to provide a SAP-primitive interface
-Modify MeshForwarder.cpp to use primitives instead of directly building frames, and send indirect transmissions direct to the MAC to queue in the indirect queue
-A few very small modifications dotted around other parts of the stack to allow for getting/giving necessary information for/from the external PIB

I have 2 main questions

1. Would a modification like this have the potential to be merged into the main openthread branch as an optional module? I see a few advantages:
    1. Better platform compatibility by providing an alternate adaptation point
    2. Delegate all of the tougher real-time requirements of 802.15.4 to a coprocessor/real time module which is responsible for the 802.15.4 MAC
    3. Isolate the security processing at the 802.15.4 layer from the rest of openthread

    and I see a few different potential ways of implementation:

    1. As I have it structured currently: a completely different path from meshForwarder down, configured using compile time options to compile different files
    2. Restructure the stack to use SAP primitives entirely and have an optional Mac class which can run below this for interacting with a PHY radio

    Personally, I see i as the more realistic option, although it would result in a less unified stack. Option ii would probably result in more resource usage than openthread's current model for the PHY configuration.

2. I have some questions about the usage of Key Id mode 2 at the MAC layer, specifically the usage of the 'Mode 2 Extended Address'. The Thread spec seems to indicate (and openthread currently seems to match) that when a message is sent with mode2, the originating address should be aExtendedAddress, but the address used in the nonce for CCM should be 0x3506feb823d48712. This seems to be unachievable with a plain 802.15.4 MAC, as section 7.6.3.2 in the IEEE802.15.4 spec states that in the construction of the nonce "The source address shall be set to the extended address aExtendedAddress of the device originating the frame". I also note that section 3.2 of the thread spec states that 7.6.3.2 of the IEEE802.15.4 spec is normative.

    The rest of the thread spec seems to very naturally fit with a MAC-level implementation. And even the rest of the mode2 description seems to fit with the 802.15.4 spec, such as structuring the entire thing around the device descriptor (including forcing a frame counter reset every received frame so it can be used globally).

    So:

    1. Which is correct, using 0x3506feb823d48712 as aExtendedAddress temporarily, or injecting it into the nonce outside of the 802.15.4 spec's guidelines?
    2. I know the mode2 is supposed to aid in 'filtering', but seems to cause a great deal of extra processing, how is it supposed to work?

Apologies for the long issue, I would appreciate any help or comments,

Thanks in advance,
Ciaran Woodward

jwhui, 2017-11-14:
@CiaranWoodward, thanks for submitting this proposal.  There have been a few other cursory requests to support the IEEE 802.15.4 MLME-SAP and MCPS-SAP.  However, it sounds like you have some working code, which is great.

I think you outlined the advantages of supporting the MLME-SAP and MCPS-SAP nicely.  Some of the reasons why OpenThread went with the current model are:

1. Support IEEE 802.15.4 radios that do not provide an MLME-SAP and MCPS-SAP.
2. Minimize MAC-layer resource requirements.  Thread uses a small subset of the IEEE 802.15.4 MAC sublayer.  Linking in a full MLME-SAP and MCPS-SAP implementation can pull in unnecessary code and memory requirements.
3. Allow for cross-layer optimizations.  In particular, OpenThread uses a single message buffer queue across the network and link layers to reduce memory consumption.  Similarly, OpenThread maintains a single neighbor table shared between the network and link layers.
4. More easily implement forward-looking features (i.e. proposed Thread enhancements as part of its Beyond Home initiative).  Some of these forward-looking features rely on lower-layer primitives which are not provided by IEEE 802.15.4 MLME-SAP or MCPS-SAP.

That said, I am supportive of finding a way to support the MLME-SAP and MCPS-SAP.  I'm aware of IEEE 802.15.4 platforms that completely offload the MAC to a separate peripheral.  I agree that having a different path down from MeshForwarder is not ideal but may be the more realistic solution going forward.

As for your question around Key ID Mode 2, I believe setting the Nonce outside IEEE 802.15.4's spec is "more correct."  As mentioned in Section 7.2.1.3:
> For both outgoing and incoming processing, the nonce value for the Key ID Mode 2 frames
input to the CCM* algorithm MUST be set to the ExtAddress value of the Key ID Mode 2
Device Descriptor instead of the value of aExtendedAddress of the originating node.

If the MAC implementation cannot support this Thread-specific behavior, this may be a roadblock to supporting Thread 1.1.  This may be something worth submitting in a [Thread Group JIRA](https://threadgroup.atlassian.net/).

CiaranWoodward, 2017-11-14:
@jwhui Thanks for your response.

I do have some working code using the Cascoda CA-8210 radio, and am currently doing some major refactoring and fixing bugs as they pop up. I have builds which have been stable and able to form flexible networks while handling large loads, and have run the stack using the CA-8210 on several platforms from low-power bare-metal systems to desktop linux systems. 

I agree, and feel that the MLME-SAP and MCPS-SAP interface is largely going to be useful for externally hosted MAC layers. After I've stabilised my current refactor, I can look at sorting out some compile-time options for enabling the SAP path and submitting a pull request. There is a lot of changed code here, so this might be a few weeks.

Regarding the Key ID Mode 2: In that case, a plain MAC is not capable of this behaviour. As far as I am aware from my implementation, this is the only behaviour in Thread 1.1 which is not compatible with normal 802.15.4 MAC behaviour (other than the reconfigurable extended address, but that is common).

What permissions are required to submit a JIRA? It seems to be blocked for me, and im wondering if that's because the company I work for only has Affiliate level Thread Group membership (we're very small). Is there any other path I could take?

CiaranWoodward, 2017-11-16:
I'd also like to clarify that receiving and decrypting Mode 2 messages in this way is fine (although the mode 2 key device descriptor should have UniqueDevice=1 as an exception to what is specified in 7.2.2.2.5 of the thread spec 1.1.1). 

It is transmitting that is the problem. I notice that the only required usage is an announce - which does not seem to require the source address for any other processing. Because of this, I don't think it would be critically damaging to use the mode2 address for SAP platforms - assuming a spec change is out of the question.

jwhui, 2017-11-27:
@CiaranWoodward, I would contact help@threadgroup.org about gaining access to the JIRA.

CiaranWoodward, 2017-11-27:
@jwhui I did that and it appears that affiliates don't have access to it unfortunately

>  We use several tools made by Atlassian - JIRA, Bitbucket and Confluence. Nearly all of the tool usage is tied to certification test harness users, commissioning app users and specification development - which is not something Affiliates have access to unfortunately.

CiaranWoodward, 2018-02-01:
Update: I am still working on this after being on hold for a bit, submitted JIRA issues [SPEC-754](https://threadgroup.atlassian.net/browse/SPEC-754) and [SPEC-753](https://threadgroup.atlassian.net/browse/SPEC-753), and will soon be testing against the harness to make sure everything else is compliant.

jwhui, 2022-12-07:
Closing stale issue.

closed 2022-12-07 by jwhui